immortalwrt/immortalwrt #328: 编译失败18.06-k5.4

jh0072110, 2021-03-22:
日志节选如下：
make[3]: Entering directory '/workdir/openwrt/package/ctcgfw/r8168'
1200
touch /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/r8168-8.048.03/.prepared_1eeb18f7a2d94a926cd52101def62ecf_6664517399ebbbc92a37c5bb081b5c53_check
1201
. /workdir/openwrt/include/shell.sh; gzip -dc /workdir/openwrt/dl/r8168-8.048.03.tar.gz | tar -C /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/r8168-8.048.03/.. -xf -
1202
gzip: /workdir/openwrt/dl/r8168-8.048.03.tar.gz: No such file or directory
1203
tar: This does not look like a tar archive
1204
tar: Exiting with failure status due to previous errors
1205
Makefile:46: recipe for target '/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/r8168-8.048.03/.prepared_1eeb18f7a2d94a926cd52101def62ecf_6664517399ebbbc92a37c5bb081b5c53' failed
1206
make[3]: *** [/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/r8168-8.048.03/.prepared_1eeb18f7a2d94a926cd52101def62ecf_6664517399ebbbc92a37c5bb081b5c53] Error 2
1207
make[3]: Leaving directory '/workdir/openwrt/package/ctcgfw/r8168'
1208
time: package/ctcgfw/r8168/compile#0.15#0.04#0.18
1209
    ERROR: package/ctcgfw/r8168 failed to build.
1210
make[2]: *** [package/ctcgfw/r8168/compile] Error 1
1211
package/Makefile:112: recipe for target 'package/ctcgfw/r8168/compile' failed
1212
make[2]: Leaving directory '/workdir/openwrt'
1213
make[1]: *** [/workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
1214
package/Makefile:108: recipe for target '/workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
1215
make[1]: Leaving directory '/workdir/openwrt'
1216
make: *** [world] Error 2
1217
/workdir/openwrt/include/toplevel.mk:228: recipe for target 'world' failed
1218
Error: Process completed with exit code 2.

jh0072110, 2021-03-22:
[11_Compile the firmware.txt](https://github.com/immortalwrt/immortalwrt/files/6184385/11_Compile.the.firmware.txt)
and…完整日志文件……

1715173329, 2021-03-23:
I will fix this later.

1715173329, 2021-03-23:
Fixed.

closed 2021-03-23 by 1715173329