immortalwrt/immortalwrt #333: master分支编译报错

asiaon123, 2021-03-27:
* check_conflicts_for: The following packages conflict with kmod-fs-exfat0:
 * check_conflicts_for: 	kmod-fs-exfat * 
 * opkg_install_cmd: Cannot install package kmod-fs-exfat0.
 * check_conflicts_for: The following packages conflict with kmod-fs-exfat0:
 * check_conflicts_for: 	kmod-fs-exfat * 
 * opkg_install_cmd: Cannot install package automount.
make[2]: *** [package/install] Error 255
make[1]: *** [/workdir/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.package_install] Error 2
make: *** [world] Error 2
package/Makefile:66: recipe for target 'package/install' failed
make[2]: Leaving directory '/workdir/openwrt'
package/Makefile:109: recipe for target '/workdir/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.package_install' failed
make[1]: Leaving directory '/workdir/openwrt'
/workdir/openwrt/include/toplevel.mk:228: recipe for target 'world' failed

asiaon123, 2021-03-27:
之前编译一直都好的

AmadeusGhost, 2021-03-27:
旧配置删掉重新选就行

qinyuhang, 2021-03-27:
.config里面的 `kmod-fs-exfat` 注释掉再重新编译。

asiaon123, 2021-03-27:
好的，我试试看

closed 2021-03-27 by 1715173329