immortalwrt/immortalwrt #335: luci-app-kodexplorer

lexin8, 2021-03-30:
`Collected errors:
 * pkg_hash_fetch_best_installation_candidate: Packages for nginx-ssl found, but incompatible with the architectures configured
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-kodexplorer:
 * 	nginx-ssl-util (>= 1.5-1)
 * opkg_install_cmd: Cannot install package luci-app-kodexplorer.
package/Makefile:65: recipe for target 'package/install' failed
make[2]: *** [package/install] Error 255
make[2]: Leaving directory '/home/xin/immortalwrt'
package/Makefile:108: recipe for target '/home/xin/immortalwrt/staging_dir/target-x86_64_musl/stamp/.package_install' failed
make[1]: *** [/home/xin/immortalwrt/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/xin/immortalwrt'
Build failed - please re-run with -j1 to see the real error message
/home/xin/immortalwrt/include/toplevel.mk:228: recipe for target 'world' failed
make: *** [world] Error 1`

BabXhwZUNBeTV, 2021-03-30:
> `Collected errors:
>
> * pkg_hash_fetch_best_installation_candidate: Packages for nginx-ssl found, but incompatible with the architectures configured
> * satisfy_dependencies_for: Cannot satisfy the [...]

**Build failed - please re-run with -j1 to see the real error message**  看这个。先尝试 -j1 参数运行
另外，提 issue 请提供尽可能详细的信息 例如使用的是源码哪个分支 什么目标设备

lexin8, 2021-03-30:
> **Build failed - please re-run with -j1 to see the real error message** 看这个。先尝试 -j1 参数运行
> 另外，提 issue 请提供尽可能详细的信息 例如使用的是源码哪个分支 什么目标设备

openwrt-18.06

`Collected errors:
 * pkg_hash_fetch_best_installation_candidate: Packages for nginx-ssl found, but incompatible with the architectures configured
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-kodexplorer:
 * 	nginx-ssl-util (>= 1.5-1)
 * opkg_install_cmd: Cannot install package luci-app-kodexplorer.
package/Makefile:65: recipe for target 'package/install' failed
make[2]: *** [package/install] Error 255
make[2]: Leaving directory '/home/xin/immortalwrt'
package/Makefile:108: recipe for target '/home/xin/immortalwrt/staging_dir/target-x86_64_musl/stamp/.package_install' failed
make[1]: *** [/home/xin/immortalwrt/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/xin/immortalwrt'
/home/xin/immortalwrt/include/toplevel.mk:228: recipe for target 'world' failed
make: *** [world] Error 2
`

1715173329, 2021-03-30:
Fixed.

commit 6f85331c (2021-03-30): luci-app-kodexplorer: fix dependencies

Renamed `nginx` to `nginx-ssl`.

Fixes: #335
closed 2021-03-30 by 1715173329