immortalwrt/immortalwrt #336: luci-app-kodexplorer and exfat error

chiwaicw, 2021-03-31:
openwrt-18.06-k5.4

 * check_conflicts_for: The following packages conflict with kmod-fs-exfat0:
 * check_conflicts_for: 	kmod-fs-exfat * 
 * opkg_install_cmd: Cannot install package kmod-fs-exfat0.
 * check_conflicts_for: The following packages conflict with kmod-fs-exfat0:
 * check_conflicts_for: 	kmod-fs-exfat * 
 * opkg_install_cmd: Cannot install package automount.
 * pkg_hash_fetch_best_installation_candidate: Packages for nginx-ssl found, but incompatible with the architectures configured
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-kodexplorer:
 * 	nginx-ssl-util (>= 1.5-1)
 * opkg_install_cmd: Cannot install package luci-app-kodexplorer.
make[2]: *** [package/install] Error 255
make[1]: *** [/workdir/openwrt/staging_dir/target-aarch64_generic_musl/stamp/.package_install] Error 2

hue715, 2021-03-31:
@chiwaicw 5.4用exfat0，5.10用exfat，要重新生成config

AmadeusGhost, 2021-04-01:
旧配置删掉重新选择即可

closed 2021-04-01 by AmadeusGhost