immortalwrt/immortalwrt #338: rx errors with proprietary driver (5.0.4.0)?

i3roly, 2021-04-01:
hey my dudes,

hope you are well.

i'm writing today to inquire about whether any of you experience RX errors using the proprietary driver. while this driver may seem "uncool", both personal and external accounts suggest it still superior to the mt76 version (i have both codepaths).

in light of this, i am wondering how you deal with this RX error situation. it seems there are packets being dropped due to overflow. i know i did not experience this phenomenon on the 4.x drivers and i'm curious to learn if this is a known issue with an easy workaround.

i would like to increase the buffer to prevent them from being dropped, or fiddle with CRC support, but i thought one of you would know better about this peculiarity.

AmadeusGhost, 2021-04-01:
In fact, we don’t even know these. I don’t know why you think some of us know these, but I suggest that you can go to these following repositories to ask this questions: https://github.com/hanwckf/rt-n56u , https://github.com/MeIsReallyBa/Redmi2100-WIFI-ProprietaryDriver-linux5.4 . Due to we are unable to answer your question, I choose to close this issue.

closed 2021-04-01 by AmadeusGhost

i3roly, 2021-04-01:
Aight fam.

I thought maybe you guys would know. It's all good!

From: AmadeusGhost
Sent: Thursday, 1 April 2021 7:53 AM
To: immortalwrt/immortalwrt
Reply To: immortalwrt/immortalwrt
Cc: gagan sidhu; Author
Subject: Re: [immortalwrt/immortalwrt] rx errors with proprietary driver (5.0.4.0)? (#338)

In fact, we don’t even know these. I don’t know why you think some of us know these, but I suggest that you can go to these following repositories to ask this questions: https://github.com/hanwckf/rt-n56u , https://github.com/MeIsReallyBa/Redmi2100-WIFI-ProprietaryDriver-linux5.4 . Due to we are unable to answer your question, I choose to close this issue.