immortalwrt/immortalwrt #347: GL-MT1300无线无法选择2.4G频段

CWZ7605, 2021-04-05:
使用github action编译。
使用master分支。
2月7日时候编译过，正常可以正常使用2.4G与5G。
4月1号的时候，重新编译了一次，发现无法选择到2.4G。手动修改wireless配置文件也不行。
于是测试编译时只设置编译型号为GL-MT1300，其他使用默认值，编译的固件依然存在这个问题。
系统日志中报错内容如下：
`Sun Apr  4 21:49:15 2021 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy0.conf (phy wlan0)
Sun Apr  4 21:49:15 2021 daemon.err hostapd: Interface name wlan0 already in use
Sun Apr  4 21:49:15 2021 daemon.notice netifd: radio0 (10727): Command failed: Invalid argument
Sun Apr  4 21:49:15 2021 daemon.notice netifd: radio0 (10727): Device setup failed: HOSTAPD_START_FAILED
Sun Apr  4 21:49:15 2021 daemon.notice netifd: radio0 (10816): WARNING: Variable 'data' does not exist or is not an array/object`
[image]

尝试寻找问题，找了4天没搞定，故提交issue

1715173329, 2021-04-05:
You might need to report this to upstream.

CWZ7605, 2021-04-05:
> 
> 
> You might need to report this to upstream.

天灵大神，上游是openwrt官方源吗？

1715173329, 2021-04-05:
@CWZ7605 Yes

CWZ7605, 2021-04-05:
@CWZ7605 我使用openwrt官方源编译测试一次先。如果存在问题，我向上游报告

CWZ7605, 2021-04-05:
@1715173329 果然是上游的问题。这就去提交issue。感谢！

AmadeusGhost, 2021-04-05:
mt76 的问题，你可以换成 luci-app-mtwifi

closed 2021-04-05 by AmadeusGhost

CWZ7605, 2021-04-06:
> 
> 
> mt76 的问题，你可以换成 luci-app-mtwifi

换了luci-app-mtwifi，发现问题依旧，高级设置里死活选不到2.4G