immortalwrt/immortalwrt #356: 18.06 k5.4 R4S-RK3399 Jellyfin HEVC 10bit 转码问题

JSZMonkey, 2021-04-12:
RK3399安装Jellyfin 转码 HEVC 10bit 的时候 很卡 CPU 100% 不知道是不是不支持还是什么 网上都说 支持 H264 H265 HEVC 10bit

1715173329, 2021-04-12:
你勾选了对应的内核模块吗？

JSZMonkey, 2021-04-12:
勾了

1715173329, 2021-04-12:
根据 data sheet 所描述，不支持 hevc 硬解。

See also:
http://opensource.rock-chips.com/images/2/20/RK3399_Multimedia_Codec_Benchmark_v1.0.pdf
http://rockchip.wikidot.com/rk3399

Ref:
https://t.me/ctcgfw_openwrt_discuss/280360
https://t.me/ctcgfw_openwrt_discuss/280363

closed 2021-04-12 by 1715173329

112288999, 2021-09-09:
> 根据 data sheet 所描述，不支持 hevc 硬解。
> 
> See also:
> http://opensource.rock-chips.com/images/2/20/RK3399_Multimedia_Codec_Benchmark_v1.0.pdf
> http://rockchip.wikidot.com/rk3399
> 
> Ref:
> https://t.me/ctcgfw_openwrt_discuss/280360
> https://t.me/ctcgfw_openwrt_discuss/280363
data sheet 描述支持 H.265/HEVC 10bit: 2160p@60fps(4096x2304) 但是这个支持是由VPU来支持的 并不是gpu  其驱动为https://github.com/rockchip-linux/mpp

zhu327, 2026-09-25:
https://gist.github.com/zhu327/394a8ec77f23602c59b60f900a9d33a9