immortalwrt/immortalwrt #359: 1806最新分支代码编译出错

blueveryday, 2021-04-17:
`make[2]: Leaving directory '/home/ago/immortalwrt'
make[2]: Entering directory '/home/ago/immortalwrt'
make[3]: Entering directory '/home/ago/immortalwrt/target/linux'
make[4]: Entering directory '/home/ago/immortalwrt/target/linux/ramips'
rm -f /home/ago/immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.229/localversion
/home/ago/immortalwrt/scripts/kconfig.pl  + /home/ago/immortalwrt/target/linux/generic/config-4.14 /home/ago/immortalwrt/target/linux/ramips/mt7621/config-4.14 > /home/ago/immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.229/.config.target
awk '/^(#[[:space:]]+)?CONFIG_KERNEL/{sub("CONFIG_KERNEL_","CONFIG_");print}' /home/ago/immortalwrt/.config >> /home/ago/immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.229/.config.target
echo "# CONFIG_KALLSYMS_EXTRA_PASS is not set" >> /home/ago/immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.229/.config.target
echo "# CONFIG_KALLSYMS_ALL is not set" >> /home/ago/immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.229/.config.target
echo "CONFIG_KALLSYMS_UNCOMPRESSED=y" >> /home/ago/immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.229/.config.target
/home/ago/immortalwrt/scripts/package-metadata.pl kconfig /home/ago/immortalwrt/tmp/.packageinfo /home/ago/immortalwrt/.config 4.14 > /home/ago/immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.229/.config.override
/home/ago/immortalwrt/scripts/kconfig.pl 'm+' '+' /home/ago/immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.229/.config.target /dev/null /home/ago/immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.229/.config.override > /home/ago/immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.229/.config.set
mv /home/ago/immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.229/.config.set /home/ago/immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.229/.config.old
grep -v INITRAMFS /home/ago/immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.229/.config.old > /home/ago/immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.229/.config.set
echo 'CONFIG_INITRAMFS_SOURCE=""' >> /home/ago/immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.229/.config.set
echo '# CONFIG_INITRAMFS_FORCE is not set' >> /home/ago/immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.229/.config.set
rm -rf /home/ago/immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/modules
cmp -s /home/ago/immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.229/.config.set /home/ago/immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.229/.config.prev || { cp /home/ago/immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.229/.config.set /home/ago/immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.229/.config; cp /home/ago/immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.229/.config.set /home/ago/immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.229/.config.prev; }
export MAKEFLAGS= ; [ -d /home/ago/immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.229/user_headers ] || make -C /home/ago/immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.229 KCFLAGS="-fmacro-prefix-map=/home/ago/immortalwrt/build_dir/target-mipsel_24kc_musl=target-mipsel_24kc_musl" HOSTCFLAGS="-O2 -I/home/ago/immortalwrt/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="mipsel-openwrt-linux-musl-" ARCH="mips" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Fri Apr 16 16:59:37 2021" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/home/ago/immortalwrt/staging_dir/host/lib" KBUILD_HOSTLDLIBS="-L/home/ago/immortalwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KERNELRELEASE=4.14.229 CC="ccache_cc" INSTALL_HDR_PATH=/home/ago/immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.229/user_headers headers_install
grep '=[ym]' /home/ago/immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.229/.config.set | LC_ALL=C sort | mkhash md5 > /home/ago/immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.229/.vermagic
touch /home/ago/immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.229/.configured
rm -f /home/ago/immortalwrt/build_dir/target-mipsel_24kc_musl/root-ramips/init
make -C /home/ago/immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.229 KCFLAGS="-fmacro-prefix-map=/home/ago/immortalwrt/build_dir/target-mipsel_24kc_musl=target-mipsel_24kc_musl" HOSTCFLAGS="-O2 -I/home/ago/immortalwrt/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="mipsel-openwrt-linux-musl-" ARCH="mips" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Fri Apr 16 16:59:37 2021" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/home/ago/immortalwrt/staging_dir/host/lib" KBUILD_HOSTLDLIBS="-L/home/ago/immortalwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KERNELRELEASE=4.14.229 CC="ccache_cc" all modules
make[5]: Entering directory '/home/ago/immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.229'  CHK     include/config/kernel.release
  CHK     include/generated/uapi/linux/version.h
  CHK     include/generated/utsrelease.h
  CHK     include/generated/bounds.h
  CHK     include/generated/timeconst.h
  CHK     include/generated/asm-offsets.h
  CALL    scripts/checksyscalls.sh
  CHK     scripts/mod/devicetable-offsets.h
  CHK     include/generated/compile.h
  CC      drivers/net/ethernet/mediatek/mt7530.o
drivers/net/ethernet/mediatek/mt7530.c: In function 'mt7530_r32':
drivers/net/ethernet/mediatek/mt7530.c:327:3: error: implicit declaration of function '__mdiobus_write'; did you mean 'mdiobus_write'? [-Werror=implicit-function-declaration]
   __mdiobus_write(priv->bus, 0x1f, 0x1f, (reg >> 6) & 0x3ff);
   ^~~~~~~~~~~~~~~
   mdiobus_write
drivers/net/ethernet/mediatek/mt7530.c:328:9: error: implicit declaration of function '__mdiobus_read'; did you mean 'mdiobus_read'? [-Werror=implicit-function-declaration]
   low = __mdiobus_read(priv->bus, 0x1f, (reg >> 2) & 0xf);
         ^~~~~~~~~~~~~~
         mdiobus_read
cc1: some warnings being treated as errors
make[9]: *** [scripts/Makefile.build:330: drivers/net/ethernet/mediatek/mt7530.o] Error 1
make[8]: *** [scripts/Makefile.build:588: drivers/net/ethernet/mediatek] Error 2
make[7]: *** [scripts/Makefile.build:588: drivers/net/ethernet] Error 2
make[6]: *** [scripts/Makefile.build:588: drivers/net] Error 2
make[5]: *** [Makefile:1054: drivers] Error 2
make[5]: Leaving directory '/home/ago/immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.229'
make[4]: *** [Makefile:27: /home/ago/immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.229/.image] Error 2
make[4]: Leaving directory '/home/ago/immortalwrt/target/linux/ramips'
make[3]: *** [Makefile:13: install] Error 2
make[3]: Leaving directory '/home/ago/immortalwrt/target/linux'
time: target/linux/install#5.73#1.29#7.41
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:25: target/linux/install] Error 1
make[2]: Leaving directory '/home/ago/immortalwrt'
make[1]: *** [target/Makefile:19: /home/ago/immortalwrt/staging_dir/target-mipsel_24kc_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/ago/immortalwrt'
make: *** [/home/ago/immortalwrt/include/toplevel.mk:230: world] Error 2
ago@V310:~/immortalwrt$`

blueveryday, 2021-04-17:
commit
dc9a1187b940c0b062746a1aa3f4280aa6ec2f6f

1715173329, 2021-04-17:
Ping @AmadeusGhost.

AmadeusGhost, 2021-04-17:
`git revert a5b9ff72cff3f25516b9c8142da9c875eb41e044`
It seems that this commit can only work on the kernel 5.4 branch.

blueveryday, 2021-04-18:
回退以后，在install阶段，还是报错。
`make[2]: Leaving directory '/home/ago/immortalwrt'
make[2]: Entering directory '/home/ago/immortalwrt'
make[3]: Entering directory '/home/ago/immortalwrt/target/linux'
make[4]: Entering directory '/home/ago/immortalwrt/target/linux/ramips'
rm -f /home/ago/immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.229/localversion
/home/ago/immortalwrt/scripts/kconfig.pl  + /home/ago/immortalwrt/target/linux/generic/config-4.14 /home/ago/immortalwrt/target/linux/ramips/mt7621/config-4.14 > /home/ago/immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.229/.config.target
awk '/^(#[[:space:]]+)?CONFIG_KERNEL/{sub("CONFIG_KERNEL_","CONFIG_");print}' /home/ago/immortalwrt/.config >> /home/ago/immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.229/.config.target
echo "# CONFIG_KALLSYMS_EXTRA_PASS is not set" >> /home/ago/immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.229/.config.target
echo "# CONFIG_KALLSYMS_ALL is not set" >> /home/ago/immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.229/.config.target
echo "CONFIG_KALLSYMS_UNCOMPRESSED=y" >> /home/ago/immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.229/.config.target
/home/ago/immortalwrt/scripts/package-metadata.pl kconfig /home/ago/immortalwrt/tmp/.packageinfo /home/ago/immortalwrt/.config 4.14 > /home/ago/immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.229/.config.override
/home/ago/immortalwrt/scripts/kconfig.pl 'm+' '+' /home/ago/immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.229/.config.target /dev/null /home/ago/immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.229/.config.override > /home/ago/immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.229/.config.set
mv /home/ago/immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.229/.config.set /home/ago/immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.229/.config.old
grep -v INITRAMFS /home/ago/immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.229/.config.old > /home/ago/immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.229/.config.set
echo 'CONFIG_INITRAMFS_SOURCE=""' >> /home/ago/immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.229/.config.set
echo '# CONFIG_INITRAMFS_FORCE is not set' >> /home/ago/immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.229/.config.set
rm -rf /home/ago/immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/modules
cmp -s /home/ago/immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.229/.config.set /home/ago/immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.229/.config.prev || { cp /home/ago/immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.229/.config.set /home/ago/immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.229/.config; cp /home/ago/immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.229/.config.set /home/ago/immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.229/.config.prev; }
export MAKEFLAGS= ; [ -d /home/ago/immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.229/user_headers ] || make -C /home/ago/immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.229 KCFLAGS="-fmacro-prefix-map=/home/ago/immortalwrt/build_dir/target-mipsel_24kc_musl=target-mipsel_24kc_musl" HOSTCFLAGS="-O2 -I/home/ago/immortalwrt/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="mipsel-openwrt-linux-musl-" ARCH="mips" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Sat Apr 17 13:23:27 2021" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/home/ago/immortalwrt/staging_dir/host/lib" KBUILD_HOSTLDLIBS="-L/home/ago/immortalwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KERNELRELEASE=4.14.229 CC="ccache_cc" INSTALL_HDR_PATH=/home/ago/immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.229/user_headers headers_install
grep '=[ym]' /home/ago/immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.229/.config.set | LC_ALL=C sort | mkhash md5 > /home/ago/immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.229/.vermagic
touch /home/ago/immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.229/.configured
rm -f /home/ago/immortalwrt/build_dir/target-mipsel_24kc_musl/root-ramips/init
make -C /home/ago/immortalwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.229 KCFLAGS="-fmacro-prefix-map=/home/ago/immortalwrt/build_dir/target-mipsel_24kc_musl=target-mipsel_24kc_musl" HOSTCFLAGS="-O2 -I/home/ago/immortalwrt/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="mipsel-openwrt-linux-musl-" ARCH="mips" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Sat Apr 17 13:23:27 2021" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/home/ago/immortalwrt/staging_dir/host/lib" KBUILD_HOSTLDLIBS="-L/home/ago/immortalwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KERNELRELEA [... cut ...]
make[3]: Leaving directory '/home/ago/immortalwrt/target/linux'
time: target/linux/install#5.17#1.28#6.35
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:25: target/linux/install] Error 1
make[2]: Leaving directory '/home/ago/immortalwrt'
make[1]: *** [target/Makefile:19: /home/ago/immortalwrt/staging_dir/target-mipsel_24kc_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/ago/immortalwrt'
make: *** [/home/ago/immortalwrt/include/toplevel.mk:230: world] Error 2
ago@V310:~/immortalwrt$`

1715173329, 2021-04-18:
先运行 make target/clean，然后再 make 试试

closed 2021-04-18 by 1715173329

blueveryday, 2021-04-19:
OK，搞定了。

commit fad6849f (2021-04-18): Revert "ramips: mt7530 swconfig: fix race condition in register access"

This reverts commit a5b9ff72cff3f25516b9c8142da9c875eb41e044. Fix: #359