immortalwrt/immortalwrt #363: 修改任何配置应用时都会卡在 /etc/config/firewall 步骤

zhyi828, 2021-04-20:
设备：Nonopi R2S；版本：ImmortalWrt 18.06-SNAPSHOT r9323-e541e2a0c
问题：
  已有几条端口转发记录，之后在修改任何配置时“正在应用更改”都会卡在“/etc/config/firewall”步骤，可以通过点击其他条目解决，同时目测修改应该已经生效，只是每次配置过程都会卡在这一步 无法自动完成。尝试恢复出厂设置也在增加了两条端口转发 过了一夜之后复现了。辛苦大佬帮忙看看。
[image]
[image]

1715173329, 2021-04-20:
防火墙卡了
SSH 运行 `killall -9 fw3; fw3 reload` 看看是什么卡住的

zhyi828, 2021-04-20:
> 防火墙卡了
> SSH 运行 `killall -9 fw3; fw3 reload` 看看是什么卡住的

[image]
是因为最后的这几个path error嘛？

1715173329, 2021-04-20:
> 因为最后的这几个path error嘛？

你的防火墙因为玄学卡住了，现在又因为玄学恢复正常了。

zhyi828, 2021-04-20:
> > 因为最后的这几个path error嘛？
> 
> 你的防火墙因为玄学卡住了，现在又因为玄学恢复正常了。

啊这😂但是我刚刚尝试修改了一下配置 还是会卡在防火墙设置这里😭

1715173329, 2021-04-20:
也许是一些插件导致的问题
adbyby or mia (上网控制)

zhyi828, 2021-04-20:
> 也许是一些插件导致的问题
> adbyby or mia (上网控制)

刚刚尝试把adbyby相关的三个软件包都移除了，然后重启防火墙 还是会卡住。。我看看尝试更新一下固件吧

ricksuzade-maker, 2021-04-20:
最新版源码好像还有一个问题，就是ssr+怎么都起不来，提示tcp relay error

1715173329, 2021-04-29:
Closed due to stale, reopen is welcome.

closed 2021-04-29 by 1715173329

bi-kai, 2021-12-21:
在配置端口转发时，遇到了相同的问题。

<html>
<body>

主机名 | ImmortalWrt
-- | --
型号 | FriendlyElec NanoPi R4S
架构 | ARMv8 Processor rev 4 (v8l) x 6 (1608MHz 1200MHz, 45.0°C)
固件版本 | ImmortalWrt 18.06-SNAPSHOT r10525-4af9b12df / LuCI openwrt-18.06-k5.4 branch (git-21.204.22671-a14e5d4)
内核版本 | 5.4.134
本地时间 | Tue Dec 21 21:14:18 2021

</body>
</html>

zuosc, 2022-09-23:
@1715173329  一样 反复出现。

iwanglei1, 2023-01-31:
> 我实测关闭广告屏蔽大师后就正常了
对的，你说的很对，关闭之后就可以正常配置了。
如果不关闭的话，卡住的时候刷新下页面，然后进防火墙页面重新点下应用就可以了