immortalwrt/immortalwrt #364: Failed compile for nanopi-r4s

anmansky, 2021-04-20:
[1_build (1).txt](https://github.com/immortalwrt/immortalwrt/files/6347196/1_build.1.txt)
[image]

ricksuzade-maker, 2021-04-21:
刷新代码后，依然编译失败

1715173329, 2021-04-21:
日志？

ricksuzade-maker, 2021-04-21:
make[4]: Leaving directory '/home/zhiliang/scripts/immortal/r4s/immortalwrt/build_dir/target-aarch64_generic_musl/rapidjson-2021-01-06-585042c0'
touch /home/zhiliang/scripts/immortal/r4s/immortalwrt/build_dir/target-aarch64_generic_musl/rapidjson-2021-01-06-585042c0/.built
rm -rf /home/zhiliang/scripts/immortal/r4s/immortalwrt/tmp/stage-rapidjson
mkdir -p /home/zhiliang/scripts/immortal/r4s/immortalwrt/tmp/stage-rapidjson/host /home/zhiliang/scripts/immortal/r4s/immortalwrt/staging_dir/target-aarch64_generic_musl/packages /home/zhiliang/scripts/immortal/r4s/immortalwrt/staging_dir/host/packages
install -d -m0755 /home/zhiliang/scripts/immortal/r4s/immortalwrt/tmp/stage-rapidjson
cp -fpR /home/zhiliang/scripts/immortal/r4s/immortalwrt/build_dir/target-aarch64_generic_musl/rapidjson-2021-01-06-585042c0/ipkg-install/* /home/zhiliang/scripts/immortal/r4s/immortalwrt/tmp/stage-rapidjson/
find /home/zhiliang/scripts/immortal/r4s/immortalwrt/tmp/stage-rapidjson -name '*.la' | xargs -r rm -f; 
if [ -f /home/zhiliang/scripts/immortal/r4s/immortalwrt/staging_dir/target-aarch64_generic_musl/packages/rapidjson.list ]; then /home/zhiliang/scripts/immortal/r4s/immortalwrt/scripts/clean-package.sh "/home/zhiliang/scripts/immortal/r4s/immortalwrt/staging_dir/target-aarch64_generic_musl/packages/rapidjson.list" "/home/zhiliang/scripts/immortal/r4s/immortalwrt/staging_dir/target-aarch64_generic_musl"; fi
if [ -d /home/zhiliang/scripts/immortal/r4s/immortalwrt/tmp/stage-rapidjson ]; then (cd /home/zhiliang/scripts/immortal/r4s/immortalwrt/tmp/stage-rapidjson; find ./ > /home/zhiliang/scripts/immortal/r4s/immortalwrt/tmp/stage-rapidjson.files); 	SHELL= flock /home/zhiliang/scripts/immortal/r4s/immortalwrt/tmp/.staging-dir.flock -c ' mv /home/zhiliang/scripts/immortal/r4s/immortalwrt/tmp/stage-rapidjson.files /home/zhiliang/scripts/immortal/r4s/immortalwrt/staging_dir/target-aarch64_generic_musl/packages/rapidjson.list && cp -fpR /home/zhiliang/scripts/immortal/r4s/immortalwrt/tmp/stage-rapidjson/* /home/zhiliang/scripts/immortal/r4s/immortalwrt/staging_dir/target-aarch64_generic_musl/; '; fi
rm -rf /home/zhiliang/scripts/immortal/r4s/immortalwrt/tmp/stage-rapidjson
touch /home/zhiliang/scripts/immortal/r4s/immortalwrt/staging_dir/target-aarch64_generic_musl/stamp/.rapidjson_installed
make[3]: Leaving directory '/home/zhiliang/scripts/immortal/r4s/immortalwrt/feeds/packages/libs/rapidjson'
time: package/feeds/packages/rapidjson/compile#15.51#1.69#18.44
                                                                                17839800 ->   4059012   22.75%   linux/arm64   main

Packed 1 file.
CONFIG_GOLANG_MOD_CACHE_WORLD_READABLE="" GO_BUILD_CACHE_DIR="/home/zhiliang/scripts/immortal/r4s/immortalwrt/tmp/go-build" GO_MOD_CACHE_DIR="/home/zhiliang/scripts/immortal/r4s/immortalwrt/dl/go-mod-cache" GO_MOD_ARGS="-modcacherw" /usr/bin/env bash ../../lang/golang//golang-build.sh cache_cleanup
touch /home/zhiliang/scripts/immortal/r4s/immortalwrt/build_dir/target-aarch64_generic_musl/Xray-core-1.4.2/.built
rm -rf /home/zhiliang/scripts/immortal/r4s/immortalwrt/build_dir/target-aarch64_generic_musl/Xray-core-1.4.2/.pkgdir/xray-core.installed /home/zhiliang/scripts/immortal/r4s/immortalwrt/build_dir/target-aarch64_generic_musl/Xray-core-1.4.2/.pkgdir/xray-core
mkdir -p /home/zhiliang/scripts/immortal/r4s/immortalwrt/build_dir/target-aarch64_generic_musl/Xray-core-1.4.2/.pkgdir/xray-core
CONFIG_GOLANG_MOD_CACHE_WORLD_READABLE="" GO_BUILD_CACHE_DIR="/home/zhiliang/scripts/immortal/r4s/immortalwrt/tmp/go-build" GO_MOD_CACHE_DIR="/home/zhiliang/scripts/immortal/r4s/immortalwrt/dl/go-mod-cache" GO_MOD_ARGS="-modcacherw" GO_PKG="github.com/xtls/xray-core" GO_INSTALL_EXTRA="" GO_INSTALL_ALL="" GO_SOURCE_ONLY="" GO_BUILD_PKG="github.com/xtls/xray-core/main" GO_EXCLUDES="" GO_GO_GENERATE="" GO_INSTALL_BIN_PATH="/usr/bin" BUILD_DIR="/home/zhiliang/scripts/immortal/r4s/immortalwrt/build_dir/target-aarch64_generic_musl/Xray-core-1.4.2" GO_BUILD_DIR="/home/zhiliang/scripts/immortal/r4s/immortalwrt/build_dir/target-aarch64_generic_musl/Xray-core-1.4.2/.go_work/build" GO_BUILD_BIN_DIR="/home/zhiliang/scripts/immortal/r4s/immortalwrt/build_dir/target-aarch64_generic_musl/Xray-core-1.4.2/.go_work/build/bin/linux_arm64" GO_BUILD_DEPENDS_PATH="/usr/share/gocode" GO_BUILD_DEPENDS_SRC="/home/zhiliang/scripts/immortal/r4s/immortalwrt/staging_dir/target-aarch64_generic_musl/usr/share/gocode/src" /usr/bin/env bash ../../lang/golang//golang-build.sh install_bin "/home/zhiliang/scripts/immortal/r4s/immortalwrt/build_dir/target-aarch64_generic_musl/Xray-core-1.4.2/ipkg-install"
install -d -m0755 /home/zhiliang/scripts/immortal/r4s/immortalwrt/build_dir/target-aarch64_generic_musl/Xray-core-1.4.2/.pkgdir/xray-core/usr/bin/
install -m0755 /home/zhiliang/scripts/immortal/r4s/immortalwrt/build_dir/target-aarch64_generic_musl/Xray-core-1.4.2/ipkg-install/usr/bin/main /home/zhiliang/scripts/immortal/r4s/immortalwrt/build_dir/target-aarch64_generic_musl/Xray-core-1.4.2/.pkgdir/xray-core/usr/bin/xray
ln -sf xray /home/zhiliang/scripts/immortal/r4s/immortalwrt/build_dir/target-aarch64_generic_musl/Xray-core-1.4.2/.pkgdir/xray-core/usr/bin/v2ray
install -d -m0755 /home/zhiliang/scripts/immortal/r4s/immortalwrt/build_dir/target-aarch64_generic_musl/Xray-core-1.4.2/.pkgdir/xray-core/etc/xray/
[... cut ...]
make[3]: Leaving directory '/home/zhiliang/scripts/immortal/r4s/immortalwrt/feeds/packages/net/xray-core'
time: package/feeds/packages/xray-core/compile#37.00#4.56#41.35
make[2]: Leaving directory '/home/zhiliang/scripts/immortal/r4s/immortalwrt'
make[1]: *** [package/Makefile:108: /home/zhiliang/scripts/immortal/r4s/immortalwrt/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/zhiliang/scripts/immortal/r4s/immortalwrt'
make: *** [/home/zhiliang/scripts/immortal/r4s/immortalwrt/include/toplevel.mk:230：world] 错误 2

1715173329, 2021-04-21:
在你给的日志中，我找不到任何有价值的内容

noreph, 2021-04-21:
我也编译出错了，日志：https://paste.ubuntu.com/p/HPSk6b6PYQ/

1715173329, 2021-04-21:
看来是一些函数名发生了变动

1715173329, 2021-04-21:
已修复。

closed 2021-04-21 by 1715173329

ricksuzade-maker, 2021-04-21:
完美通过编译，给天灵大佬点个大赞！

commit 9d8bca43 (2021-04-21): uboot-rockchip: refresh patches

Fix: #364
commit f1e564e7 (2021-04-21): uboot-rockchip: nanopi4: rename DM_GET_DRIVER to DM_DRIVER_GET

Upstream commit: u-boot/u-boot@65e25bea597cca ("dm: Rename DM_GET_DRIVER() to DM_DRIVER_GET()")

Fixes: #364
commit 290eb024 (2021-04-21): uboot-rockchip: refresh patches

Fix: #364
commit fb3bac8f (2021-04-21): uboot-rockchip: nanopi4: rename DM_GET_DRIVER to DM_DRIVER_GET

Upstream commit: u-boot/u-boot@65e25bea597cca ("dm: Rename DM_GET_DRIVER() to DM_DRIVER_GET()")

Fixes: #364