immortalwrt/immortalwrt #369: 18.06 k5.4 failed, lm-sensor brocen pipe

anmansky, 2021-04-23:
mkdir -p /workdir/immortalwrt/dl
2021-04-22T23:08:17.1115153Z SHELL= flock /workdir/immortalwrt/tmp/.lm-sensors-3.6.0.tar.gz.flock -c '  	/workdir/immortalwrt/scripts/download.pl "/workdir/immortalwrt/dl" "lm-sensors-3.6.0.tar.gz" "0591f9fa0339f0d15e75326d0365871c2d4e2ed8aa1ff759b3a55d3734b7d197" "" "https://codeload.github.com/lm-sensors/lm-sensors/tar.gz/V3-6-0?"    '
2021-04-22T23:08:17.1688973Z mkdir -p /workdir/immortalwrt/staging_dir/target-aarch64_generic_musl/root-rockchip/stamp
2021-04-22T23:08:17.1712863Z echo '5' | cmp -s - /workdir/immortalwrt/staging_dir/target-aarch64_generic_musl/pkginfo/libsensors.version || { echo '5' > /workdir/immortalwrt/staging_dir/target-aarch64_generic_musl/pkginfo/libsensors.version;  } 
2021-04-22T23:08:17.1737496Z SHELL= flock /workdir/immortalwrt/tmp/.root-copy.flock -c 'cp -fpR /workdir/immortalwrt/build_dir/target-aarch64_generic_musl/lm-sensors-3-6-0/.pkgdir/libsensors/. /workdir/immortalwrt/staging_dir/target-aarch64_generic_musl/root-rockchip/'
2021-04-22T23:08:17.1782388Z touch /workdir/immortalwrt/staging_dir/target-aarch64_generic_musl/root-rockchip/stamp/.libsensors_installed
2021-04-22T23:08:17.1872724Z echo "libsensors" >> /workdir/immortalwrt/staging_dir/target-aarch64_generic_musl/pkginfo/lm-sensors.default.install
2021-04-22T23:08:18.1149931Z <built-in>: fatal error: when writing output to : Broken pipe
2021-04-22T23:08:18.1150990Z mkdir -p /workdir/immortalwrt/staging_dir/target-aarch64_generic_musl/root-rockchip/stamp
2021-04-22T23:08:18.1158865Z compilation terminated.

AmadeusGhost, 2021-04-23:
Just tested that this package can compile normally, maybe you can try again with this command `make clean`.

anmansky, 2021-04-25:
Ok, close!

closed 2021-04-25 by anmansky