immortalwrt/immortalwrt #375: 4月30日版18.06-k5.4 刷入R2S和R4S后qBittorrent无法正常启动，提示缺少libboost_system.so.1.75.0

ricksuzade-maker, 2021-04-30:
Fri Apr 30 21:38:07 2021 user.info watchcat: starting all tasks
Fri Apr 30 21:38:07 2021 user.info watchcat: started task (mode=ping_reboot;period=21600;pinghosts=8.8.8.8;pingperiod=60;forcedelay=30;pingsize=standard)
Fri Apr 30 21:38:07 2021 user.info watchcat: all tasks started
Fri Apr 30 21:38:07 2021 daemon.notice procd: /etc/rc.d/S98beardropper: beardropper started!
Fri Apr 30 21:38:07 2021 authpriv.notice beardropper[10563]: loadState() loaded 0 entries
Fri Apr 30 21:38:07 2021 daemon.info ksmbd: triggering kill_server
Fri Apr 30 21:38:07 2021 daemon.notice ksmbd: Starting Ksmbd userspace service.
Fri Apr 30 21:38:07 2021 kern.info kernel: [   76.500561] ksmbd: kill command received
Fri Apr 30 21:38:07 2021 authpriv.notice beardropper[10563]: loadState() loaded 0 entries
Fri Apr 30 21:38:07 2021 authpriv.notice beardropper[10563]: Running in follow mode
Fri Apr 30 21:38:07 2021 daemon.info dnsmasq[10414]: Connected to system UBus
Fri Apr 30 21:38:07 2021 authpriv.notice beardropper[10563]: loadState() loaded 0 entries
Fri Apr 30 21:38:16 2021 user.info mwan3track[5719]: Check (ping) failed for target "8.8.8.8" on interface wan (eth0)
Fri Apr 30 21:38:17 2021 daemon.info frps: Instance "main" disabled.
Fri Apr 30 21:38:17 2021 daemon.notice procd: /etc/rc.d/S99frps: [info] Instance "main" disabled.
Fri Apr 30 21:38:17 2021 daemon.notice procd: /etc/rc.d/S99gost: uci: Entry not found
Fri Apr 30 21:38:18 2021 daemon.notice procd: /etc/rc.d/S99qbittorrent: Error loading shared library libboost_system.so.1.75.0: No such file or directory (needed by /usr/bin/qbittorrent-nox)
Fri Apr 30 21:38:18 2021 daemon.notice procd: /etc/rc.d/S99serverchan: serverchan is starting now ...
Fri Apr 30 21:38:18 2021 daemon.err miniupnpd[5897]: try_sendto(sock=6, len=464, dest=239.255.255.250:1900): sendto: Operation not permitted
Fri Apr 30 21:38:18 2021 daemon.err miniupnpd[5897]: try_sendto(sock=6, len=464, dest=239.255.255.250:1900): sendto: Operation not permitted
Fri Apr 30 21:38:18 2021 daemon.err miniupnpd[5897]: try_sendto(sock=6, len=480, dest=239.255.255.250:1900): sendto: Operation not permitted
Fri Apr 30 21:38:18 2021 daemon.err miniupnpd[5897]: try_sendto(sock=6, len=462, dest=239.255.255.250:1900): sendto: Operation not permitted
Fri Apr 30 21:38:18 2021 daemon.err miniupnpd[5897]: try_sendto(sock=6, len=409, dest=239.255.255.250:1900): sendto: Operation not permitted
Fri Apr 30 21:38:18 2021 daemon.err miniupnpd[5897]: try_sendto(sock=6, len=448, dest=239.255.255.250:1900): sendto: Operation not permitted
Fri Apr 30 21:38:18 2021 daemon.err miniupnpd[5897]: try_sendto(sock=6, len=409, dest=239.255.255.250:1900): sendto: Operation not permitted
Fri Apr 30 21:38:18 2021 daemon.err miniupnpd[5897]: try_sendto(sock=6, len=468, dest=239.255.255.250:1900): sendto: Operation not permitted
Fri Apr 30 21:38:18 2021 daemon.err miniupnpd[5897]: try_sendto(sock=6, len=409, dest=239.255.255.250:1900): sendto: Operation not permitted
Fri Apr 30 21:38:18 2021 daemon.err miniupnpd[5897]: try_sendto(sock=6, len=472, dest=239.255.255.250:1900): sendto: Operation not permitted
Fri Apr 30 21:38:18 2021 daemon.err miniupnpd[5897]: try_sendto(sock=6, len=400, dest=239.255.255.250:1900): sendto: Operation not permitted
Fri Apr 30 21:38:18 2021 daemon.err miniupnpd[5897]: try_sendto failed to send 11 packets
Fri Apr 30 21:38:19 2021 cron.err crond[12072]: crond (busybox 1.33.0) started, log level 8
Fri Apr 30 21:38:20 2021 user.notice Subconverter: Starting Subconverter service
Fri Apr 30 21:38:21 2021 daemon.notice ttyd[12451]: ttyd 1.6.3 (libwebsockets 4.1.6-e636a2fb6)
Fri Apr 30 21:38:21 2021 daemon.notice ttyd[12451]: tty configuration:
Fri Apr 30 21:38:21 2021 daemon.notice ttyd[12451]:   start command: /bin/login
Fri Apr 30 21:38:21 2021 daemon.notice ttyd[12451]:   close signal: SIGHUP (1)
Fri Apr 30 21:38:21 2021 daemon.notice ttyd[12451]:   terminal type: xterm-256color
Fri Apr 30 21:38:21 2021 daemon.notice ttyd[12451]: LWS: 4.1.6-e636a2fb6, loglevel 7
Fri Apr 30 21:38:21 2021 daemon.notice ttyd[12451]: NET CLI SRV H1 H2 WS IPV6-off
Fri Apr 30 21:38:21 2021 daemon.notice ttyd[12451]:    /usr/lib/libwebsockets-evlib_uv.so
Fri Apr 30 21:38:21 2021 daemon.notice ttyd[12451]:  Using foreign event loop...
Fri Apr 30 21:38:21 2021 daemon.notice ttyd[12451]:  Listening on port: 7681
Fri Apr 30 21:38:21 2021 daemon.info subconverter[12400]: \033]0;SubConverter v0.6.4\007\c
Fri Apr 30 21:38:21 2021 daemon.notice vnstatd[12767]: vnStat daemon 1.18 started. (pid:12767 uid:0 gid:0)
Fri Apr 30 21:38:21 2021 daemon.notice vnstatd[12767]: Monitoring: br-lan (1000 Mbit) eth0 (1000 Mbit)
Fri Apr 30 21:38:21 2021 user.info wsdd2[12851]: starting.
Fri Apr 30 21:38:21 2021 daemon.notice procd: /etc/rc.d/S99zerotier: disabled in config
Fri Apr 30 21:38:21 2021 daemon.info procd: - init complete -

1715173329, 2021-04-30:
重新编译 qBittorrent，或重新编译整个 OpenWrt。

ricksuzade-maker, 2021-04-30:
收到，make clean后重新编译看能不能解决，待会儿汇报

zxlhhyccc, 2021-04-30:
> 收到，make clean后重新编译看能不能解决，待会儿汇报

加入boost_system依赖即可解决编译问题。。。

ricksuzade-maker, 2021-04-30:
make clean后全部重新编译，果然解决问题了，谢谢天灵大.

closed 2021-04-30 by 1715173329

KnowledegAndAction, 2025-08-29:
> > 收到，make clean后重新编译看能不能解决，待会儿汇报
> 
> 加入boost_system或安装libboost_system即可问题。。。

请问如何加入或安装？

lyin888, 2025-08-29:
Libraries  --->  <*必须选择后才会出现子选项> boost  --->Select Boost libraries  --->< > Boost system library. (NEW)