immortalwrt/immortalwrt #382: 21.02的编译k3错误，请各位大神看看什么情况，谢谢！

ly20202020, 2021-05-19:
rm -f /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/tplink_archer-c5-v2-kernel.bin
cp /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/zImage /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/tplink_archer-c5-v2-kernel.bin
cat /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/image-bcm47081-tplink-archer-c5-v2.dtb >> /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/tplink_archer-c5-v2-kernel.bin
/workdir/openwrt/staging_dir/host/bin/lzma e /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/tplink_archer-c5-v2-kernel.bin -d16  /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/tplink_archer-c5-v2-kernel.bin.new
mv /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/tplink_archer-c5-v2-kernel.bin.new /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/tplink_archer-c5-v2-kernel.bin
[ -f /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/tplink_archer-c5-v2-kernel.bin -a -f /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/root.squashfs ]
dd if=/workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/root.squashfs >> /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/tmp/immortalwrt-bcm53xx-generic-tplink_archer-c5-v2-squashfs.bin
34067+1 records in
34067+1 records out
17442758 bytes (17 MB, 17 MiB) copied, 0.0912715 s, 191 MB/s
/workdir/openwrt/staging_dir/host/bin/trx -o /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/tmp/immortalwrt-bcm53xx-generic-tplink_archer-c5-v2-squashfs.bin.trx -m 33554432 -f /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/tplink_archer-c5-v2-kernel.bin -a 1024
mjn3's trx replacement - v0.81.1
WARNING: maxlen exceeds default maximum!  Beware of overwriting nvram!
/workdir/openwrt/staging_dir/host/bin/tplink-safeloader -B ARCHER-C5-V2 -k /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/tmp/immortalwrt-bcm53xx-generic-tplink_archer-c5-v2-squashfs.bin.trx -r /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/tmp/immortalwrt-bcm53xx-generic-tplink_archer-c5-v2-squashfs.bin -j -o /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/tmp/immortalwrt-bcm53xx-generic-tplink_archer-c5-v2-squashfs.bin.new
os-image partition too big (more than 2097152 bytes): Success
make[5]: *** [Makefile:456: /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/tmp/immortalwrt-bcm53xx-generic-tplink_archer-c5-v2-squashfs.bin] Error 1
make[5]: Leaving directory '/workdir/openwrt/target/linux/bcm53xx/image'
make[4]: *** [Makefile:30: install] Error 2
make[4]: Leaving directory '/workdir/openwrt/target/linux/bcm53xx'
make[3]: *** [Makefile:11: install] Error 2
make[3]: Leaving directory '/workdir/openwrt/target/linux'
time: target/linux/install#58.99#10.62#58.75
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:25: target/linux/install] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [target/Makefile:19: /workdir/openwrt/staging_dir/target-arm_cortex-a9_musl_eabi/stamp/.target_install] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2

[详细日志.txt](https://github.com/immortalwrt/immortalwrt/files/6507955/default.txt)

啥插件都没增加，就默认的编译，请教各位大神这个什么情况？

EnnawYang, 2021-05-19:
`os-image partition too big (more than 2097152 bytes)`

ly20202020, 2021-05-19:
> `os-image partition too big (more than 2097152 bytes)`

感谢，我自己百度了一下也大概这个意思，但是不懂什么原因造成的，我就默认编译，也没加啥插件，也没修改什么文件啊，奇怪

ElonH, 2021-05-19:
@ly20202020 不是写着了吗？分区最大就2M, 能加的了插件？

ly20202020, 2021-05-19:
> @ly20202020 不是写着了吗？分区最大就2M, 能加的了插件？

感谢！

AmadeusGhost, 2021-05-19:
Error: make[5]: *** [Makefile:456: /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/tmp/immortalwrt-bcm53xx-generic-tplink_archer-c5-v2-squashfs.bin] Error 1
Related source code:
```
define Device/tplink_archer-c5-v2
  BROKEN := y
endef
TARGET_DEVICES += tplink_archer-c5-v2
```
此设备的固件已被禁止生成，你是不是搞错了什么。。

ly20202020, 2021-05-19:
> Error: make[5]: *** [Makefile:456: /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/tmp/immortalwrt-bcm53xx-generic-tplink_archer-c5-v2-squashfs.bin] Error 1
> Related [...]

我也一脸懵啊，以前用18.06的时候没这样错误的，搞21.02编译了x86、k2p、新3的都没问题，就K3的出错了

Ozzy-hzy, 2021-12-12:
> 我也一脸懵啊，以前用18.06的时候没这样错误的，搞21.02编译了x86、k2p、新3的都没问题，就K3的出错了

请问搞定了吗，我用官方21.02.1源码编译k3也出现了同样的问题

zxlhhyccc, 2021-12-12:
> 请问搞定了吗，我用官方21.02.1源码编译k3也出现了同样的问题

和tplink设备有啥关系？这个设备估计是默认了勾选，取消试试吧，我master分支编译一点问题没有，2102和master分支差不多。。。

zxlhhyccc, 2021-12-12:
> 请问搞定了吗，我用官方21.02.1源码编译k3也出现了同样的问题

如果要纯编译k3，可以使用下面的命令：
·····
sed -i 's|^TARGET_|# TARGET_|g; s|# TARGET_DEVICES += phicomm_k3|TARGET_DEVICES += phicomm_k3|' target/linux/bcm53xx/image/Makefile
············
然后再编译试试，目前编译k3如果不屏蔽其他固件的话，会自动把所有的固件都打包出来编译！！！

Ozzy-hzy, 2021-12-12:
> > > > Error: make[5]: *** [Makefile:456: /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/tmp/immortalwrt-bcm53xx-generic-tplink_archer-c5-v2-squashfs.bin] Error 1
> > > > Related source code:
> > > > ```
> > > > define Device/tplink_archer-c5-v2
> > > >   BROKEN := y
> > > > endef
> > > > TARGET_DEVICES += tplink_archer-c5-v2
> > > > ```
> > > > 
> > > > 
> > > >     
> > > >       
> > > >     
> > > > 
> > > >       
> > > >     
> > > > 
> > > >     
> > > >   
> > > > 此设备的固件已被禁止生成，你是不是搞错了什么。。
> > > 
> > > 
> > > 我也一脸懵啊，以前用18.06的时候没这样错误的，搞21.02编译了x86、k2p、新3的都没问题，就K3的出错了
> > 
> > 
> > 请问搞定了吗，我用官方21.02.1源码编译k3也出现了同样的问题
> 
> 如果要纯编译k3，可以使用下面的命令：
> 
> ```
> sed -i 's|^TARGET_|# TARGET_|g; s|# TARGET_DEVICES += phicomm_k3|TARGET_DEVICES += phicomm_k3|' target/linux/bcm53xx/image/Makefile
> ```
> 
> 然后再编译试试，目前编译k3如果不屏蔽其他固件的话，编译完成时会自动把所有的固件都打包出来！！！

谢谢，已经解决了，我把openwrt/target/linux/bcm53xx/image/Makefile文件中关于tplink_archer-c9-v1和tplink_archer-c5-v2这两个型号的行用#注释掉了，编译成功