immortalwrt/immortalwrt #384: 编译openwrt-18.06-k5.4分支出错

mmeiwenlong, 2021-05-21:
应该不是我这里问题……[image]
怎么处理这问题？

AmadeusGhost, 2021-05-22:
请按红字部分提示 j1 跑一遍再发送 log…

mmeiwenlong, 2021-05-22:
> 请按红字部分提示 j1 跑一遍再发送 log…

[image]

zxlhhyccc, 2021-05-22:
应该是nginx-util的编译报错，是因为netifd更新后引起的，将nginx-ssl-util.hpp修改一下就可以编译通过了，补丁为：
```
--- a/net/nginx-util/src/nginx-ssl-util.hpp
+++ b/net/nginx-util/src/nginx-ssl-util.hpp
@@ -166,8 +166,10 @@ static constexpr auto _escape = _Line{
         std::string ret{};
         for (char c : strptr) {
             switch (c) {
-                case '^': ret += '\\'; [[fallthrough]];
-                case '_': [[fallthrough]];
+                case '^': ret += '\\';
+                         /* fallthrough */
+                case '_':
+                         /* fallthrough */
                 case '-': ret += c; break;
                 default:
                     if ((isalpha(c) != 0) || (isdigit(c) != 0)) {
`````
或
````
diff --git a/net/nginx-util/src/nginx-ssl-util.hpp b/net/nginx-util/src/nginx-ssl-util.hpp
index 4f12925db..4484b3550 100644
--- a/net/nginx-util/src/nginx-ssl-util.hpp
+++ b/net/nginx-util/src/nginx-ssl-util.hpp
@@ -166,8 +166,8 @@ static constexpr auto _escape = _Line{
         std::string ret{};
         for (char c : strptr) {
             switch (c) {
-                case '^': ret += '\\'; [[fallthrough]];
-                case '_': [[fallthrough]];
+                case '^': ret += '\\'; /* fallthrough */
+                case '_': /* fallthrough */
                 case '-': ret += c; break;
                 default:
                     if ((isalpha(c) != 0) || (isdigit(c) != 0)) {