immortalwrt/immortalwrt #385: 请问下为什么7620的hc5661一直打不出factory的包？

asiaon123, 2021-05-23:


AmadeusGhost, 2021-05-24:
这个设备本来就没有定义 factory.bin
[image]

asiaon123, 2021-05-24:
问下这个可以增加factory么？

asiaon123, 2021-05-24:
或者问下如果没有factory我通过什么方式可以刷入？😂

BabXhwZUNBeTV, 2021-05-24:
@asiaon123 test

AmadeusGhost, 2021-05-24:
> 问下这个可以增加factory么？

并非所有设备都需要 factory.bin，通常的做法是刷入 breed，然后在 breed 刷入 sysupgrade.bin

asiaon123, 2021-05-24:
> > 问下这个可以增加factory么？
> 
> 并非所有设备都需要 factory.bin，通常的做法是刷入 breed，然后在 breed 刷入 sysupgrade.bin

好的，我已经是刷入breed了，但是不是说breed刷入的需要factory包的么，sysupgrade也是可以的么？我现在刷的是padavan的

AmadeusGhost, 2021-05-24:
> 但是不是说breed刷入的需要factory包的么，sysupgrade也是可以的么？

通常都是刷入 sysupgrade.bin（小米那些奇葩分区设备除外），你试试就知道了

asiaon123, 2021-05-24:
> > 但是不是说breed刷入的需要factory包的么，sysupgrade也是可以的么？
> 
> 通常都是刷入 sysupgrade.bin（小米那些奇葩分区设备除外），你试试就知道了

好的，非常感谢，我晚上回去试试

closed 2021-05-24 by asiaon123