immortalwrt/immortalwrt #386: 问下这个错的话怎么排查是那个包引起的

asiaon123, 2021-05-24:
* check_data_file_clashes: Package libustream-openssl20201210 wants to install file /workdir/openwrt/build_dir/target-mipsel_24kc_musl/root-ramips/lib/libustream-ssl.so
	But that file is already provided by package  * libustream-mbedtls20201210
 * opkg_install_cmd: Cannot install package libustream-openssl20201210.

我发现两个冲突包的都是在同一个里面

1715173329, 2021-05-24:
去掉 libustream-mbedtls 即可
在冲突的包名上按 `?` 即可查询依赖情况

asiaon123, 2021-05-24:
define Package/ipv6helper
	SECTION:=ipv6
	TITLE:=IPv6 Helper and Dynamic Update he.net of ip
	DEPENDS:=+luci-proto-ipv6 +wget-ssl +libip6tc +ip6tables +ip6tables-mod-nat \
		+kmod-ipt-nat6 +kmod-nf-nat6 +odhcpd-ipv6only +odhcp6c +6in4
	PKGARCH:=all
endef

查到是因为ip6helper导致的，本来我打算用这个包来支持IPv6的，问下如果要支持IPv6的话需要哪些包？

1715173329, 2021-05-24:
。。。查错了
建议删掉配置重选

closed 2021-05-24 by 1715173329

asiaon123, 2021-05-24:
> 。。。查错了
> 建议删掉配置重选

问下是Aria2 的openssl 和 那个冲突了，请问下怎么解决？

asiaon123, 2021-05-24:
> 。。。查错了
> 建议删掉配置重选

其实我就是选出来编译报错的，所以找不出来的话不是到哪个不选v😂