immortalwrt/immortalwrt #398: Not Foundbin file

midorima5791, 2021-06-07:
[image]
 where's the bin file after finished build I can't find the bin file ?

1715173329, 2021-06-07:
I can only assume that you selected too many packages so OpenWrt cannot generate a image within the flash size.

midorima5791, 2021-06-07:
> I can only assume that you selected too many packages so OpenWrt cannot generate a image within the flash size.

im using newifi D2 router so it means limited package can I install?

1715173329, 2021-06-07:
> im using newifi D2 router so it means limited package can I install?

Yes, try to deselect some packages, and build again.

midorima5791, 2021-06-07:
> > im using newifi D2 router so it means limited package can I install?
> 
> Yes, try to deselect some packages, and build again.

ok thanks for the tips

closed 2021-06-07 by 1715173329