immortalwrt/immortalwrt #404: [18.06-k5.4] Newifi-d2 build failed

EnnawYang, 2021-06-14:
logs here:  
```
make[3]: Entering directory '/home/xw/openwrt/openwrt/package/kernel/mt-drivers/mt7612e'
rm -f /home/xw/openwrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76x2/.built
touch /home/xw/openwrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76x2/.built_check
make -C "/home/xw/openwrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.124" V=1 CROSS_COMPILE="mipsel-openwrt-linux-musl-" ARCH="mips" M="/home/xw/openwrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76x2/mt76x2_ap" CONFIG_RT_FIRST_CARD=7603 CONFIG_RT_SECOND_IF_MT7612E=y CONFIG_RT_SECOND_CARD=7612 CONFIG_AP_LLTD=y CONFIG_AP_LED=y CONFIG_AP_WDS=y CONFIG_AP_MBSS=y CONFIG_AP_MBSS_NEW_MBSSID_MODE=y CONFIG_AP_APCLI=y CONFIG_AP_80211AC_VHT=y  modules
make[4]: Entering directory '/home/xw/openwrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.124'
test -e include/generated/autoconf.h -a -e include/config/auto.conf || (		\
echo >&2;							\
echo >&2 "  ERROR: Kernel configuration is invalid.";		\
echo >&2 "         include/generated/autoconf.h or include/config/auto.conf are missing.";\
echo >&2 "         Run 'make oldconfig && make prepare' on kernel src to fix it.";	\
echo >&2 ;							\
/bin/false)
make -f ./scripts/Makefile.build obj=/home/xw/openwrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76x2/mt76x2_ap \
single-build= \
need-builtin=1 need-modorder=1
  mipsel-openwrt-linux-musl-gcc -Wp,-MD,/home/xw/openwrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76x2/mt76x2_ap/../mt76x2/ap/.ap_wds.o.d  -nostdinc -isystem /home/xw/openwrt/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/lib/gcc/mipsel-openwrt-linux-musl/8.4.0/include -I./arch/mips/include -I./arch/mips/include/generated  -I./include -I./arch/mips/include/uapi -I./arch/mips/include/generated/uapi -I./include/uapi -I./include/generated/uapi -include ./include/linux/kconfig.h -include ./include/linux/compiler_types.h -D__KERNEL__ -DVMLINUX_LOAD_ADDRESS=0xffffffff80001000 -DLINKER_LOAD_ADDRESS=0x80001000 -DDATAOFFSET=0 -Wall -Wundef -Werror=strict-prototypes -Wno-trigraphs -fno-strict-aliasing -fno-common -fshort-wchar -fno-PIE -Werror=implicit-function-declaration -Werror=implicit-int -Werror=return-type -Wno-format-security -std=gnu89 -mno-check-zero-division -mabi=32 -G 0 -mno-abicalls -fno-pic -pipe -mno-branch-likely -msoft-float -DGAS_HAS_SET_HARDFLOAT -Wa,-msoft-float -ffreestanding -fno-stack-check -march=mips32r2 -mtune=34kc -Wa,--trap -DTOOLCHAIN_SUPPORTS_VIRT -DTOOLCHAIN_SUPPORTS_XPA -DTOOLCHAIN_SUPPORTS_CRC -DTOOLCHAIN_SUPPORTS_DSP -DTOOLCHAIN_SUPPORTS_GINV -I./arch/mips/include/asm/mach-ralink -I./arch/mips/include/asm/mach-ralink/mt7621 -I./arch/mips/include/asm/mach-generic -fno-asynchronous-unwind-tables -fno-delete-null-pointer-checks -Wno-frame-address -Wno-format-truncation -Wno-format-overflow -O2 -fno-caller-saves --param=allow-store-data-races=0 -Wframe-larger-than=1024 -fstack-protector -Wimplicit-fallthrough -Wno-unused-but-set-variable -Wno-unused-const-variable -fomit-frame-pointer -fno-var-tracking-assignments -g -femit-struct-debug-baseonly -fno-var-tracking -Wdeclaration-after-statement -Wvla -Wno-pointer-sign -Wno-stringop-truncation -Wno-array-bounds -Wno-stringop-overflow -Wno-restrict -Wno-maybe-uninitialized -fno-strict-overflow -fno-merge-all-constants -fmerge-constants -fno-stack-check -fconserve-stack -Werror=date-time -Werror=incompatible-pointer-types -Werror=designated-init -fmacro-prefix-map=./= -Wno-packed-not-aligned -Wall -Wstrict-prototypes -Wno-trigraphs -Wno-unused -Wno-declaration-after-statement -I/home/xw/openwrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76x2/mt76x2_ap/../mt76x2/include -DAGGREGATION_SUPPORT -DPIGGYBACK_SUPPORT -DWMM_SUPPORT -DLINUX -DCONFIG_AP_SUPPORT -DUAPSD_SUPPORT -DDOT11_N_SUPPORT -DDOT1X_SUPPORT -DAP_SCAN_SUPPORT -DSCAN_SUPPORT -DSTATS_COUNT_SUPPORT -DENHANCED_STAT_DISPLAY -DRSSI_FEEDBACK -DRESOURCE_PRE_ALLOC -DEAPOL_QUEUE_SUPPORT -DDATA_QUEUE_RESERVE -DFORCE_ANNOUNCE_CRITICAL_AMPDU -DEEPROM_COUNTRY_UNLOCK -DCONFIG_RA_NAT_NONE -DRTMP_FLASH_SUPPORT -DREAD_MAC_FROM_EEPROM -DCONFIG_RT_FIRST_IF_RF_OFFSET=0x0000 -DCONFIG_RT_FIRST_CARD=7603 -DCONFIG_RT_FIRST_CARD_EEPROM="flash" -DCONFIG_RT_SECOND_IF_RF_OFFSET=0x8000 -DCONFIG_RT_SECOND_CARD_EEPROM="flash" -DCONFIG_RT_SECOND_CARD=7612 -DRT65xx -DMT76x2 -DMT7612 -DRTMP_PCI_SUPPORT -DRTMP_MAC_PCI -DRLT_MAC -DRLT_BBP -DMT_RF -DRTMP_RF_RW_SUPPORT -DA_BAND_SUPPORT -DRX_DMA_SCATTER -DCONFIG_ANDES_SUPPORT -DNEW_RATE_ADAPT_SUPPORT -DFIFO_EXT_SUPPORT -DDYNAMIC_VGA_SUPPORT -DMCS_LUT_SUPPORT -DPEER_DELBA_TX_ADAPT -DDROP_MASK_SUPPORT -DMULTI_CLIENT_SUPPORT -DRTMP_TEMPERATURE_TX_ALC -DDBG -DUAPSD_DEBUG -DDOT11_VHT_AC -DLED_CONTROL_SUPPORT -DLLTD_SUPPORT -DWDS_SUPPORT -DMBSS_SUPPORT -DNEW_MBSSID_MODE -DENHANCE_NEW_MBSSID_MODE -DAPCLI_SUPPORT -DMAT_SUPPORT -DTRAFFIC_BASED_TXOP -DAPCLI_AUTO_CONNECT_SUPPORT -DMAC_APCLI_SUPPORT  -DMODULE -mno-long-calls  -DKBUILD_BASENAME='"ap_wds"' -DKBUILD_MODNAME='"mt76x2_ap"' -c -o /home/xw/openwrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76x2/mt76x2_ap/../mt76x2/ap/ap_wds.o /home/xw/openwrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76x2/mt76x2_ap/../mt76x2/ap/ap_wds.c
In file included from ./include/linux/bitmap.h:9,
                 from ./include/linux/cpumask.h:12,
                 from ./arch/mips/include/asm/processor.h:15,
                 from ./arch/mips/include/asm/thread_info.h:16,
                 from ./include/linux/thread_info.h:39,
                 from ./include/asm-generic/preempt.h:5,
                 from ./arch/mips/include/generated/asm/preempt.h:1,
                 from ./include/linux/preempt.h:78,
                 from ./include/linux/spinlock.h:51,
                 from ./include/linux/seqlock.h:36,
                 from ./include/linux/time.h:6,
                 from ./include/linux/stat.h:19,
                 from ./include/linux/module.h:10,
                 from /home/xw/openwrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76x2/mt76x2_ap/../mt76x2/include/os/rt_linux.h:18,
                 from /home/xw/openwrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76x2/mt76x2_ap/../mt76x2/include/rtmp_os.h:46,
                 from /home/xw/openwrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76x2/mt76x2_ap/../mt76x2/include/rtmp_comm.h:81,
                 from /home/xw/openwrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76x2/mt76x2_ap/../mt76x2/include/rt_config.h:39,
                 from /home/xw/openwrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76x2/mt76x2_ap/../mt76x2/ap/ap_wds.c:32:
In function 'memmove',
    inlined from 'WdsPeerBeaconProc' at /home/xw/openwrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76x2/mt76x2_ap/../mt76x2/ap/ap_wds.c:886:4:
./include/linux/string.h:414:4: error: call to '__read_overflow2' declared with attribute error: detected read beyond size of object passed as 2nd parameter
    __read_overflow2();
    ^~~~~~~~~~~~~~~~~~
make[5]: *** [scripts/Makefile.build:262: /home/xw/openwrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76x2/mt76x2_ap/../mt76x2/ap/ap_wds.o] Error 1
make[4]: *** [Makefile:1734: /home/xw/openwrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76x2/mt76x2_ap] Error 2
```

AmadeusGhost, 2021-06-14:
试试这样改能不能正常工作吧（乱改的）：
```
--- a/package/kernel/mt-drivers/mt7612e/src/mt76x2/ap/ap_wds.c
+++ b/package/kernel/mt-drivers/mt7612e/src/mt76x2/ap/ap_wds.c
@@ -275,11 +275,11 @@ MAC_TABLE_ENTRY *MacTableInsertWDSEntry(
 		if ((wdev->PhyMode == MODE_VHT) &&
 			(pAd->CommonCfg.Channel > 14))
 		{
-			VHT_CAP_IE vht_cap;
+			VHT_CAP_IE *vht_cap = NULL;
 			VHT_CAP_INFO *vht_cap_info;
-			build_vht_cap_ie(pAd, (UCHAR *)&vht_cap);
+			build_vht_cap_ie(pAd, (UCHAR *)vht_cap);
 
-			vht_cap_info = (VHT_CAP_INFO *)&vht_cap;
+			vht_cap_info = (VHT_CAP_INFO *)vht_cap;
 
 			
 			pEntry->MaxHTPhyMode.field.MODE = MODE_VHT;
@@ -287,25 +287,25 @@ MAC_TABLE_ENTRY *MacTableInsertWDSEntry(
 				pEntry->MaxHTPhyMode.field.BW = BW_80;
 
 			/* TODO: implement get_vht_max_mcs to get peer max MCS */
-			if (vht_cap.mcs_set.rx_mcs_map.mcs_ss1 == VHT_MCS_CAP_9) {
+			if (vht_cap->mcs_set.rx_mcs_map.mcs_ss1 == VHT_MCS_CAP_9) {
 				if ((pEntry->MaxHTPhyMode.field.BW == BW_20))
 					pEntry->MaxHTPhyMode.field.MCS = 8;
 				else
 					pEntry->MaxHTPhyMode.field.MCS = 9;
-			} else if (vht_cap.mcs_set.rx_mcs_map.mcs_ss1 == VHT_MCS_CAP_8) {
+			} else if (vht_cap->mcs_set.rx_mcs_map.mcs_ss1 == VHT_MCS_CAP_8) {
 				pEntry->MaxHTPhyMode.field.MCS = 8;
-			} else if (vht_cap.mcs_set.rx_mcs_map.mcs_ss1 == VHT_MCS_CAP_7) {
+			} else if (vht_cap->mcs_set.rx_mcs_map.mcs_ss1 == VHT_MCS_CAP_7) {
 				pEntry->MaxHTPhyMode.field.MCS = 7;
 			}
 			
-			if (vht_cap.mcs_set.rx_mcs_map.mcs_ss2 == VHT_MCS_CAP_9) {
+			if (vht_cap->mcs_set.rx_mcs_map.mcs_ss2 == VHT_MCS_CAP_9) {
 				if ((pEntry->MaxHTPhyMode.field.BW == BW_20))
 					pEntry->MaxHTPhyMode.field.MCS = ((1 << 4) | 8);
 				else
 					pEntry->MaxHTPhyMode.field.MCS = ((1 << 4) | 9);
-			} else if (vht_cap.mcs_set.rx_mcs_map.mcs_ss2 == VHT_MCS_CAP_8) {
+			} else if (vht_cap->mcs_set.rx_mcs_map.mcs_ss2 == VHT_MCS_CAP_8) {
 				pEntry->MaxHTPhyMode.field.MCS = ((1 << 4) | 8);
-			} else if (vht_cap.mcs_set.rx_mcs_map.mcs_ss2 == VHT_MCS_CAP_7) {
+			} else if (vht_cap->mcs_set.rx_mcs_map.mcs_ss2 == VHT_MCS_CAP_7) {
 				pEntry->MaxHTPhyMode.field.MCS = ((1 << 4) | 7);
 			}
 
@@ -333,7 +333,7 @@ MAC_TABLE_ENTRY *MacTableInsertWDSEntry(
 				if (vht_cap_info->rx_stbc)
 					CLIENT_STATUS_SET_FLAG(pEntry, fCLIENT_STATUS_VHT_RXSTBC_CAPABLE);
 			}
-			NdisMoveMemory(&pEntry->vht_cap_ie, &vht_cap, sizeof(VHT_CAP_IE));
+			NdisMoveMemory(&pEntry->vht_cap_ie, vht_cap, sizeof(VHT_CAP_IE));
 		}
 		else
 		{
@@ -707,7 +707,7 @@ VOID WdsPeerBeaconProc(
 	IN ULONG ClientRalinkIe,
 #ifdef DOT11_VHT_AC
 	IN UCHAR vht_cap_len,
-	IN VHT_CAP_IE *vht_cap,
+	IN VHT_CAP_IE vht_cap[],
 #endif /* DOT11_VHT_AC */
 	IN HT_CAPABILITY_IE *pHtCapability,
 	IN UCHAR HtCapabilityLen)
@@ -883,7 +883,7 @@ VOID WdsPeerBeaconProc(
 				if (vht_cap_info->rx_stbc)
 					CLIENT_STATUS_SET_FLAG(pEntry, fCLIENT_STATUS_VHT_RXSTBC_CAPABLE);
 			}
-			NdisMoveMemory(&pEntry->vht_cap_ie, &vht_cap, sizeof(VHT_CAP_IE));
+			NdisMoveMemory(&pEntry->vht_cap_ie, vht_cap, sizeof(VHT_CAP_IE));
 		}
 		else
 		{
```

EnnawYang, 2021-06-15:
> 试试这样改能不能正常工作吧（乱改的）：

可以了

commit 3300c8e7 (2021-06-16): mtwifi: mt7612: fix build error

Fixes: #404
(cherry picked from commit c8d26d2b915ecd1edbb53c2fbe294cee198b8883)
commit 67d20d16 (2021-06-16): mtwifi: mt7612: fix build error

Fixes: #404
(cherry picked from commit c8d26d2b915ecd1edbb53c2fbe294cee198b8883)
closed 2021-06-16 by 1715173329