immortalwrt/immortalwrt #406: openwrt-18.06分支云编译x86-64错误

281677160, 2021-06-15:
rm -rf /workdir/openwrt/tmp/stage-ntfs-3g
touch /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.ntfs-3g_installed
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/utils/ntfs-3g'
time: package/feeds/packages/ntfs-3g/compile#37.86#5.37#42.40
make[3]: Entering directory '/workdir/openwrt/package/kernel/exfat'
mkdir -p /workdir/openwrt/dl
SHELL= flock /workdir/openwrt/tmp/.exfat-5.12.3.tar.gz.flock -c '  	/workdir/openwrt/scripts/download.pl "/workdir/openwrt/dl" "exfat-5.12.3.tar.gz" "43889c73af76c466bbc904aff80354a62ecaa24c7b20e354ff735f5949907982" "" "https://codeload.github.com/namjaejeon/linux-exfat-oot/tar.gz/5.12.3?"    '
rm -f /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-exfat-oot-5.12.3/.built
touch /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-exfat-oot-5.12.3/.built_check
cat /dev/null > /workdir/openwrt/staging_dir/target-x86_64_musl/pkginfo/exfat.symvers; for subdir in .; do cat /workdir/openwrt/staging_dir/target-x86_64_musl/pkginfo/*.symvers 2>/dev/null > /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-exfat-oot-5.12.3/$subdir/Module.symvers; done
make -C /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-4.19.194 KCFLAGS="-fmacro-prefix-map=/workdir/openwrt/build_dir/target-x86_64_musl=target-x86_64_musl" HOSTCFLAGS="-O2 -I/workdir/openwrt/staging_dir/host/include -I/workdir/openwrt/staging_dir/hostpkg/include -I/workdir/openwrt/staging_dir/target-x86_64_musl/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="x86_64-openwrt-linux-musl-" ARCH="x86" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Mon Jun 14 18:32:23 2021" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/workdir/openwrt/staging_dir/host/lib" KBUILD_HOSTLDLIBS="-L/workdir/openwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KERNELRELEASE=4.19.194 M="/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-exfat-oot-5.12.3" EXTRA_CFLAGS=""  CONFIG_EXFAT_FS=m modules
make[4]: Entering directory '/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-4.19.194'
  CC [M]  /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-exfat-oot-5.12.3/balloc.o
/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-exfat-oot-5.12.3/balloc.c: In function 'exfat_trim_fs':
/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-exfat-oot-5.12.3/balloc.c:325:7: error: implicit declaration of function 'fatal_signal_pending'; did you mean 'mod_timer_pending'? [-Werror=implicit-function-declaration]
   if (fatal_signal_pending(current)) {
       ^~~~~~~~~~~~~~~~~~~~
       mod_timer_pending
cc1: some warnings being treated as errors
make[5]: *** [scripts/Makefile.build:304: /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-exfat-oot-5.12.3/balloc.o] Error 1
make[4]: *** [Makefile:1545: _module_/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-exfat-oot-5.12.3] Error 2
make[3]: *** [Makefile:50: /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-exfat-oot-5.12.3/.built] Error 2
make[4]: Leaving directory '/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-4.19.194'
make[3]: Leaving directory '/workdir/openwrt/package/kernel/exfat'
time: package/kernel/exfat/compile#1.25#0.37#1.47
    ERROR: package/kernel/exfat failed to build.
make[2]: *** [package/Makefile:115: package/kernel/exfat/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:109: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2
make[1]: Leaving directory '/workdir/openwrt'

EnnawYang, 2021-06-15:
不是已经修过了？

281677160, 2021-06-15:
> 不是已经修过了？

早之前不是这个错误，就是修复了上一个错误，就立马编译了，现在又出错

1715173329, 2021-06-15:
So, said above, this was fixed already.

closed 2021-06-15 by 1715173329

281677160, 2021-06-15:
好吧，感谢了，我今天第一个发的错误是ERROR: tools/zstd failed to build，已经修复，修复了ERROR: tools/zstd failed to build，之后编译出现ERROR: package/kernel/exfat failed to build，我再看看我什么地方弄错了吧，感谢回复！