immortalwrt/immortalwrt #407: arm-trusted-firmware-rk3328: download failed

EdiTurn, 2021-06-16:
`make[1]: Entering directory '/home/editurn/immortalwrt'
make[2]: Entering directory '/home/editurn/immortalwrt/package/boot/arm-trusted-firmware-rk3328'
Makefile:55: WARNING: skipping arm-trusted-firmware-rk3328 -- package has no install section
mkdir -p /home/editurn/immortalwrt/dl
/home/editurn/immortalwrt/scripts/download.pl "/home/editurn/immortalwrt/dl" "arm-trusted-firmware-rk3328-2020-10-30-0bb1c512.tar.xz" "6dffe95b83f37a280f98c105fce529f45e39ce333b24709a9645aac1352457ce" "" || (  /home/editurn/immortalwrt/scripts/dl_github_archive.py --dl-dir="/home/editurn/immortalwrt/dl" --url="https://github.com/rockchip-linux/rkbin.git" --version="0bb1c512492386a72a3a0b5a0e18e49c636577b9" --subdir="arm-trusted-firmware-rk3328-2020-10-30-0bb1c512" --source="arm-trusted-firmware-rk3328-2020-10-30-0bb1c512.tar.xz" --hash="6dffe95b83f37a280f98c105fce529f45e39ce333b24709a9645aac1352457ce" || (   echo "Checking out files from the git repository..."; mkdir -p /home/editurn/immortalwrt/tmp/dl && cd /home/editurn/immortalwrt/tmp/dl && rm -rf arm-trusted-firmware-rk3328-2020-10-30-0bb1c512 && [ \! -d arm-trusted-firmware-rk3328-2020-10-30-0bb1c512 ] && git clone  https://github.com/rockchip-linux/rkbin.git arm-trusted-firmware-rk3328-2020-10-30-0bb1c512 && (cd arm-trusted-firmware-rk3328-2020-10-30-0bb1c512 && git checkout 0bb1c512492386a72a3a0b5a0e18e49c636577b9 && git submodule update --init --recursive) && echo "Packing checkout..." && export TAR_TIMESTAMP=`cd arm-trusted-firmware-rk3328-2020-10-30-0bb1c512 && git log -1 --format='@%ct'` && rm -rf arm-trusted-firmware-rk3328-2020-10-30-0bb1c512/.git &&  tar --numeric-owner --owner=0 --group=0 --mode=a-s --sort=name ${TAR_TIMESTAMP:+--mtime="$TAR_TIMESTAMP"} -c arm-trusted-firmware-rk3328-2020-10-30-0bb1c512 |  xz -zc -7e > /home/editurn/immortalwrt/tmp/dl/arm-trusted-firmware-rk3328-2020-10-30-0bb1c512.tar.xz && mv /home/editurn/immortalwrt/tmp/dl/arm-trusted-firmware-rk3328-2020-10-30-0bb1c512.tar.xz /home/editurn/immortalwrt/dl/ && rm -rf arm-trusted-firmware-rk3328-2020-10-30-0bb1c512; );  )
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://mirror01.download.immortalwrt.eu.org/arm-trusted-firmware-rk3328-2020-10-30-0bb1c512.tar.xz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     8    0     0    0     0      0      0 --:--:--  0:00:02 --:--:--     0
curl: (22) The requested URL returned error: 404
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://mirror02.download.immortalwrt.eu.org/arm-trusted-firmware-rk3328-2020-10-30-0bb1c512.tar.xz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100    52  100    52    0     0     55      0 --:--:-- --:--:-- --:--:--    55
Hash of the downloaded file does not match (file: f23aad655a6bba7b8a5033a5a07cafa85c440b2bca72fc9fbf3365cc80ff16c8, requested: 6dffe95b83f37a280f98c105fce529f45e39ce333b24709a9645aac1352457ce) - deleting download.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://openwrt.cc/dl/immortalwrt/master/arm-trusted-firmware-rk3328-2020-10-30-0bb1c512.tar.xz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   138  100   138    0     0    152      0 --:--:-- --:--:-- --:--:--   152
  0     0    0     0    0     0      0      0 --:--:--  0:00:01 --:--:--     0
curl: (22) The requested URL returned error: 404
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://sources.cdn.openwrt.org/arm-trusted-firmware-rk3328-2020-10-30-0bb1c512.tar.xz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0   169    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://sources.openwrt.org/arm-trusted-firmware-rk3328-2020-10-30-0bb1c512.tar.xz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0   169    0     0    0     0      0      0 --:--:--  0:00:01 --:--:--     0
curl: (22) The requested URL returned error: 404
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://mirror2.openwrt.org/sources/arm-trusted-firmware-rk3328-2020-10-30-0bb1c512.tar.xz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0   169    0     0    0     0      0      0 --:--:--  0:00:01 --:--:--     0
curl: (22) The requested URL returned error: 404
Download failed.
No more mirrors to try - giving up.
arm-trusted-firmware-rk3328-2020-10-30-0bb1c512.tar.xz: Download from https://github.com/rockchip-linux/rkbin.git failed
arm-trusted-firmware-rk3328-2020-10-30-0bb1c512.tar.xz: Wrong hash (probably caused by .gitattributes), expecting 6dffe95b83f37a280f98c105fce529f45e39ce333b24709a9645aac1352457ce, got bb91f311ff7a26524059b7b49516140bfdfa892826b515518fa1fa09b5eb0db5
Checking out files from the git repository...
Cloning into 'arm-trusted-firmware-rk3328-2020-10-30-0bb1c512'...
remote: Enumerating objects: 8608, done.
remote: Counting objects: 100% (3958/3958), done.
remote: Compressing objects: 100% (1137/1137), done.
remote: Total 8608 (delta 2840), reused 3914 (delta 2817), pack-reused 4650
Receiving objects: 100% (8608/8608), 78.17 MiB | 10.17 MiB/s, done.
Resolving deltas: 100% (6490/6490), done.
fatal: reference is not a tree: 0bb1c512492386a72a3a0b5a0e18e49c636577b9
make[2]: *** [Makefile:56: /home/editurn/immortalwrt/dl/arm-trusted-firmware-rk3328-2020-10-30-0bb1c512.tar.xz] Error 128
make[2]: Leaving directory '/home/editurn/immortalwrt/package/boot/arm-trusted-firmware-rk3328'
time: package/boot/arm-trusted-firmware-rk3328/download#13.12#2.19#31.09
    ERROR: package/boot/arm-trusted-firmware-rk3328 failed to build.
make[1]: *** [package/Makefile:116: package/boot/arm-trusted-firmware-rk3328/download] Error 1
make[1]: Leaving directory '/home/editurn/immortalwrt'
make: *** [/home/editurn/immortalwrt/include/toplevel.mk:230：package/arm-trusted-firmware-rk3328/download] 错误 2`

AmadeusGhost, 2021-06-16:
强烈建议去把 rockchip 打一顿，不过应该是能 checkout 出来的

1715173329, 2021-06-16:
I just wonder why my mirrors don't work...

EdiTurn, 2021-06-16:
手动去 checkout 也是提示 `fatal: 引用不是一个树：0bb1c512492386a72a3a0b5a0e18e49c636577b9`，这个 commit 没有所属分支了。

1715173329, 2021-06-16:
> I just wonder why my mirrors don't work...

Fine, this is target specific...

closed 2021-06-16 by 1715173329

AmadeusGhost, 2021-06-16:
> 手动去 checkout 也是提示 `fatal: 引用不是一个树：0bb1c512492386a72a3a0b5a0e18e49c636577b9`，这个 commit 没有所属分支了。

See: https://github.com/immortalwrt/immortalwrt/commit/1c3cf01b93bdff3e45f34c0018bd6cbf14ef7589 , just git pull.