immortalwrt/immortalwrt #408: 18.06 x86 不能启动

noreph, 2021-06-16:
在物理机和虚拟机上均有复现，开机到 grub loading 就重启了

[image]

1715173329, 2021-06-16:
Same issue can be found here (based on immortalwrt x86 source):
coolsnowwolf/lede#6977
coolsnowwolf/lede#6979

1715173329, 2021-06-17:
Please try with this commit: c01acf6c4d01efba2000ca7fe4627eec763a63cc
If you're at openwrt-18.06-k5.4 branch, run the following command:
```bash
curl -fsSL https://github.com/immortalwrt/immortalwrt/commit/c01acf6c4d01efba2000ca7fe4627eec763a63cc.patch | git apply
```

1715173329, 2021-06-17:
Please try with latest git HEAD.

noreph, 2021-06-18:
最新代码编译的在虚拟测试已经正常开机了

1715173329, 2021-06-18:
However sysupgrade function is broken...
I have no idea how to address it.

1715173329, 2021-06-18:
> However sysupgrade function is broken...
> I have no idea how to address it.

I saw the same issue with master branch... Does anyone have any idea?

1715173329, 2021-06-18:
However it boots, so I guess I can close this issue now.

closed 2021-06-18 by 1715173329