immortalwrt/immortalwrt #410: wpad compile error

DeborahCox, 2021-06-23:
` /home/runner/lede/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/lib/gcc/aarch64-openwrt-linux-musl/8.4.0/../../../../aarch64-openwrt-linux-musl/bin/ld: /home/runner/lede/tmp/cc7AekQI.ltrans25.ltrans.o: in function `ap_sta_add.part.13':

<artificial>:(.text.ap_sta_add.part.13+0x18c): undefined reference to `sta_track_claim_taxonomy_info'
/home/runner/lede/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/lib/gcc/aarch64-openwrt-linux-musl/8.4.0/../../../../aarch64-openwrt-linux-musl/bin/ld: <artificial>:(.text.ap_sta_add.part.13+0x19c): undefined reference to `sta_track_claim_taxonomy_info'
collect2: error: ld returned 1 exit status
make[3]: *** [Makefile:682: /home/runner/lede/build_dir/target-aarch64_generic_musl/hostapd-wpad-full-internal/hostapd-2021-05-22-b102f19b/.built] Error 1
make[3]: Leaving directory '/home/runner/lede/package/network/services/hostapd'
time: package/network/services/hostapd/wpad-full-internal/compile#35.77#1.50#36.45
    ERROR: package/network/services/hostapd failed to build (build variant: wpad-full-internal).
make[2]: *** [package/Makefile:116: package/network/services/hostapd/compile] Error 1
make[2]: Leaving directory '/home/runner/lede'
make[1]: Leaving directory '/home/runner/lede'
make[1]: *** [package/Makefile:110: /home/runner/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make: *** [/home/runner/lede/include/toplevel.mk:230: world] Error 2
Error: Process completed with exit code 2.`

1715173329, 2021-06-23:
- No branch
- No device
- use wpad-openssl instead

closed 2021-06-23 by 1715173329

DeborahCox, 2021-06-23:
Sorry, It was on [18.06 k5.4] R2S.
so wpad-wolfssl is not available I assume?

1715173329, 2021-06-23:
The package you used is `wpad-full-internal`.
Remember to compile mac80211 first.