immortalwrt/immortalwrt #416: 新路由3编译21.02，没有交换机插件，而且cpu使用率较高

Thanf-code, 2021-07-01:
[image]
[image]

1715173329, 2021-07-01:
- Use Google to find what is called DSA.
- Use luci-app-turboacc to enable hardware offloading.

closed 2021-07-01 by 1715173329

Thanf-code, 2021-07-01:
> * Use Google to find what DSA is.
> * Use luci-app-turboacc to enable hardware offloading.

Directory System Agent？ 
我想用应用过滤，但应用过滤与加速模块冲突

Gzxhwq, 2021-07-01:
https://forum.openwrt.org/t/mini-tutorial-for-dsa-network-config/96998

1715173329, 2021-07-01:
> Directory System Agent？

https://www.kernel.org/doc/Documentation/networking/dsa/dsa.txt

> 我想用应用过滤，但应用过滤与加速模块冲突

So it's up to you, appfilter or offloading.

Thanf-code, 2021-07-01:
> > Directory System Agent？
> 
> https://www.kernel.org/doc/Documentation/networking/dsa/dsa.txt
> 
> > 我想用应用过滤，但应用过滤与加速模块冲突
> 
> So it's up to you, appfilter or offloading.

1.可不可以在ssh界面重新选上交换机插件，该如何操作？
2.lede源码不用加速模块，cpu使用率也还可以，可以借鉴一下？

1715173329, 2021-07-01:
Use 18.06 branch.