immortalwrt/immortalwrt #418: 请教一个小白问题 openclash在哪个目录里啊

zhangj3, 2021-07-05:
大佬好！
我知道immortalwrt里集成了openclash，是在哪个目录里啊 我想看下源码有没有跟作者的github同步 https://github.com/vernesong/OpenClash/tree/master

谢谢

zhangj3, 2021-07-05:
immortalwrt/package/里点了一些文件夹 没找到，所以只能问问大佬了

AmadeusGhost, 2021-07-05:
在 feeds/luci 下：https://github.com/immortalwrt/luci

closed 2021-07-05 by AmadeusGhost

zhangj3, 2021-07-05:
> 在 feeds/luci 下：https://github.com/immortalwrt/luci

谢谢大佬，原来不在一个地方。

是没有同步，有什么办法可以手动同步吗？
最近openclash总是出现“{"message":"Resource not found"}”问题 https://github.com/vernesong/OpenClash/issues/1439 ，启动慢，作者好像修复了

[image]

[image]

我有什么办法手动同步吗？