immortalwrt/immortalwrt #420: What is the problem? I followed the steps to build image but always has the same error

qqsir-dev, 2021-07-20:
Installing base-files (137-r7526-5a6225d58) to root...
Installing libgcc1 (8.4.0-3) to root...
Installing libc (1.1.24-3) to root...
Installing libubox20191228 (2020-07-11-f4e9bf73-2) to root...
Installing libuci20130104 (2019-05-17-f199b961-5) to root...
Installing libnl-tiny1 (2020-08-05-c291088f-2) to root...
Installing libubus20191227 (2020-02-05-171469e3-2) to root...
Installing libjson-c5 (0.15-2) to root...
Installing libblobmsg-json (2020-07-11-f4e9bf73-2) to root...
Installing ubusd (2020-02-05-171469e3-2) to root...
Installing ubus (2020-02-05-171469e3-2) to root...
Installing jshn (2020-07-11-f4e9bf73-2) to root...
Installing netifd (2020-06-06-51e9fb81-1) to root...
Installing jsonfilter (2018-02-04-c7e938d6-1) to root...
Installing usign (2020-05-23-f1f65026-1) to root...
Installing openwrt-keyring (2018-05-18-103a32e9-1) to root...
Installing ubox (2019-12-31-0e34af14-4) to root...
Installing fstools (2019-03-28-ff1ded63-6) to root...
Installing fwtool (1) to root...
Installing grub2 (2.06-1) to root...
Installing grub2-efi (2.06-1) to root...
Installing grub2-bios-setup (2.06-1) to root...
Installing kmod-button-hotplug (4.19.197-3) to root...
Installing kernel (4.19.197-1-db9a8a3d4463322055e60950366259a7) to root...
Installing kmod-input-core (4.19.197-1) to root...
Installing kmod-cryptodev (4.19.197+1.11-x86-1) to root...
Installing kmod-crypto-hash (4.19.197-1) to root...
Installing kmod-crypto-null (4.19.197-1) to root...
Installing kmod-crypto-aead (4.19.197-1) to root...
Installing kmod-crypto-pcompress (4.19.197-1) to root...
Installing kmod-crypto-manager (4.19.197-1) to root...
Installing kmod-crypto-authenc (4.19.197-1) to root...
Installing kmod-fs-exfat (4.19.197+5.11.1-1) to root...
Installing kmod-nls-base (4.19.197-1) to root...
Installing kmod-ipt-fullconenat (4.19.197+2019-10-21-0cf3b48f-1) to root...
Installing kmod-nf-ipt (4.19.197-1) to root...
Installing kmod-nf-conntrack (4.19.197-1) to root...
Installing kmod-nf-nat (4.19.197-1) to root...
Package kernel (4.19.197-1-db9a8a3d4463322055e60950366259a7) installed in root is up to date.
Installing kmod-scsi-core (4.19.197-1) to root...
Installing kmod-crypto-acompress (4.19.197-1) to root...
Package kmod-crypto-aead (4.19.197-1) installed in root is up to date.
Installing kmod-crypto-arc4 (4.19.197-1) to root...
Package kmod-crypto-authenc (4.19.197-1) installed in root is up to date.
Installing kmod-crypto-cbc (4.19.197-1) to root...
Installing kmod-crypto-crc32 (4.19.197-1) to root...
Installing kmod-crypto-crc32c (4.19.197-1) to root...
Installing kmod-crypto-deflate (4.19.197-1) to root...
Installing kmod-lib-zlib-inflate (4.19.197-1) to root...
Installing kmod-lib-zlib-deflate (4.19.197-1) to root...
Installing kmod-crypto-des (4.19.197-1) to root...
Installing kmod-crypto-ecb (4.19.197-1) to root...
Installing kmod-crypto-echainiv (4.19.197-1) to root...
Package kmod-crypto-hash (4.19.197-1) installed in root is up to date.
Installing kmod-crypto-hmac (4.19.197-1) to root...
Package kmod-crypto-manager (4.19.197-1) installed in root is up to date.
Installing kmod-crypto-md5 (4.19.197-1) to root...
Package kmod-crypto-null (4.19.197-1) installed in root is up to date.
Package kmod-crypto-pcompress (4.19.197-1) installed in root is up to date.
Installing kmod-crypto-sha1 (4.19.197-1) to root...
Installing kmod-crypto-user (4.19.197-1) to root...
Installing kmod-fs-btrfs (4.19.197-1) to root...
Installing kmod-lib-crc32c (4.19.197-1) to root...
Installing kmod-lib-lzo (4.19.197-1) to root...
Installing kmod-lib-raid6 (4.19.197-1) to root...
Installing kmod-lib-xor (4.19.197-1) to root...
Installing kmod-lib-zstd (4.19.197-1) to root...
Installing kmod-fs-ext4 (4.19.197-1) to root...
Installing kmod-lib-crc16 (4.19.197-1) to root...
Installing kmod-fs-f2fs (4.19.197-1) to root...
[... 1488 lines cut ...]
Number of inodes 7500
Number of files 6361
Number of fragments 208
Number of symbolic links  577
Number of device nodes 1
Number of fifo nodes 0
Number of socket nodes 0
Number of directories 561
Number of ids (unique uids + gids) 1
Number of uids 1
        root (0)
Number of gids 1
        root (0)
make[5]: Leaving directory '/home/qqsir/immortalwrt/target/linux/x86/image'
make[4]: *** [Makefile:29: install] Error 2
make[4]: Leaving directory '/home/qqsir/immortalwrt/target/linux/x86'
make[3]: *** [Makefile:13: install] Error 2
make[3]: Leaving directory '/home/qqsir/immortalwrt/target/linux'
time: target/linux/install#157.95#6.09#77.97
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:25: target/linux/install] Error 1
make[2]: Leaving directory '/home/qqsir/immortalwrt'
make[1]: *** [target/Makefile:19: /home/qqsir/immortalwrt/staging_dir/target-x86_64_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/qqsir/immortalwrt'
make: *** [/home/qqsir/immortalwrt/include/toplevel.mk:230: world] Error 2

1715173329, 2021-07-20:
> error: ext4_allocate_best_fit_partial: failed to allocate 3 blocks, out of space?

Increase ROOTFS size.

closed 2021-07-20 by 1715173329

qqsir-dev, 2021-07-20:
> > error: ext4_allocate_best_fit_partial: failed to allocate 3 blocks, out of space?
> 
> Increase ROOTFS size.

How to do it?

1715173329, 2021-07-20:
Ask Google.

qqsir-dev, 2021-07-20:
技术不行，玩不了，换lede吧 ，至少能编译成功

EnnawYang, 2021-07-20:
[image]

AmadeusGhost, 2021-07-20:
> 技术不行，玩不了，换lede吧 ，至少能编译成功

你自己东西选太多了怪这里不行？

qqsir-dev, 2021-07-21:
> > 技术不行，玩不了，换lede吧 ，至少能编译成功
> 
> 你自己东西选太多了怪这里不行？

你是不中文不行么？我说我技术不行，我哪句说怪这里不行了？

EnnawYang, 2021-07-21:
> > 技术不行，玩不了，换lede吧 ，至少能编译成功
> > 
> 我哪句说怪这里不行了？
> 
你这整句话都挺有争议的

jh0072110, 2021-07-21:
我也觉得整句话都阴阳怪气的…可能是我中文造诣还不够精深吧…

qqsir-dev, 2021-07-21:
好，那我重说，我自己技术太菜！打字还是会误解意思，其实没有阴阳怪气，确实技术水平有限。专业人士可以忽略我这种人，不用较真，因为和我差不多的人太多了，太较真自己容易血压飙升：）
我上面的意思是Lede的编译预处理方面可能就是为了大多数和我差不多的人准备的，避免我们由于技术不行导致的瞎嚷嚷，然后步骤教程写的细致一些，才让我这种菜人成功率也高一些，也编译失败过很多次，没有哪个好和不好的意思。
不知道这么说行不行，还会不会有什么误解