immortalwrt/immortalwrt #421: openwrt-18.06-k5.4 分支编译失败，681-NET-add-mtd-mac-address-support-to-of_get_mac_addres.patch Patch failed!

PigodChen, 2021-07-22:
patch: **** malformed patch at line 110: +                      continue;

Patch failed!  Please fix /opt/immortalwrt-k54/target/linux/generic/pending-5.4/681-NET-add-mtd-mac-address-support-to-of_get_mac_addres.patch!

pikachullk, 2021-07-22:
X86同样的错误~我换了5.10也是同样的错误~

EnnawYang, 2021-07-22:
target: x86_64
branch: 18.06-k5.4
environment: Actions (ubuntu 20.04)，PC (ubuntu 20.04)
logs: 
```
Applying /home/runner/work/build-openwrt-and-repack/build-openwrt-and-repack/openwrt/target/linux/generic/pending-5.4/670-ipv6-allow-rejecting-with-source-address-failed-policy.patch using plaintext: 
patching file include/net/netns/ipv6.h
patching file include/uapi/linux/fib_rules.h
patching file include/uapi/linux/rtnetlink.h
patching file net/ipv4/fib_semantics.c
patching file net/ipv4/fib_trie.c
patching file net/ipv4/ipmr.c
patching file net/ipv6/fib6_rules.c
patching file net/ipv6/ip6mr.c
patching file net/ipv6/route.c

Applying /home/runner/work/build-openwrt-and-repack/build-openwrt-and-repack/openwrt/target/linux/generic/pending-5.4/671-net-provide-defines-for-_POLICY_FAILED-until-all-cod.patch using plaintext: 
patching file include/uapi/linux/fib_rules.h
patching file include/uapi/linux/icmpv6.h
patching file include/uapi/linux/rtnetlink.h

Applying /home/runner/work/build-openwrt-and-repack/build-openwrt-and-repack/openwrt/target/linux/generic/pending-5.4/680-NET-skip-GRO-for-foreign-MAC-addresses.patch using plaintext: 
patching file include/linux/netdevice.h
patching file include/linux/skbuff.h
Hunk #1 succeeded at 836 (offset 12 lines).
patching file net/core/dev.c
Hunk #1 succeeded at 5501 (offset 12 lines).
Hunk #2 succeeded at 7299 (offset 12 lines).
Hunk #3 succeeded at 7391 (offset 12 lines).
Hunk #4 succeeded at 7485 (offset 12 lines).
Hunk #5 succeeded at 8216 (offset 12 lines).
patching file net/ethernet/eth.c

Applying /home/runner/work/build-openwrt-and-repack/build-openwrt-and-repack/openwrt/target/linux/generic/pending-5.4/681-NET-add-mtd-mac-address-support-to-of_get_mac_addres.patch using plaintext: 
patching file drivers/of/of_net.c
Hunk #2 succeeded at 40 (offset -6 lines).
patch: **** malformed patch at line 110: +			continue;

Patch failed!  Please fix /home/runner/work/build-openwrt-and-repack/build-openwrt-and-repack/openwrt/target/linux/generic/pending-5.4/681-NET-add-mtd-mac-address-support-to-of_get_mac_addres.patch!
make[3]: *** [Makefile:122: /home/runner/work/build-openwrt-and-repack/build-openwrt-and-repack/openwrt/build_dir/toolchain-x86_64_gcc-8.4.0_musl/linux-5.4.132/.prepared] Error 1
```

AmadeusGhost, 2021-07-22:
emmm，明天起来再修，本来想先跳过这些 commit （好像还有 bug）结果忘了全同步了🌚

commit 8553655c (2021-07-23): kernel/pending: refresh some patches

Fix: #421
closed 2021-07-23 by 1715173329