immortalwrt/immortalwrt #437: R4S编译报错

JSZMonkey, 2021-08-20:
make[7]: *** [scripts/Makefile.build:262: kernel/sched/fair.o] Error 1
make[6]: *** [scripts/Makefile.build:497: kernel/sched] Error 2
make[5]: *** [Makefile:1734: kernel] Error 2
make[4]: *** [Makefile:25: /home/runner/work/R4S-OpenWrt/R4S-OpenWrt/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.137/.image] Error 2
make[3]: *** [Makefile:11: install] Error 2
make[5]: Leaving directory '/home/runner/work/R4S-OpenWrt/R4S-OpenWrt/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.137'
make[4]: Leaving directory '/home/runner/work/R4S-OpenWrt/R4S-OpenWrt/openwrt/target/linux/rockchip'
make[2]: *** [target/Makefile:25: target/linux/install] Error 1
make[3]: Leaving directory '/home/runner/work/R4S-OpenWrt/R4S-OpenWrt/openwrt/target/linux'
time: target/linux/install#1.69#0.56#2.21
    ERROR: target/linux failed to build.
make[2]: Leaving directory '/home/runner/work/R4S-OpenWrt/R4S-OpenWrt/openwrt'
make[1]: *** [target/Makefile:19: /home/runner/work/R4S-OpenWrt/R4S-OpenWrt/openwrt/staging_dir/target-aarch64_generic_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/runner/work/R4S-OpenWrt/R4S-OpenWrt/openwrt'
make: *** [/home/runner/work/R4S-OpenWrt/R4S-OpenWrt/openwrt/include/toplevel.mk:230: world] Error 2
Error: Process completed with exit code 2.

1715173329, 2021-08-20:
- 没有分支
- 日志不足

1715173329, 2021-08-20:
You're not using ImmortalWrt's source tree. Closing.

closed 2021-08-20 by 1715173329