immortalwrt/immortalwrt #438: k5.4编译出错

jh0072110, 2021-08-22:
先上报错：
tmp/.config-package.in:111902:error: recursive dependency detected!
tmp/.config-package.in:111902:  symbol PACKAGE_ethtool-full depends on PACKAGE_ethtool
tmp/.config-package.in:111889:  symbol PACKAGE_ethtool is selected by PACKAGE_ethtool-full
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"
[image]
用的是Action云编译，有点没看明白，应该是Makefile的问题（吧）？各位大佬给康康？

EnnawYang, 2021-08-22:
非源码问题，就两个原因：
1.你自己取消的
2.被github中断了

jh0072110, 2021-08-22:
> 非源码问题，就两个原因：
> 1.你自己取消的
> 2.被github中断了

1可以排除掉=。=被github中断可还行……

jh0072110, 2021-08-22:
太诡异了，我开了三四次都在一小时左右“被cannel”，容我reopen一下康康有木有大佬知道这是肿么回事……

closed 2021-08-22 by AmadeusGhost

jh0072110, 2021-08-23:
@AmadeusGhost 很尴尬，之前那个被天灵大佬spam了，这里附上我找到的actions完整logs，issue我也不reopen了（也没法reopen……），您空了看看有没有帮助吧，如果没能帮到忙就算了…毕竟没技术只能流下泪水T.T加油
[未命名2.txt](https://github.com/immortalwrt/immortalwrt/files/7030998/2.txt)
[未命名.txt](https://github.com/immortalwrt/immortalwrt/files/7030999/default.txt)

hue715, 2021-08-23:
> 很尴尬，之前那个被天灵大佬spam了，这里附上我找到的actions完整logs，issue我也不reopen了（也没法reopen……），您空了看看有没有帮助吧，如果没能帮到忙就算了…毕竟没技术只能流下泪水T.T加油

[image]
在你的action代码里增加V=s，使用单线程-j1编译，再查看完整debug信息

jh0072110, 2021-08-23:
> > 很尴尬，之前那个被天灵大佬spam了，这里附上我找到的actions完整logs，issue我也不reopen了（也没法reopen……），您空了看看有没有帮助吧，如果没能帮到忙就算了…毕竟没技术只能流下泪水T.T加油
> 
> [image]
> 在你的action代码里增加V=s，使用单线程-j1编译，再查看完整debug信息

好嘞