immortalwrt/immortalwrt #439: k5.4编译https-dns-proxy错误

jh0072110, 2021-08-23:
研究了半天，锅应该在它身上（吧）……
附actions产生的两版logs：
1⃣️
Run cd openwrt
2 thread compile
 make[1] world
 make[2] tools/compile
 make[2] package/cleanup
 make[3] -C tools/flock compile
 make[3] -C tools/xz compile
 make[3] -C tools/sed compile
 make[3] -C tools/tar compile
 make[3] -C tools/patch compile
 make[3] -C tools/pkg-config compile
 make[3] -C tools/ninja compile
 make[3] -C tools/libressl compile
 make[3] -C tools/cmake compile
 make[3] -C tools/zstd compile
 make[3] -C tools/ccache compile
 make[3] -C tools/autoconf-archive compile
 make[3] -C tools/m4 compile
 make[3] -C tools/cpio compile
 make[3] -C tools/zlib compile
 make[3] -C tools/lzma compile
 make[3] -C tools/make-ext4fs compile
 make[3] -C tools/mkimage compile
 make[3] -C tools/mtools compile
 make[3] -C tools/patch-image compile
 make[3] -C tools/squashfskit4 compile
 make[3] -C tools/sstrip compile
 make[3] -C tools/ucl compile
 make[3] -C tools/xxd compile
 make[3] -C tools/zip compile
 make[3] -C tools/expat compile
 make[3] -C tools/autoconf compile
 make[3] -C tools/firmware-utils compile
 make[3] -C tools/automake compile
 make[3] -C tools/missing-macros compile
 make[3] -C tools/dosfstools compile
 make[3] -C tools/upx compile
 make[3] -C tools/libtool compile
 make[3] -C tools/flex compile
 make[3] -C tools/e2fsprogs compile
 make[3] -C tools/fakeroot compile
 make[3] -C tools/gengetopt compile
 make[3] -C tools/mklibs compile
 make[3] -C tools/mm-macros compile
 make[3] -C tools/mtd-utils compile
 make[3] -C tools/patchelf compile
 make[3] -C tools/gmp compile
 make[3] -C tools/bison compile
 make[3] -C tools/mpfr compile
 make[3] -C tools/bc compile
 make[3] -C tools/findutils compile
 make[3] -C tools/mpc compile
 make[3] -C tools/padjffs2 compile
 make[3] -C tools/quilt compile
 make[2] toolchain/compile
 make[3] -C toolchain/binutils compile
 make[3] -C toolchain/gdb compile
 make[3] -C toolchain/fortify-headers compile
 make[3] -C toolchain/nasm compile
 make[3] -C toolchain/gcc/initial compile
 make[3] -C toolchain/kernel-headers compile
 make[3] -C toolchain/musl compile
 make[3] -C toolchain/gcc/final compile
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Terminated
Error: The operation was canceled.
2⃣️
Run cd openwrt
2 thread compile
 make[1] world
 make[2] tools/compile
 make[2] package/cleanup
 make[3] -C tools/flock compile
 make[3] -C tools/xz compile
 make[3] -C tools/sed compile
 make[3] -C tools/tar compile
 make[3] -C tools/patch compile
 make[3] -C tools/pkg-config compile
 make[3] -C tools/ninja compile
 make[3] -C tools/libressl compile
 make[3] -C tools/cmake compile
 make[3] -C tools/zstd compile
 make[3] -C tools/ccache compile
 make[3] -C tools/m4 compile
 make[3] -C tools/autoconf-archive compile
 make[3] -C tools/cpio compile
 make[3] -C tools/zlib compile
 make[3] -C tools/lzma compile
 make[3] -C tools/make-ext4fs compile
 make[3] -C tools/mkimage compile
 make[3] -C tools/mtools compile
 make[3] -C tools/patch-image compile
 make[3] -C tools/squashfskit4 compile
 make[3] -C tools/sstrip compile
 make[3] -C tools/ucl compile
 make[3] -C tools/xxd compile
 make[3] -C tools/zip compile
 make[3] -C tools/expat compile
 make[3] -C tools/autoconf compile
 make[3] -C tools/firmware-utils compile
 make[3] -C tools/automake compile
 make[3] -C tools/missing-macros compile
 make[3] -C tools/dosfstools compile
 make[3] -C tools/upx compile
 make[3] -C tools/libtool compile
 make[3] -C tools/e2fsprogs compile
 make[3] -C tools/flex compile
 make[3] -C tools/fakeroot compile
 make[3] -C tools/gengetopt compile
 make[3] -C tools/mklibs compile
 make[3] -C tools/mm-macros compile
 make[3] -C tools/mtd-utils compile
 make[3] -C tools/patchelf compile
 make[3] -C tools/gmp compile
 make[3] -C tools/bison compile
 make[3] -C tools/mpfr compile
 make[3] -C tools/bc compile
 make[3] -C tools/findutils compile
 make[3] -C tools/mpc compile
 make[3] -C tools/quilt compile
 make[3] -C tools/padjffs2 compile
 make[2] toolchain/compile
 make[3] -C toolchain/gdb compile
 make[3] -C toolchain/binutils compile
 make[3] -C toolchain/fortify-headers compile
 make[3] -C toolchain/nasm compile
 make[3] -C toolchain/gcc/initial compile
make: *** wait: No child processes.  Stop.
make: *** Waiting for unfinished jobs....
make: *** wait: No child processes.  Stop.
Error: Process completed with exit code 143.

AmadeusGhost, 2021-08-23:
没有详细 log 就不要提 issue，浪费别人时间。

closed 2021-08-23 by AmadeusGhost

jh0072110, 2021-08-23:
> 没有详细 log 就不要提 issue，浪费别人时间。

哥logs一共就这么多行就卡住了，我也想给您多发点儿啊…而且这个问题就在选中https-dns-proxy后出现，不选就是好的

QiuSimons, 2021-08-23:
> > 没有详细 log 就不要提 issue，浪费别人时间。
> 
> 哥logs一共就这么多行就卡住了，我也想给您多发点儿啊…而且这个问题就在选中https-dns-proxy后出现，不选就是好的

V=s

jh0072110, 2021-08-23:
> 没有详细 log 就不要提 issue，浪费别人时间。

[image]
[image]
我特么也很绝望啊…我是真的想多提供一些的……