immortalwrt/immortalwrt #444: wan口协商速率

taowuhan, 2021-08-29:
我是电信PPOE拨号，首次连接正常协商模式为1Gbps全双工但是时间不会超过6小时，有时10分钟就会强制断线拨号失败，2分钟左右重拨成功，网卡速率被协商为100Mbps，用ethtool更改1Gbps全双工禁用自动协商，ppoe断开无法拨号，重启路由器又陷入循环。硬件设备已全部排查均支持1Gbps。由于LUCI界面无法查看网口速率且家庭带宽为300Mbps为一人使用，以前都没发现，偶然一次下载电信APP发现宽带24小时频繁掉线。以上问题我试着通过添加启动脚本ethtool -s eth0 speed 1000 duplex full autoneg off，无法成功拨号。用光猫拨号路由模式不会有此问题，我试着用硬路由强制网卡1Gbps全双工，正常，我不确定是哪里有问题，希望大神解答。

Un1Gfn, 2021-08-29:
你：full duplex
電信：auto negotiation
結果：這樣絕對不可能會正常的，一定會爆炸的 

https://www.nojitter.com/killer-duplex-mismatch
&bullet; auto negotiation + ( auto negotiation / half duplex ) = OK
&bullet; auto negotiation + full duplex = BOOM

先強制 half duplex 試試看

taowuhan, 2021-08-29:
我試試看

> 你：力全双工
> 电信：汽车
> 结果：这样绝对不可能会正常的，一定会爆炸的
> 
> https://www.nojitter.com/killer-duplex-mismatch
> • 自动协商+（自动协商OK/强制半双工）=
> • 自动协商+强制全双工= BOOM
> 
> 先强制1000M半双工学习看看

Un1Gfn, 2021-08-29:
https://t.me/ctcgfw_openwrt_discuss/370796  
https://t.me/ctcgfw_openwrt_discuss/370797  
https://t.me/ctcgfw_openwrt_discuss/370798

如果強制半雙工也不正常，就換網線試試

taowuhan, 2021-08-29:
> 你：force full duplex
> 電信：auto negotiation
> 結果：這樣絕對不可能會正常的，一定會爆炸的
> 
> https://www.nojitter.com/killer-duplex-mismatch
> • auto negotiation + ( auto negotiation / force half duplex ) = OK
> • auto negotiation + force full duplex = BOOM
> 
> 先強制 1000M half duplex 試試看

> https://t.me/ctcgfw_openwrt_discuss/370796
> https://t.me/ctcgfw_openwrt_discuss/370797
> https://t.me/ctcgfw_openwrt_discuss/370798
> 
> https://t.me/ctcgfw_openwrt_discuss/370803
> 
> 如果強制半雙工也不正常，就換網線試試
行不通，無法成功撥號，只有設置AUTO才可以。

taowuhan, 2021-08-29:
>

[  462.808812] rk_gmac-dwmac ff540000.ethernet eth0: Link is Down
[  474.489585] rk_gmac-dwmac ff540000.ethernet eth0: PHY [stmmac-0:01] driver [RTL8211E Gigabit Ethernet]
[  474.500029] rk_gmac-dwmac ff540000.ethernet eth0: No Safety Features support found
[  474.500705] rk_gmac-dwmac ff540000.ethernet eth0: PTP not supported by HW
[  474.501302] rk_gmac-dwmac ff540000.ethernet eth0: configuring for phy/rgmii link mode
[  478.584857] rk_gmac-dwmac ff540000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[  478.585649] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[  484.728267] rk_gmac-dwmac ff540000.ethernet eth0: Link is Down
[  494.809082] rk_gmac-dwmac ff540000.ethernet eth0: PHY [stmmac-0:01] driver [RTL8211E Gigabit Ethernet]
[  494.819619] rk_gmac-dwmac ff540000.ethernet eth0: No Safety Features support found
[  494.820297] rk_gmac-dwmac ff540000.ethernet eth0: PTP not supported by HW
[  494.820895] rk_gmac-dwmac ff540000.ethernet eth0: configuring for phy/rgmii link mode
[  498.904426] rk_gmac-dwmac ff540000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[  498.905222] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[  499.928037] rk_gmac-dwmac ff540000.ethernet eth0: Link is Down
[  510.077810] rk_gmac-dwmac ff540000.ethernet eth0: PHY [stmmac-0:01] driver [RTL8211E Gigabit Ethernet]
[  510.090283] rk_gmac-dwmac ff540000.ethernet eth0: No Safety Features support found
[  510.090961] rk_gmac-dwmac ff540000.ethernet eth0: PTP not supported by HW
[  510.091602] rk_gmac-dwmac ff540000.ethernet eth0: configuring for phy/rgmii link mode
[  525.320885] rk_gmac-dwmac ff540000.ethernet eth0: PHY [stmmac-0:01] driver [RTL8211E Gigabit Ethernet]
[  525.333311] rk_gmac-dwmac ff540000.ethernet eth0: No Safety Features support found
[  525.333988] rk_gmac-dwmac ff540000.ethernet eth0: PTP not supported by HW
[  525.334587] rk_gmac-dwmac ff540000.ethernet eth0: configuring for phy/rgmii link mode
[  533.527734] rk_gmac-dwmac ff540000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[  533.528525] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[  535.575280] rk_gmac-dwmac ff540000.ethernet eth0: Link is Down
[  540.597022] rk_gmac-dwmac ff540000.ethernet eth0: PHY [stmmac-0:01] driver [RTL8211E Gigabit Ethernet]
[  540.606661] rk_gmac-dwmac ff540000.ethernet eth0: No Safety Features support found
[  540.607338] rk_gmac-dwmac ff540000.ethernet eth0: PTP not supported by HW
[  540.607936] rk_gmac-dwmac ff540000.ethernet eth0: configuring for phy/rgmii link mode
[  555.887923] rk_gmac-dwmac ff540000.ethernet eth0: PHY [stmmac-0:01] driver [RTL8211E Gigabit Ethernet]
[  555.898388] rk_gmac-dwmac ff540000.ethernet eth0: No Safety Features support found
[  555.899065] rk_gmac-dwmac ff540000.ethernet eth0: PTP not supported by HW
[  555.899664] rk_gmac-dwmac ff540000.ethernet eth0: configuring for phy/rgmii link mode
[  571.128355] rk_gmac-dwmac ff540000.ethernet eth0: PHY [stmmac-0:01] driver [RTL8211E Gigabit Ethernet]
[  571.138135] rk_gmac-dwmac ff540000.ethernet eth0: No Safety Features support found
[  571.138813] rk_gmac-dwmac ff540000.ethernet eth0: PTP not supported by HW
[  571.139412] rk_gmac-dwmac ff540000.ethernet eth0: configuring for phy/rgmii link mode
[  580.342926] rk_gmac-dwmac ff540000.ethernet eth0: Link is Up - 100Mbps/Full - flow control rx/tx
[  580.343719] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[  581.264095] pppoe-wan: renamed from ppp0
线我换了都试了
哎不解决了，可能是中国特色吧，喜欢忽悠人，我直接降100M的资费算了。

Un1Gfn, 2021-08-29:
我输入法挂了，简繁乱跳，别介意
你用AUTO掉到100M
你用强制1000M半双工又不能工作

> 行不通，無法成功撥號，只有設置AUTO才可以。

网线换了吗?

taowuhan, 2021-08-29:
电信给我的回复是，必须用他们的光猫设备拨号路由，还问我怎么设置的桥接，要上门给我换设备，/(ㄒoㄒ)/~~ 算了 不折腾了 谢谢大神，掉线就掉线把。

taowuhan, 2021-08-29:
> 我输入法挂了，简繁乱跳，别介意 你用AUTO掉到100M 你用强制1000M半双工又不能工作
> 行不通，無法成功撥號，只有設置AUTO才可以。
> 网线换了吗?

我重接了一根短线，白橙，橙，白绿，蓝，白蓝，绿，白棕，棕

closed 2021-08-29 by taowuhan

Un1Gfn, 2021-08-29:
作爲最後手段
你還可以關閉路由器的全部自動聯網程式然後純手動聯網試試看，排除一下配置的問題
電腦上我是這樣寫進腳本手動上網的，但是路由我不會
如果你想這樣排除一下的話需要調整命令以適應openwrt環境，並且採取措施防止路由器斷線後失聯（例如用串口）
1）殺掉全部網路相關的行程
busybox killallall -9 dhcpcd dhclient dnsmasq ...
2）強制掉線
cd /sys/class/net
for i in *; do busybox ip link set dev "$i" down; done
for i in {0..9}; do rfkill block "$i"; done
3）拔掉網線
4）等五秒插上網線
5）上線
busybox ip link set dev lo up
busybox ip link set dev en0 up
6）攞IP位址
dhcpcd --timeout 0 --ipv4only en0

Un1Gfn, 2021-08-29:
https://t.me/ctcgfw_openwrt_discuss/371069  
https://t.me/ctcgfw_openwrt_discuss/371070  
可能根本和你路由器無關，而是電信給了你一個垃圾翻修貓
你把路由器收起來，電腦直連貓上網一段時間看看有沒有問題

taowuhan, 2021-08-29:
我第一条已经说了，不用软路由，用光猫自带路由功能拨号一切正常不会断网重拨号。速率可以跑满理论300M,用软或硬路由拨号就会在一段时间断线循环协商为100Mbps，但是硬路由只需强制wan口1Gbps全双工即可稳定工作。我还试过官方原版无任何插件的快照固件。一样开机可以协商1Gbps，但是还是隔一段时间就断了被协商成100Mbps。

taowuhan, 2021-08-29:
这个问题就此结束吧，感谢大神，再折腾下去，我怕电信要强制上门给我换设备了，彻底沦为大数据肉鸡，收集我的各种网络数据，我可不想用他们得新设备，听说根本没有任何权限，连桥接都改不了。超级密码还是随机变化的，根本无法破解。好在这个猫的控制权还在我手上，目前最好得办法只能是降100M的资费了，这应该是最好的解决办法了。

aeeq, 2023-04-21:
尝试 ethtool -s eth0 speed 1000 duplex half autoneg on ,会自动识别到千兆全双工