immortalwrt/immortalwrt #451: Build failure: `openwrt-21.02` branch

imWildCat, 2021-09-03:
Sample output: `nproc == 24`
```
$ make -j$(nproc) V=s

TARGET_OBJ2ELF= sh genmod.sh moddep.lst gcry_whirlpool.module build-grub-module-verifier gcry_whirlpool.mod
make[7]: Leaving directory '/home/kitten/Downloads/202109/temp/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.04/grub-core'
make[6]: Leaving directory '/home/kitten/Downloads/202109/temp/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.04/grub-core'
Making all in po
make[6]: Entering directory '/home/kitten/Downloads/202109/temp/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.04/po'
make[6]: Nothing to be done for 'all'.
make[6]: Leaving directory '/home/kitten/Downloads/202109/temp/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.04/po'
Making all in docs
make[6]: Entering directory '/home/kitten/Downloads/202109/temp/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.04/docs'
make[6]: Nothing to be done for 'all'.
make[6]: Leaving directory '/home/kitten/Downloads/202109/temp/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.04/docs'
Making all in util/bash-completion.d
make[6]: Entering directory '/home/kitten/Downloads/202109/temp/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.04/util/bash-completion.d'
../../config.status --file=grub:grub-completion.bash.in
config.status: creating grub
make[6]: Leaving directory '/home/kitten/Downloads/202109/temp/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.04/util/bash-completion.d'
make[5]: Leaving directory '/home/kitten/Downloads/202109/temp/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.04'
make[4]: Leaving directory '/home/kitten/Downloads/202109/temp/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.04'
touch /home/kitten/Downloads/202109/temp/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.04/.built
removed '/home/kitten/Downloads/202109/temp/immortalwrt/bin/targets/x86/64/packages/grub2-efi_2.04-3_x86_64.ipk'
mkdir -p /home/kitten/Downloads/202109/temp/immortalwrt/bin/targets/x86/64/packages /home/kitten/Downloads/202109/temp/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.04/ipkg-x86_64/grub2-efi/CONTROL /home/kitten/Downloads/202109/temp/immortalwrt/staging_dir/target-x86_64_musl/pkginfo
sed 's#msdos1#gpt1#g' ./files/grub-early.cfg >/home/kitten/Downloads/202109/temp/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.04/grub-early.cfg
/home/kitten/Downloads/202109/temp/immortalwrt/staging_dir/host/bin/grub-mkimage -d /home/kitten/Downloads/202109/temp/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.04/grub-core -p /boot/grub -O "x86_64"-efi -c /home/kitten/Downloads/202109/temp/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.04/grub-early.cfg -o /home/kitten/Downloads/202109/temp/immortalwrt/staging_dir/target-x86_64_musl/image/grub2/bootx64.efi at_keyboard boot chain configfile fat linux ls part_gpt reboot serial efi_gop efi_uga
/home/kitten/Downloads/202109/temp/immortalwrt/staging_dir/host/bin/grub-mkimage -d /home/kitten/Downloads/202109/temp/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.04/grub-core -p /boot/grub -O "x86_64"-efi -c ./files/grub-early.cfg -o /home/kitten/Downloads/202109/temp/immortalwrt/staging_dir/target-x86_64_musl/image/grub2/iso-bootx64.efi at_keyboard boot chain configfile fat iso9660 linux ls part_msdos part_gpt reboot serial test efi_gop efi_uga
find /home/kitten/Downloads/202109/temp/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.04/ipkg-x86_64/grub2-efi -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
export CROSS="x86_64-openwrt-linux-musl-"  NO_RENAME=1 ; NM="x86_64-openwrt-linux-musl-nm" STRIP="/home/kitten/Downloads/202109/temp/immortalwrt/staging_dir/host/bin/sstrip -z" STRIP_KMOD="/home/kitten/Downloads/202109/temp/immortalwrt/scripts/strip-kmod.sh" PATCHELF="/home/kitten/Downloads/202109/temp/immortalwrt/staging_dir/host/bin/patchelf" /home/kitten/Downloads/202109/temp/immortalwrt/scripts/rstrip.sh /home/kitten/Downloads/202109/temp/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.04/ipkg-x86_64/grub2-efi
(cd /home/kitten/Downloads/202109/temp/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.04/ipkg-x86_64/grub2-efi/CONTROL; ( echo "$CONTROL"; printf "Description: "; echo "$DESCRIPTION" | sed -e 's,^[[:space:]]*, ,g'; ) > control; chmod 644 control; ( echo "#!/bin/sh"; echo "[ \"\${IPKG_NO_SCRIPT}\" = \"1\" ] && exit 0"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_postinst \$0 \$@"; ) > postinst; ( echo "#!/bin/sh"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_prerm \$0 \$@"; ) > prerm; chmod 0755 postinst prerm;  )
install -d -m0755 /home/kitten/Downloads/202109/temp/immortalwrt/bin/targets/x86/64/packages
/home/kitten/Downloads/202109/temp/immortalwrt/staging_dir/host/bin/fakeroot /home/kitten/Downloads/202109/temp/immortalwrt/scripts/ipkg-build -m "" /home/kitten/Downloads/202109/temp/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.04/ipkg-x86_64/grub2-efi /home/kitten/Downloads/202109/temp/immortalwrt/bin/targets/x86/64/packages
Packaged contents of /home/kitten/Downloads/202109/temp/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.04/ipkg-x86_64/grub2-efi into /home/kitten/Downloads/202109/temp/immortalwrt/bin/targets/x86/64/packages/grub2-efi_2.04-3_x86_64.ipk
rm -rf /home/kitten/Downloads/202109/temp/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.04/.pkgdir/grub2-efi.installed /home/kitten/Downloads/202109/temp/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.04/.pkgdir/grub2-efi
mkdir -p /home/kitten/Downloads/202109/temp/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.04/.pkgdir/grub2-efi
sed 's#msdos1#gpt1#g' ./files/grub-early.cfg >/home/kitten/Downloads/202109/temp/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.04/grub-early.cfg
/home/kitten/Downloads/202109/temp/immortalwrt/staging_dir/host/bin/grub-mkimage -d /home/kitten/Downloads/202109/temp/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.04/grub-core -p /boot/grub -O "x86_64"-efi -c /home/kitten/Downloads/202109/temp/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.04/grub-early.cfg -o /home/kitten/Downloads/202109/temp/immortalwrt/staging_dir/target-x86_64_musl/image/grub2/bootx64.efi at_keyboard boot chain configfile fat linux ls part_gpt reboot serial efi_gop efi_uga
/home/kitten/Downloads/202109/temp/immortalwrt/staging_dir/host/bin/grub-mkimage -d /home/kitten/Downloads/202109/temp/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.04/grub-core -p /boot/grub -O "x86_64"-efi -c ./files/grub-early.cfg -o /home/kitten/Downloads/202109/temp/immortalwrt/staging_dir/target-x86_64_musl/image/grub2/iso-bootx64.efi at_keyboard boot chain configfile fat iso9660 linux ls part_msdos part_gpt reboot serial test efi_gop efi_uga
touch /home/kitten/Downloads/202109/temp/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.04/.pkgdir/grub2-efi.installed
mkdir -p /home/kitten/Downloads/202109/temp/immortalwrt/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/kitten/Downloads/202109/temp/immortalwrt/tmp/.root-copy.flock -c 'cp -fpR /home/kitten/Downloads/202109/temp/immortalwrt/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.04/.pkgdir/grub2-efi/. /home/kitten/Downloads/202109/temp/immortalwrt/staging_dir/target-x86_64_musl/root-x86/'
touch /home/kitten/Downloads/202109/temp/immortalwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.grub2-efi_installed
echo "grub2-efi" >> /home/kitten/Downloads/202109/temp/immortalwrt/staging_dir/target-x86_64_musl/pkginfo/grub2.efi.install
make[3]: Leaving directory '/home/kitten/Downloads/202109/temp/immortalwrt/package/boot/grub2'
time: package/boot/grub2/efi/compile#24.51#7.63#30.21
make[2]: Leaving directory '/home/kitten/Downloads/202109/temp/immortalwrt'
[... cut ...]
CONFIG_PACKAGE_tcping=y
CONFIG_PACKAGE_uclient-fetch=y
CONFIG_V2RAY_PLUGIN_COMPRESS_UPX=y
CONFIG_PACKAGE_vlmcsd=y

CONFIG_PACKAGE_fdisk=y
CONFIG_PACKAGE_sysfsutils=y

CONFIG_PACKAGE_bash=y

CONFIG_PACKAGE_coreutils=y
CONFIG_PACKAGE_coreutils-base64=y
CONFIG_PACKAGE_coreutils-nohup=y
CONFIG_FILEBROWSER_COMPRESS_UPX=y
CONFIG_PACKAGE_libjson-script=y
CONFIG_PACKAGE_lm-sensors=y
CONFIG_PACKAGE_openssl-util=y
CONFIG_STRACE_NONE=y
CONFIG_SYNCTHING_COMPRESS_UPX=y
CONFIG_YQ_COMPRESS_UPX=y

# end of Xorg
```
</details>

1715173329, 2021-09-03:
Please re-run with `make -j1 V=s` to see the real error.

imWildCat, 2021-09-04:
Thanks for your prompt reply!

Here's the real error, likely related to Ruby:

```
make[4]: Entering directory '/home/kitten/Downloads/202109/temp/immortalwrt/build_dir/hostpkg/ruby-3.0.2'
	BASERUBY = /home/kitten/Downloads/202109/temp/immortalwrt/staging_dir/hostpkg/bin/ruby --disable=gems
	CC = ccache gcc
	LD = ld
	LDSHARED = ccache gcc -shared
	CFLAGS = -O2 -I/home/kitten/Downloads/202109/temp/immortalwrt/staging_dir/host/include -I/home/kitten/Downloads/202109/temp/immortalwrt/staging_dir/hostpkg/include -I/home/kitten/Downloads/202109/temp/immortalwrt/staging_dir/target-x86_64_musl/host/include
	XCFLAGS = -D_FORTIFY_SOURCE=2 -fstack-protector-strong -fno-strict-overflow -fvisibility=hidden -fexcess-precision=standard -DRUBY_EXPORT -fPIE -I. -I.ext/include/x86_64-linux-gnu -I./include -I. -I./enc/unicode/12.1.0
	CPPFLAGS = -I/home/kitten/Downloads/202109/temp/immortalwrt/staging_dir/host/include -I/home/kitten/Downloads/202109/temp/immortalwrt/staging_dir/hostpkg/include -I/home/kitten/Downloads/202109/temp/immortalwrt/staging_dir/target-x86_64_musl/host/include
	DLDFLAGS = -L/home/kitten/Downloads/202109/temp/immortalwrt/staging_dir/host/lib -L/home/kitten/Downloads/202109/temp/immortalwrt/staging_dir/hostpkg/lib -L/home/kitten/Downloads/202109/temp/immortalwrt/staging_dir/target-x86_64_musl/host/lib -Wl,--compress-debug-sections=zlib -fstack-protector-strong -pie
	SOLIBS = -lz -lpthread -lrt -lrt -ldl -lcrypt -lm
	LANG = C
	LC_ALL = C
	LC_CTYPE =
	MFLAGS = -w -j1
gcc (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

cd -P . && \
type aclocal >/dev/null 2>&1 && exec aclocal; \
touch aclocal.m4
cd -P . && exec autoconf -o configure
./tool/file2lastrev.rb:6:in `require': cannot load such file -- optparse (LoadError)
	from ./tool/file2lastrev.rb:6:in `<main>'
make[4]: *** [uncommon.mk:1156: revision.tmp] Error 1
make[4]: Leaving directory '/home/kitten/Downloads/202109/temp/immortalwrt/build_dir/hostpkg/ruby-3.0.2'
make[3]: *** [Makefile:1134: /home/kitten/Downloads/202109/temp/immortalwrt/build_dir/hostpkg/ruby-3.0.2/.built] Error 2
make[3]: Leaving directory '/home/kitten/Downloads/202109/temp/immortalwrt/feeds/packages/lang/ruby'
time: package/feeds/packages/ruby/host-compile#1.32#0.08#1.38
    ERROR: package/feeds/packages/ruby [host] failed to build.
make[2]: *** [package/Makefile:112: package/feeds/packages/ruby/host/compile] Error 1
make[2]: Leaving directory '/home/kitten/Downloads/202109/temp/immortalwrt'
make[1]: *** [package/Makefile:108: /home/kitten/Downloads/202109/temp/immortalwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/kitten/Downloads/202109/temp/immortalwrt'
make: *** [/home/kitten/Downloads/202109/temp/immortalwrt/include/toplevel.mk:230: world] Error 2
```

I found Ruby cannot be unchecked in Languages. What can I do?

1715173329, 2021-09-04:
> I found Ruby cannot be unchecked in Languages. What can I do?

Ruby is selected by OpenClash.

Don't know how did this happen - maybe the dirty build environment?
I just found the same error: rbenv/ruby-build#1484

zxlhhyccc, 2021-09-04:
> Ruby

官方master分支没问题，不过9.3编译会出现一些莫名其妙的问题，比如我软路由，编译出来明明有防火墙，但实际根本使用不了。。。

imWildCat, 2021-09-04:
I can checkout to a commit from last week to work around this issue.

1715173329, 2021-09-13:
Well, Is there anyone can help me to reproduce this issue?
Cc: @QiuSimons

1715173329, 2021-09-13:
A new test now is launched.
https://github.com/immortalwrt/OpenWrt-Compilation-Checker/runs/3582108533?check_suite_focus=true

imWildCat, 2021-09-13:
Thanks. I will try a build using GH Actions later.

1715173329, 2021-09-13:
CI reported that the package `ruby` works fine.
https://github.com/immortalwrt/OpenWrt-Compilation-Checker/runs/3582108533?check_suite_focus=true#step:6:191

1715173329, 2021-09-13:
You may use our script to setup the dependencies quickly:
```bash
curl -s https://build-scripts.immortalwrt.eu.org/init_build_environment.sh | sudo bash
```

closed 2021-09-13 by 1715173329

imWildCat, 2021-09-13:
@1715173329 Cool, thanks!