immortalwrt/immortalwrt #453: Backport of JSON support

aparcar, 2021-09-06:
Hi, would there be interest to backport patches like [this](https://github.com/openwrt/openwrt/commit/881ed09ee6e23f6c224184bb7493253c4624fb9f) to 18.06.x? I'd be interesting for me since I could then offer ImmortalWrt via my online ImageBuilder [here](https://asu.aparcar.org/)

1715173329, 2021-09-06:
Hi, "image info" related commits have beed backported already, thanks for your work!

aparcar, 2021-09-06:
I see, great. Do you offer compiled binaries anywhere?

1715173329, 2021-09-07:
Sadly but not yet. We don't have enough resources currently ;(

aparcar, 2021-09-07:
Please see [this](https://github.com/aparcar/openwrt/blob/staging/.github/workflows/custom.yml) file on how to use GitHub actions to builds OpenWrt for all targets and upload it to a S3 storage.

1715173329, 2021-09-07:
> Please see [this](https://github.com/aparcar/openwrt/blob/staging/.github/workflows/custom.yml) file on how to use GitHub actions to builds OpenWrt for all targets and upload it to a S3 storage.

Thank you so much!

Ping @ElonH - Can you take a look?

ElonH, 2021-09-07:
> Please see [this](https://github.com/aparcar/openwrt/blob/staging/.github/workflows/custom.yml) file on how to use GitHub actions to builds OpenWrt for all targets and upload it to a S3 storage.

Hi, @aparcar
The file link is invalid (404, not found), could you provide a new link?

1715173329, 2021-09-07:
@ElonH please see https://github.com/aparcar/openwrt/blob/staging/.github/workflows/build-immortalwrt.yml

1715173329, 2021-09-07:
```
make[3]: Entering directory '/home/runner/work/openwrt/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-4.19.206'
2573
make[4]: *** No rule to make target '/home/runner/work/openwrt/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-exfat-oot-5.11.1/inode.o', needed by '/home/runner/work/openwrt/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-exfat-oot-5.11.1/exfat.o'.  Stop.
2574
make[3]: *** [Makefile:1544: _module_/home/runner/work/openwrt/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-exfat-oot-5.11.1] Error 2
2575
make[3]: Leaving directory '/home/runner/work/openwrt/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-4.19.206'
2576
make[2]: *** [Makefile:50: /home/runner/work/openwrt/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-exfat-oot-5.11.1/.built] Error 2
```

Well, the CI was failing due to exfat-oot... The log looks weird..

aparcar, 2021-09-07:
Not my expertise. Good luck. I will keep an eye on the CI and upload it to my own server, in case you iffer binaries any time soon please post them here

closed 2021-09-07 by aparcar