immortalwrt/immortalwrt #457: build failure of `exfat`

aparcar, 2021-09-09:
```
 make[4]: *** No rule to make target '/home/runner/work/openwrt/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-exfat-oot-5.11.1/inode.o', needed by '/home/runner/work/openwrt/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-exfat-oot-5.11.1/exfat.o'.  Stop.
make[3]: *** [Makefile:1544: _module_/home/runner/work/openwrt/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-exfat-oot-5.11.1] Error 2
make[3]: Leaving directory '/home/runner/work/openwrt/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-4.19.206'
make[2]: Leaving directory '/home/runner/work/openwrt/openwrt/package/kernel/exfat'
time: package/kernel/exfat/compile#0.90#0.39#1.15
make[2]: *** [Makefile:50: /home/runner/work/openwrt/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-exfat-oot-5.11.1/.built] Error 2
    ERROR: package/kernel/exfat failed to build.
make[1]: *** [package/Makefile:115: package/kernel/exfat/compile] Error 1
make[1]: Leaving directory '/home/runner/work/openwrt/openwrt'
make: *** [/home/runner/work/openwrt/openwrt/include/toplevel.mk:230: package/compile] Error 2
```

aparcar, 2021-09-09:
Details here https://github.com/aparcar/openwrt/runs/3550550360?check_suite_focus=true

1715173329, 2021-09-09:
Well, this was a increment compilation since last built has been failed, so I think the real error log may be gone.
Just tried to compile this package on my VM, and works fine.

I will try to fork your workflow and add some debug commands to see what happend ;)

aparcar, 2021-09-09:
I rerun it with the latest commits and it doesnt seem to change anything

https://github.com/aparcar/openwrt/runs/3560416196?check_suite_focus=true

1715173329, 2021-09-10:
After installed all of the dependencies, the error is gone.
```bash
sudo apt install -y ack antlr3 asciidoc autoconf automake autopoint binutils bison build-essential \
  bzip2 ccache cmake cpio curl device-tree-compiler ecj fastjar flex gawk gettext git git-core gperf haveged \
  help2man intltool lib32gcc1 libc6-dev-i386 libelf-dev libglib2.0-dev libgmp3-dev libltdl-dev libmpc-dev \
  libmpfr-dev libncurses5-dev libncurses5-dev libreadline-dev libssl-dev libtool libz-dev lrzsz mkisofs msmtp \
  nano ninja-build p7zip p7zip-full patch pkgconf python2.7 python3 python3-pip python3-ply python-docutils \
  qemu-utils re2c rsync scons squashfs-tools subversion swig texinfo uglifyjs upx-ucl unzip vim wget xmlto \
  xxd zlib1g-dev
```

aparcar, 2021-09-10:
This is a bit unspecific since many of those tools are actually build by the OpenWrt system itself, e.g. squashfs-tools. Do you know more precisely what's missing?

ElonH, 2021-09-10:
https://github.com/immortalwrt/opde/runs/3562788895?check_suite_focus=true#step:12:182

I use my build bot to check all kernel modules for x86/64, everything is fine

1715173329, 2021-09-10:
> https://github.com/immortalwrt/opde/runs/3562788895?check_suite_focus=true#step:12:182
> 
> I use my build bot to check all kernel modules for x86/64, everything is fine

Yes... It's a lack of dependencies, I will try to figure out which one is missing on GitHub Actions runner.

1715173329, 2021-09-11:
Don't know what happened, but the error was gone in a magic way...

1715173329, 2021-09-11:
Well, the error from `exfat` was gone in a magic way... 
However please install `qemu-utils` as it is required when generating images.

aparcar, 2021-09-12:
I added `qemu-utils` however that's unrelated.

Could you please try in your fork why it's breaking?

ElonH, 2021-09-12:
> This is a bit unspecific since many of those tools are actually build by the OpenWrt system itself, e.g. squashfs-tools. Do you know more precisely what's missing?

```
sudo apt install -y ack antlr3 asciidoc autoconf automake autopoint binutils bison build-essential \
  bzip2 ccache cmake cpio curl device-tree-compiler ecj fastjar flex gawk gettext git git-core gperf haveged \
  help2man intltool lib32gcc1 libc6-dev-i386 libelf-dev libglib2.0-dev libgmp3-dev libltdl-dev libmpc-dev \
  libmpfr-dev libncurses5-dev libncurses5-dev libreadline-dev libssl-dev libtool libz-dev lrzsz mkisofs msmtp \
  nano ninja-build p7zip p7zip-full patch pkgconf python2.7 python3 python3-pip python3-ply python-docutils \
  qemu-utils re2c rsync scons squashfs-tools subversion swig texinfo uglifyjs upx-ucl unzip vim wget xmlto \
  xxd zlib1g-dev
```

I know these exists lots of unnecessary tools in this issue. But it does work, although it take more space to install them.
Even if we finally find out the "precise" tools , it can't promise that other errors will be raised due to other tools are missed.
And that we need more work to figure out what going on again. It is very inefficient.

And you know that some errors can not be re-implemented because of different development environment. So I think we should keep the environment consistent, and reduce pain to each other developer.

At last, I refer a programmer humor: "If the legecy code work, don't touch it!!!" :-)

1715173329, 2021-09-12:
> I added `qemu-utils` however that's unrelated.
> 
> Could you please try in your fork why it's breaking?

Well, the reason why I said  it's "magic", is that I cannot really figure out why. I tried to setup a SSH session using [debugger-action](https://github.com/csexton/debugger-action), although `exfat` was failed to build in "Build Package" step, when I connected it via SSH 
 and run `make package/exfat/compile` it worked fine.

After I removed some packages from `include/target.mk` recently, the error was gone, but all of those packages should not be related to this issue.
You may also notice the build was passing:
https://github.com/aparcar/openwrt/runs/3577865905?check_suite_focus=true

BTW, it seems that you forgot to save the target info to `.config`, but just printed it to stdout:
https://github.com/aparcar/openwrt/commit/423d162d2b922389624ea85a3b89b61bfeb1ae9b#diff-3ed72a3d35055c6653a4e7ad148a80538d065099467581eba4f8dbbe1195c692R72

aparcar, 2021-09-12:
Great, I just started another build thank you! https://github.com/aparcar/openwrt/actions/runs/1227412610

closed 2021-09-13 by 1715173329

aparcar, 2021-09-13:
Works now, thanks. I'm storing my binaries [here](https://downloads.asu.aparcar.org/immortalwrt/targets/)

1715173329, 2021-09-13:
It's I that should say thank you. ;)
I really appreciate your work❤️