immortalwrt/immortalwrt #459: Downgrade glibc for 18.06

1715173329, 2021-09-12:
The latest version of glibc requires kernel >= 5.1.0 to work.
See also https://github.com/aparcar/openwrt/runs/3581398504?check_suite_focus=true#step:9:191

1715173329, 2021-09-12:
uboot for ipq40xx/fritz4040 is missing:
```bash
dd if=/home/runner/work/openwrt/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/image/uboot-fritz4040.bin >> /home/runner/work/openwrt/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/tmp/immortalwrt-ipq40xx-avm_fritzbox-4040-squashfs-eva.bin
dd: failed to open '/home/runner/work/openwrt/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/image/uboot-fritz4040.bin': No such file or directory
```

See also: https://github.com/aparcar/openwrt/runs/3581399058?check_suite_focus=true#step:13:1476

1715173329, 2021-09-13:
Build error for dap-1330-a1, maybe the `$(VERSION_DIST)` is too long?

```bash
mv /home/runner/work/openwrt/openwrt/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/tmp/immortalwrt-ar71xx-generic-dap-1330-a1-squashfs-factory.img.new /home/runner/work/openwrt/openwrt/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/tmp/immortalwrt-ar71xx-generic-dap-1330-a1-squashfs-factory.img
dd if=/home/runner/work/openwrt/openwrt/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/root.squashfs >> /home/runner/work/openwrt/openwrt/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/tmp/immortalwrt-ar71xx-generic-dap-1330-a1-squashfs-factory.img
8259+1 records in
8259+1 records out
4228782 bytes (4.2 MB, 4.0 MiB) copied, 0.0226022 s, 187 MB/s
/home/runner/work/openwrt/openwrt/staging_dir/host/bin/padjffs2 /home/runner/work/openwrt/openwrt/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/tmp/immortalwrt-ar71xx-generic-dap-1330-a1-squashfs-factory.img  64
padding image to 00560000
/home/runner/work/openwrt/openwrt/staging_dir/host/bin/mkdapimg2 -i /home/runner/work/openwrt/openwrt/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/tmp/immortalwrt-ar71xx-generic-dap-1330-a1-squashfs-factory.img -o /home/runner/work/openwrt/openwrt/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/tmp/immortalwrt-ar71xx-generic-dap-1330-a1-squashfs-factory.img.new -s HONEYBEE-FIRMWARE-DAP-1330 -v ImmortalWrt-r7621 -r Default -k 917504
*** buffer overflow detected ***: terminated
make[4]: *** [Makefile:71: /home/runner/work/openwrt/openwrt/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/tmp/immortalwrt-ar71xx-generic-dap-1330-a1-squashfs-factory.img] Aborted (core dumped)
make[4]: *** Deleting file '/home/runner/work/openwrt/openwrt/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/tmp/immortalwrt-ar71xx-generic-dap-1330-a1-squashfs-factory.img'
```

See also: https://github.com/aparcar/openwrt/runs/3581398424?check_suite_focus=true

1715173329, 2021-09-13:
Build error for mt7622/mt7629:
```bash
touch /home/runner/work/openwrt/openwrt/build_dir/target-arm_cortex-a7_musl_eabi/linux-mediatek_mt7629/linux-4.19.206/.configured
rm -f /home/runner/work/openwrt/openwrt/build_dir/target-arm_cortex-a7_musl_eabi/root-mediatek/init
make -C /home/runner/work/openwrt/openwrt/build_dir/target-arm_cortex-a7_musl_eabi/linux-mediatek_mt7629/linux-4.19.206 KCFLAGS="-fmacro-prefix-map=/home/runner/work/openwrt/openwrt/build_dir/target-arm_cortex-a7_musl_eabi=target-arm_cortex-a7_musl_eabi" HOSTCFLAGS="-O2 -I/home/runner/work/openwrt/openwrt/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="arm-openwrt-linux-muslgnueabi-" ARCH="arm" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Fri Sep 10 21:02:45 2021" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/home/runner/work/openwrt/openwrt/staging_dir/host/lib" KBUILD_HOSTLDLIBS="-L/home/runner/work/openwrt/openwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KERNELRELEASE=4.19.206 CC="ccache_cc" Image dtbs modules
make[4]: Entering directory '/home/runner/work/openwrt/openwrt/build_dir/target-arm_cortex-a7_musl_eabi/linux-mediatek_mt7629/linux-4.19.206'
  CALL    scripts/checksyscalls.sh
  CHK     include/generated/compile.h
  CC      drivers/net/phy/mtk/mt753x/mt753x_phy.o
drivers/net/phy/mtk/mt753x/mt753x_phy.c: In function 'phy_calibration':
drivers/net/phy/mtk/mt753x/mt753x_phy.c:838:34: warning: unused variable 'i' [-Wunused-variable]
  u32 reg_tmp,reg_tmp0, reg_tmp1, i;
                                  ^
drivers/net/phy/mtk/mt753x/mt753x_phy.c:838:14: warning: unused variable 'reg_tmp0' [-Wunused-variable]
  u32 reg_tmp,reg_tmp0, reg_tmp1, i;
              ^~~~~~~~
drivers/net/phy/mtk/mt753x/mt753x_phy.c: In function 'rx_dc_offset':
drivers/net/phy/mtk/mt753x/mt753x_phy.c:935:9: warning: unused variable 'reg_tmp1' [-Wunused-variable]
     u32 reg_tmp1;
         ^~~~~~~~
drivers/net/phy/mtk/mt753x/mt753x_phy.c: In function 'check_rx_dc_offset_pair_a':
drivers/net/phy/mtk/mt753x/mt753x_phy.c:952:8: warning: unused variable 'reg_val' [-Wunused-variable]
     u8 reg_val;
        ^~~~~~~
drivers/net/phy/mtk/mt753x/mt753x_phy.c: In function 'check_rx_dc_offset_pair_b':
drivers/net/phy/mtk/mt753x/mt753x_phy.c:973:8: warning: unused variable 'reg_val' [-Wunused-variable]
     u8 reg_val;
        ^~~~~~~
drivers/net/phy/mtk/mt753x/mt753x_phy.c: In function 'check_rx_dc_offset_pair_c':
drivers/net/phy/mtk/mt753x/mt753x_phy.c:994:8: warning: unused variable 'reg_val' [-Wunused-variable]
make[4]: Leaving directory '/home/runner/work/openwrt/openwrt/build_dir/target-arm_cortex-a7_musl_eabi/linux-mediatek_mt7629/linux-4.19.206'
     u8 reg_val;
        ^~~~~~~
drivers/net/phy/mtk/mt753x/mt753x_phy.c: In function 'check_rx_dc_offset_pair_d':
drivers/net/phy/mtk/mt753x/mt753x_phy.c:1015:8: warning: unused variable 'reg_val' [-Wunused-variable]
     u8 reg_val;
        ^~~~~~~
drivers/net/phy/mtk/mt753x/mt753x_phy.c: In function 'ge_cal_rext':
drivers/net/phy/mtk/mt753x/mt753x_phy.c:217:1: error: control reaches end of non-void function [-Werror=return-type]
 }
 ^
drivers/net/phy/mtk/mt753x/mt753x_phy.c: In function 'ge_cal_r50':
drivers/net/phy/mtk/mt753x/mt753x_phy.c:347:1: error: control reaches end of non-void function [-Werror=return-type]
 }
 ^
drivers/net/phy/mtk/mt753x/mt753x_phy.c: In function 'ge_cal_tx_offset':
drivers/net/phy/mtk/mt753x/mt753x_phy.c:473:1: error: control reaches end of non-void function [-Werror=return-type]
 }
 ^
drivers/net/phy/mtk/mt753x/mt753x_phy.c: In function 'ge_cal_tx_amp':
drivers/net/phy/mtk/mt753x/mt753x_phy.c:832:1: error: control reaches end of non-void function [-Werror=return-type]
 }
 ^
drivers/net/phy/mtk/mt753x/mt753x_phy.c: In function 'phy_calibration':
drivers/net/phy/mtk/mt753x/mt753x_phy.c:931:1: error: control reaches end of non-void function [-Werror=return-type]
 }
 ^
drivers/net/phy/mtk/mt753x/mt753x_phy.c: In function 'mt753x_phy_calibration':
drivers/net/phy/mtk/mt753x/mt753x_phy.c:1060:9: warning: 'ret' is used uninitialized in this function [-Wuninitialized]
  return ret;
         ^~~
cc1: some warnings being treated as errors
make[8]: *** [scripts/Makefile.build:304: drivers/net/phy/mtk/mt753x/mt753x_phy.o] Error 1
make[7]: *** [scripts/Makefile.build:544: drivers/net/phy/mtk/mt753x] Error 2
make[6]: *** [scripts/Makefile.build:544: drivers/net/phy] Error 2
make[5]: *** [scripts/Makefile.build:544: drivers/net] Error 2
make[4]: *** [Makefile:1070: drivers] Error 2
```

See also:
https://github.com/aparcar/openwrt/runs/3581399257?check_suite_focus=true
https://github.com/aparcar/openwrt/runs/3581399312?check_suite_focus=true

1715173329, 2021-09-13:
Failed to build uboot for omap/zync:
```
make[3]: Entering directory '/home/runner/work/openwrt/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/u-boot-zed/u-boot-2019.07'
  CHK     include/config/uboot.release
  CHK     include/generated/version_autogenerated.h
  CHK     include/generated/timestamp_autogenerated.h
  CHK     include/generated/generic-asm-offsets.h
  CHK     include/generated/asm-offsets.h
  CC      lib/efi_loader/helloworld.o
  CC      lib/efi_loader/efi_reloc.o
  CC      lib/efi_loader/efi_freestanding.o
  LD      lib/efi_loader/helloworld_efi.so
  OBJCOPY lib/efi_loader/helloworld.efi
  DTC     arch/arm/dts/bitmain-antminer-s9.dtb
FATAL ERROR: Unrecognized check name "avoid_unnecessary_addr_size"
```

See also:
https://github.com/aparcar/openwrt/runs/3581399567?check_suite_focus=true
https://github.com/aparcar/openwrt/runs/3581399838?check_suite_focus=true

Related discussion:
https://forum.openwrt.org/t/uboot-builds-failure-with-2020-01-for-openwrt-18-06/56009/3

1715173329, 2021-09-13:
@AmadeusGhost could you take a look ipq40xx & mt7622/mt7629?

AmadeusGhost, 2021-09-13:
> The latest version of glibc requires kernel >= 5.1.0 to work.

Maybe we need to rollback to this version in 18.06-4.x: https://github.com/openwrt/openwrt/blob/openwrt-19.07/toolchain/glibc/common.mk#L10
> uboot for ipq40xx/fritz4040 is missing

https://forum.openwrt.org/t/builderror-ipq40xx/26747

ElonH, 2021-09-13:
> Build error for mt7622/mt7629:
> 
> ```shell
> touch /home/runner/work/openwrt/openwrt/build_dir/target-arm_cortex-a7_musl_eabi/linux-mediatek_mt7629/linux-4.19.206/.configured
> rm -f /home/runner/work/openwrt/openwrt/build_dir/target-arm_cortex-a7_musl_eabi/root-mediatek/init
> make -C /home/runner/work/openwrt/openwrt/build_dir/target-arm_cortex-a7_musl_eabi/linux-mediatek_mt7629/linux-4.19.206 KCFLAGS="-fmacro-prefix-map=/home/runner/work/openwrt/openwrt/build_dir/target-arm_cortex-a7_musl_eabi=target-arm_cortex-a7_musl_eabi" HOSTCFLAGS="-O2 -I/home/runner/work/openwrt/openwrt/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="arm-openwrt-linux-muslgnueabi-" ARCH="arm" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Fri Sep 10 21:02:45 2021" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/home/runner/work/openwrt/openwrt/staging_dir/host/lib" KBUILD_HOSTLDLIBS="-L/home/runner/work/openwrt/openwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KERNELRELEASE=4.19.206 CC="ccache_cc" Image dtbs modules
> make[4]: Entering directory '/home/runner/work/openwrt/openwrt/build_dir/target-arm_cortex-a7_musl_eabi/linux-mediatek_mt7629/linux-4.19.206'
>   CALL    scripts/checksyscalls.sh
>   CHK     include/generated/compile.h
>   CC      drivers/net/phy/mtk/mt753x/mt753x_phy.o
> drivers/net/phy/mtk/mt753x/mt753x_phy.c: In function 'phy_calibration':
> drivers/net/phy/mtk/mt753x/mt753x_phy.c:838:34: warning: unused variable 'i' [-Wunused-variable]
>   u32 reg_tmp,reg_tmp0, reg_tmp1, i;
>                                   ^
> drivers/net/phy/mtk/mt753x/mt753x_phy.c:838:14: warning: unused variable 'reg_tmp0' [-Wunused-variable]
>   u32 reg_tmp,reg_tmp0, reg_tmp1, i;
>               ^~~~~~~~
> drivers/net/phy/mtk/mt753x/mt753x_phy.c: In function 'rx_dc_offset':
> drivers/net/phy/mtk/mt753x/mt753x_phy.c:935:9: warning: unused variable 'reg_tmp1' [-Wunused-variable]
>      u32 reg_tmp1;
>          ^~~~~~~~
> drivers/net/phy/mtk/mt753x/mt753x_phy.c: In function 'check_rx_dc_offset_pair_a':
> drivers/net/phy/mtk/mt753x/mt753x_phy.c:952:8: warning: unused variable 'reg_val' [-Wunused-variable]
>      u8 reg_val;
>         ^~~~~~~
> drivers/net/phy/mtk/mt753x/mt753x_phy.c: In function 'check_rx_dc_offset_pair_b':
> drivers/net/phy/mtk/mt753x/mt753x_phy.c:973:8: warning: unused variable 'reg_val' [-Wunused-variable]
>      u8 reg_val;
>         ^~~~~~~
> drivers/net/phy/mtk/mt753x/mt753x_phy.c: In function 'check_rx_dc_offset_pair_c':
> drivers/net/phy/mtk/mt753x/mt753x_phy.c:994:8: warning: unused variable 'reg_val' [-Wunused-variable]
> make[4]: Leaving directory '/home/runner/work/openwrt/openwrt/build_dir/target-arm_cortex-a7_musl_eabi/linux-mediatek_mt7629/linux-4.19.206'
>      u8 reg_val;
>         ^~~~~~~
> drivers/net/phy/mtk/mt753x/mt753x_phy.c: In function 'check_rx_dc_offset_pair_d':
> drivers/net/phy/mtk/mt753x/mt753x_phy.c:1015:8: warning: unused variable 'reg_val' [-Wunused-variable]
>      u8 reg_val;
>         ^~~~~~~
> drivers/net/phy/mtk/mt753x/mt753x_phy.c: In function 'ge_cal_rext':
> drivers/net/phy/mtk/mt753x/mt753x_phy.c:217:1: error: control reaches end of non-void function [-Werror=return-type]
>  }
>  ^
> drivers/net/phy/mtk/mt753x/mt753x_phy.c: In function 'ge_cal_r50':
> drivers/net/phy/mtk/mt753x/mt753x_phy.c:347:1: error: control reaches end of non-void function [-Werror=return-type]
>  }
>  ^
> drivers/net/phy/mtk/mt753x/mt753x_phy.c: In function 'ge_cal_tx_offset':
> drivers/net/phy/mtk/mt753x/mt753x_phy.c:473:1: error: control reaches end of non-void function [-Werror=return-type]
>  }
>  ^
> drivers/net/phy/mtk/mt753x/mt753x_phy.c: In function 'ge_cal_tx_amp':
> drivers/net/phy/mtk/mt753x/mt753x_phy.c:832:1: error: control reaches end of non-void function [-Werror=return-type]
>  }
>  ^
> drivers/net/phy/mtk/mt753x/mt753x_phy.c: In function 'phy_calibration':
> drivers/net/phy/mtk/mt753x/mt753x_phy.c:931:1: error: control reaches end of non-void function [-Werror=return-type]
>  }
>  ^
> drivers/net/phy/mtk/mt753x/mt753x_phy.c: In function 'mt753x_phy_calibration':
> drivers/net/phy/mtk/mt753x/mt753x_phy.c:1060:9: warning: 'ret' is used uninitialized in this function [-Wuninitialized]
>   return ret;
>          ^~~
> cc1: some warnings being treated as errors
> make[8]: *** [scripts/Makefile.build:304: drivers/net/phy/mtk/mt753x/mt753x_phy.o] Error 1
> make[7]: *** [scripts/Makefile.build:544: drivers/net/phy/mtk/mt753x] Error 2
> make[6]: *** [scripts/Makefile.build:544: drivers/net/phy] Error 2
> make[5]: *** [scripts/Makefile.build:544: drivers/net] Error 2
> make[4]: *** [Makefile:1070: drivers] Error 2
> ```
> 
> See also:
> https://github.com/aparcar/openwrt/runs/3581399257?check_suite_focus=true
> https://github.com/aparcar/openwrt/runs/3581399312?check_suite_focus=true

You need to add code `return all_ana_cal_status;` at the end of functions `ge_cal_rext, ge_cal_r50, ge_cal_tx_offset, ge_cal_tx_amp`

> Example: https://github.com/immortalwrt/immortalwrt/blob/8ffac602da94b1cfc13c88b4260f34557fb8042b/target/linux/mediatek/patches-4.19/0003-mt7531-gsw-internal_phy_calibration.patch#L180

add code `return ret;` at the end of function `phy_calibration`

https://github.com/immortalwrt/immortalwrt/blob/8ffac602da94b1cfc13c88b4260f34557fb8042b/target/linux/mediatek/patches-4.19/0003-mt7531-gsw-internal_phy_calibration.patch

1715173329, 2021-09-13:
> You need to add code `return all_ana_cal_status;` at the end of functions `ge_cal_rext, ge_cal_r50, ge_cal_tx_offset, ge_cal_tx_amp`
>
>
> add code `return ret;` at the end of function [...]

Well, can you commit these changes directly?

1715173329, 2021-09-13:
> Maybe we need to rollback to this version in 18.06-4.x: https://github.com/openwrt/openwrt/blob/openwrt-19.07/toolchain/glibc/common.mk#L10

Yes, also plan to downgrade GCC to 7.5.

ElonH, 2021-09-13:
> Well, can you commit these changes directly?

I have not PC currently, if u can wait, I can do that in the night

1715173329, 2021-09-13:
> I have not PC currently, if u can wait, I can do that in the night

Okay.

1715173329, 2021-09-13:
> https://forum.openwrt.org/t/builderror-ipq40xx/26747

Well, upstream seems to have this bug fixed already, I cannot reproduce it on `master` branch.

PurpleMoon369, 2021-09-18:
Build error for x86/64  when `hostapd-common` was selected.
`/home/hap/86/immortalwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib/gcc/x86_64-openwrt-linux-musl/8.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: /home/hap/86/immortalwrt/tmp/cci7kz7p.ltrans9.ltrans.o: in function `ap_sta_add':
<artificial>:(.text+0x681e): undefined reference to `sta_track_claim_taxonomy_info'
collect2: error: ld returned 1 exit status
make[4]: *** [Makefile:1301: hostapd] Error 1
make[4]: Leaving directory '/home/hap/86/immortalwrt/build_dir/target-x86_64_musl/hostapd-full-internal/hostapd-2021-05-22-b102f19b/hostapd'
make[3]: *** [Makefile:678: /home/hap/86/immortalwrt/build_dir/target-x86_64_musl/hostapd-full-internal/hostapd-2021-05-22-b102f19b/.built] Error 2
make[3]: Leaving directory '/home/hap/86/immortalwrt/package/network/services/hostapd'
time: package/network/services/hostapd/full-internal/compile#13.97#1.54#15.45
    ERROR: package/network/services/hostapd failed to build (build variant: full-internal).
make[2]: *** [package/Makefile:116: package/network/services/hostapd/compile] Error 1
make[2]: Leaving directory '/home/hap/86/immortalwrt'
make[1]: *** [package/Makefile:110: /home/hap/86/immortalwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/hap/86/immortalwrt'
make: *** [/home/hap/86/immortalwrt/include/toplevel.mk:230：world] 错误 2
`

PurpleMoon369, 2021-09-18:
Build error for x86/64 when `hostapd-common` was selected.

> /home/hap/86/immortalwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib/gcc/x86_64-openwrt-linux-musl/8.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: /home/hap/86/immortalwrt/tmp/cci7kz7p.ltrans9.ltrans.o: in function `ap_sta_add':
<artificial>:(.text+0x681e): undefined reference to `sta_track_claim_taxonomy_info'
collect2: error: ld returned 1 exit status
make[4]: *** [Makefile:1301: hostapd] Error 1
make[4]: Leaving directory '/home/hap/86/immortalwrt/build_dir/target-x86_64_musl/hostapd-full-internal/hostapd-2021-05-22-b102f19b/hostapd'
make[3]: *** [Makefile:678: /home/hap/86/immortalwrt/build_dir/target-x86_64_musl/hostapd-full-internal/hostapd-2021-05-22-b102f19b/.built] Error 2
make[3]: Leaving directory '/home/hap/86/immortalwrt/package/network/services/hostapd'
time: package/network/services/hostapd/full-internal/compile#13.97#1.54#15.45
    ERROR: package/network/services/hostapd failed to build (build variant: full-internal).
make[2]: *** [package/Makefile:116: package/network/services/hostapd/compile] Error 1
make[2]: Leaving directory '/home/hap/86/immortalwrt'
make[1]: *** [package/Makefile:110: /home/hap/86/immortalwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/hap/86/immortalwrt'
make: *** [/home/hap/86/immortalwrt/include/toplevel.mk:230：world] 错误 2

1715173329, 2021-09-18:
> Build error for x86/64 when hostapd-common was selected.

Well, you may use hostapd-openssl instead for now.

PurpleMoon369, 2021-09-18:
> > Build error for x86/64 when hostapd-common was selected.
> 
> Well, you may use hostapd-openssl instead for now.

It's worked!Thx.

1715173329, 2021-09-29:
Build error for x86/geode:
```
[ -f /home/runner/work/openwrt/openwrt/build_dir/target-i386_pentium-mmx_musl/linux-x86_geode/generic-kernel.bin -a -f /home/runner/work/openwrt/openwrt/build_dir/target-i386_pentium-mmx_musl/linux-x86_geode/root.squashfs ]
rm -fR /home/runner/work/openwrt/openwrt/build_dir/target-i386_pentium-mmx_musl/linux-x86_geode/tmp/immortalwrt-x86-geode-generic-squashfs-combined.img.gz.boot
install -d -m0755 /home/runner/work/openwrt/openwrt/build_dir/target-i386_pentium-mmx_musl/linux-x86_geode/tmp/immortalwrt-x86-geode-generic-squashfs-combined.img.gz.boot/boot/grub
sed -e 's#@SERIAL_CONFIG@#serial --unit=0 --speed=115200 --word=8 --parity=no --stop=1 --rtscts=off#g' -e 's#@TERMINAL_CONFIG@#terminal_input console serial; terminal_output console serial#g' -e 's#@ROOTPART@#root=PARTUUID=8cec6bc8-02 rootwait#g' -e 's#@GPT_ROOTPART@#root=PARTUUID=8cec6bc8-2683-766c-6156-704abe547102 rootwait#g' -e 's#@CMDLINE@#  console=tty0 console=ttyS0,115200n8#g' -e 's#@TIMEOUT@#0#g' -e 's#@TITLE@#OpenWrt#g' ./grub-pc.cfg > /home/runner/work/openwrt/openwrt/build_dir/target-i386_pentium-mmx_musl/linux-x86_geode/tmp/immortalwrt-x86-geode-generic-squashfs-combined.img.gz.boot/boot/grub/grub.cfg
cp -fpR /home/runner/work/openwrt/openwrt/build_dir/target-i386_pentium-mmx_musl/linux-x86_geode/bzImage /home/runner/work/openwrt/openwrt/build_dir/target-i386_pentium-mmx_musl/linux-x86_geode/tmp/immortalwrt-x86-geode-generic-squashfs-combined.img.gz.boot/boot/vmlinuz
cp -fpR /home/runner/work/openwrt/openwrt/staging_dir/target-i386_pentium-mmx_musl/root-x86/boot/. /home/runner/work/openwrt/openwrt/build_dir/target-i386_pentium-mmx_musl/linux-x86_geode/tmp/immortalwrt-x86-geode-generic-squashfs-combined.img.gz.boot/boot/
cp: cannot stat '/home/runner/work/openwrt/openwrt/staging_dir/target-i386_pentium-mmx_musl/root-x86/boot/.': No such file or directory
make[4]: [Makefile:164: /home/runner/work/openwrt/openwrt/build_dir/target-i386_pentium-mmx_musl/linux-x86_geode/tmp/immortalwrt-x86-geode-generic-squashfs-combined.img.gz] Error 1 (ignored)
cp -fpR /home/runner/work/openwrt/openwrt/staging_dir/target-i386_pentium-mmx_musl/image/grub2/boot.img /home/runner/work/openwrt/openwrt/build_dir/target-i386_pentium-mmx_musl/linux-x86_geode/tmp/immortalwrt-x86-geode-generic-squashfs-combined.img.gz.boot/boot/grub/
cp -fpR /home/runner/work/openwrt/openwrt/staging_dir/target-i386_pentium-mmx_musl/image/grub2/legacy-core.img /home/runner/work/openwrt/openwrt/build_dir/target-i386_pentium-mmx_musl/linux-x86_geode/tmp/immortalwrt-x86-geode-generic-squashfs-combined.img.gz.boot/boot/grub/core.img
PADDING="1" SIGNATURE="8cec6bc8"  /home/runner/work/openwrt/openwrt/scripts/gen_image_generic.sh /home/runner/work/openwrt/openwrt/build_dir/target-i386_pentium-mmx_musl/linux-x86_geode/tmp/immortalwrt-x86-geode-generic-squashfs-combined.img.gz  /home/runner/work/openwrt/openwrt/build_dir/target-i386_pentium-mmx_musl/linux-x86_geode/tmp/immortalwrt-x86-geode-generic-squashfs-combined.img.gz.boot 160 /home/runner/work/openwrt/openwrt/build_dir/target-i386_pentium-mmx_musl/linux-x86_geode/root.squashfs 256
+ [ 5 -ne 5 ]
+ OUTPUT=/home/runner/work/openwrt/openwrt/build_dir/target-i386_pentium-mmx_musl/linux-x86_geode/tmp/immortalwrt-x86-geode-generic-squashfs-combined.img.gz
+ KERNELSIZE=/home/runner/work/openwrt/openwrt/build_dir/target-i386_pentium-mmx_musl/linux-x86_geode/tmp/immortalwrt-x86-geode-generic-squashfs-combined.img.gz.boot
+ KERNELDIR=160
+ ROOTFSSIZE=/home/runner/work/openwrt/openwrt/build_dir/target-i386_pentium-mmx_musl/linux-x86_geode/root.squashfs
+ ROOTFSIMAGE=256
+ ALIGN=
+ rm -f /home/runner/work/openwrt/openwrt/build_dir/target-i386_pentium-mmx_musl/linux-x86_geode/tmp/immortalwrt-x86-geode-generic-squashfs-combined.img.gz
+ head=16
+ sect=63
/home/runner/work/openwrt/openwrt/scripts/gen_image_generic.sh: 20: Illegal number: /home/runner/work/openwrt/openwrt/build_dir/target-i386_pentium-mmx_musl/linux-x86_geode/tmp/immortalwrt-x86-geode-generic-squashfs-combined.img.gz.boot
make[4]: *** [Makefile:164: /home/runner/work/openwrt/openwrt/build_dir/target-i386_pentium-mmx_musl/linux-x86_geode/tmp/immortalwrt-x86-geode-generic-squashfs-combined.img.gz] Error 2
make[4]: Leaving directory '/home/runner/work/openwrt/openwrt/target/linux/x86/image'
make[3]: *** [Makefile:30: install] Error 2
make[2]: *** [Makefile:13: install] Error 2
make[1]: *** [target/Makefile:25: target/linux/install] Error 1
make[3]: Leaving directory '/home/runner/work/openwrt/openwrt/target/linux/x86'
make[2]: Leaving directory '/home/runner/work/openwrt/openwrt/target/linux'
time: target/linux/install#29.48#4.70#28.70
    ERROR: target/linux failed to build.
```

See also: https://github.com/aparcar/openwrt/runs/3711189011?check_suite_focus=true

closed 2022-04-20 by 1715173329