immortalwrt/immortalwrt #460: 何如让外网访问内网的445端口

edwardfhsiao, 2021-09-13:
设置了端口转发如图:

[image]

mydomain是ddns设置的域名
外网访问 mydomain.com:8081是成功的
外网访问 mydomain.com:4455是失败的
在局域网内部445(192.168.50.1:445)是可以连的

```
// 8081

telnet mydomain.com 8081

Trying 1xx.xxx.xxx.xxx...
Connected to mydomain.com.
Escape character is '^]'.
```

```
// 445

telnet mydomain.com 4455

Trying 1xx.xxx.xxx.xxx...
telnet: connect to address 1xx.xxx.xxx.xxx: Connection refused
telnet: Unable to connect to remote host
```

1715173329, 2021-09-13:
[image]
 issue != forum

closed 2021-09-13 by 1715173329

edwardfhsiao, 2021-09-13:
yes sir

edwardfhsiao, 2021-09-16:
For anyone who may have the same problem, my solution after 2 days of researching is to switch to the vanilla OpenWrt with the vanilla Samba.  And it works great.