immortalwrt/immortalwrt #467: [21.02] 内核报错

asiaon123, 2021-09-18:
运行过程中的内核日志里发现问题

```
[25216.950765] ------------[ cut here ]------------
[25216.955434] WARNING: CPU: 2 PID: 1647 at target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2021-06-06-22b69033/mt7603/mac.c:206 mt7603_filter_tx+0x16c/0x174 [mt7603e]
[25216.970187] Modules linked in: xt_FULLCONENAT pppoe ppp_async pppox ppp_mppe ppp_generic mt76x2e mt76x2_common mt76x02_lib mt7603e mt76 mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_socket xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TPROXY xt_TCPMSS xt_REDIRECT xt_MASQUERADE xt_LOG xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda ts_fsm ts_bm tcp_bbr slhc nf_tproxy_ipv6 nf_tproxy_ipv4 nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp
[25216.970381]  nf_conntrack_broadcast ts_kmp nf_conntrack_amanda nf_conncount macvlan iptable_raw iptable_nat iptable_mangle iptable_filter ipt_ECN ip_tables exfat crc_ccitt compat asn1_decoder fuse sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact cryptodev xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip6t_NPT nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 ifb sit tunnel4 ip_tunnel vfat fat nls_utf8 nls_iso8859_1 nls_cp437 crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg sha1_generic ecb authenc arc4
[25217.056076]  usb_storage leds_gpio xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd sd_mod scsi_mod gpio_button_hotplug ext4 mbcache jbd2 usbcore nls_base usb_common crc32c_generic
[25217.158380] CPU: 2 PID: 1647 Comm: hostapd Not tainted 5.4.143 #0
[25217.164444] Stack : 00000001 8007ff64 806e0000 806e5b10 80750000 806e5ad8 806e4b54 8e2f79cc
[25217.172769]         80890000 8ffbf668 80730d03 806766f0 00000002 00000001 8e2f7970 2b2f37a6
[25217.181090]         00000000 00000000 808d0000 00000000 30232033 000001ca 2e352064 34312e34
[25217.189412]         00000000 000004e6 00000000 00026aac 80000000 80750000 00000000 8dcb454c
[25217.197733]         00000009 00000002 00000002 8dcc0000 00000000 8036c2fc 00000008 80890008
[25217.206055]         ...
[25217.208497] Call Trace:
[25217.210972] [<8000ce7c>] show_stack+0x30/0x100
[25217.215419] [<805b5d64>] dump_stack+0xa4/0xdc
[25217.219781] [<8002e148>] __warn+0xc0/0x10c
[25217.223863] [<8002e1f0>] warn_slowpath_fmt+0x5c/0xac
[25217.228825] [<8dcb454c>] mt7603_filter_tx+0x16c/0x174 [mt7603e]
[25217.234727] [<8dcb19d4>] mt7603_sta_remove+0x94/0xfc [mt7603e]
[25217.240557] [<8dd62db0>] __mt76_sta_remove+0x78/0xec [mt76]
[25217.246119] [<8dd62efc>] mt76_sta_state+0xd8/0x1e4 [mt76]
[25217.251630] [<8de09f24>] sta_set_sinfo+0xfb4/0x1034 [mac80211]
[25217.257670] ---[ end trace 417b719e1fc9e712 ]---
[211496.094767] ------------[ cut here ]------------
[211496.099685] WARNING: CPU: 3 PID: 1147 at backports-5.10.42-1/net/mac80211/airtime.c:457 ieee80211_ocb_leave+0x564/0x644 [mac80211]
[211496.111494] Modules linked in: xt_FULLCONENAT pppoe ppp_async pppox ppp_mppe ppp_generic mt76x2e mt76x2_common mt76x02_lib mt7603e mt76 mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_socket xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TPROXY xt_TCPMSS xt_REDIRECT xt_MASQUERADE xt_LOG xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda ts_fsm ts_bm tcp_bbr slhc nf_tproxy_ipv6 nf_tproxy_ipv4 nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp
[211496.111688]  nf_conntrack_broadcast ts_kmp nf_conntrack_amanda nf_conncount macvlan iptable_raw iptable_nat iptable_mangle iptable_filter ipt_ECN ip_tables exfat crc_ccitt compat asn1_decoder fuse sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact cryptodev xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip6t_NPT nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 ifb sit tunnel4 ip_tunnel vfat fat nls_utf8 nls_iso8859_1 nls_cp437 crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg sha1_generic ecb authenc arc4
[211496.197514]  usb_storage leds_gpio xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd sd_mod scsi_mod gpio_button_hotplug ext4 mbcache jbd2 usbcore nls_base usb_common crc32c_generic
[211496.300104] CPU: 3 PID: 1147 Comm: mt76-tx phy1 Tainted: G        W         5.4.143 #0
[211496.308071] Stack : 00000000 8007ff64 806e0000 806e5b10 80750000 806e5ad8 806e4b54 8e34dc94
[211496.316483]         80890000 8fe07668 80730d03 806766f0 00000003 00000001 8e34dc38 2b2f37a6
[211496.324891]         00000000 00000000 808d0000 00000000 30230000 000001e1 00000010 20204720
[211496.333299]         00000000 0000291a 00000000 20572020 00000000 80750000 00000000 8de57090
[211496.341708]         00000009 00000014 00000004 00000003 00000010 8036c2fc 0000000c 8089000c
[211496.350115]         ...
[211496.352637] Call Trace:
[211496.355187] [<8000ce7c>] show_stack+0x30/0x100
[211496.359719] [<805b5d64>] dump_stack+0xa4/0xdc
[211496.364165] [<8002e148>] __warn+0xc0/0x10c
[211496.368332] [<8002e1f0>] warn_slowpath_fmt+0x5c/0xac
[211496.373485] [<8de57090>] ieee80211_ocb_leave+0x564/0x644 [mac80211]
[211496.379956] ---[ end trace 417b719e1fc9e713 ]---
```

AmadeusGhost, 2021-09-18:
You should post this issue at https://github.com/openwrt/mt76/issues .
Btw, mt76 support for mt7603 and mt7612 has been eol.

asiaon123, 2021-09-19:
谢谢，已经找到上面有相同的问题了

closed 2021-09-19 by asiaon123