immortalwrt/immortalwrt #474: 18.06dev编译失败

jh0072110, 2021-09-30:
首先，很抱歉我没有详细版的log（我没有打开详细日志）…
但是简版的log好像也比较明显，问题出在fstools这里…
附上我的三次编译失败的简版log错误部分：
1.
 make[3] -C package/utils/e2fsprogs compile
 make[3] -C package/utils/f2fs-tools compile
    ERROR: package/system/fstools failed to build.
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 1
Error: Process completed with exit code 2.
2.
 make[3] -C package/utils/e2fsprogs compile
 make[3] -C package/utils/f2fs-tools compile
 make[3] -C package/system/fstools compile
    ERROR: package/system/fstools failed to build.
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 1
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
Error: Process completed with exit code 2.
3.
 make[3] -C package/utils/e2fsprogs compile
 make[3] -C package/utils/f2fs-tools compile
 make[3] -C package/system/fstools compile
    ERROR: package/system/fstools failed to build.
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 1
Error: Process completed with exit code 2.
如果能有所帮助…是最好的…如果还是需要我提供详细log请不要客气直接告诉我就好啦，感谢付出！

AmadeusGhost, 2021-09-30:
[image]
本地编译未复现，加上没有详细 log 所以不视为问题。

closed 2021-09-30 by AmadeusGhost

kendraase, 2021-10-03:
同样的问题，以下是我的log：
```
make[3]: Entering directory '/workdir/openwrt/package/system/fstools'
mkdir -p /workdir/openwrt/dl
SHELL= flock /workdir/openwrt/tmp/.fstools-2019-03-28-ff1ded63.tar.xz.flock -c '  	/workdir/openwrt/scripts/download.pl "/workdir/openwrt/dl" "fstools-2019-03-28-ff1ded63.tar.xz" "2731bbca42c0eafda557d545ebeca243fa4048c433c3b27d31256aca356886bc" "" || (  	echo "Checking out files from the git repository..."; mkdir -p /workdir/openwrt/tmp/dl && cd /workdir/openwrt/tmp/dl && rm -rf fstools-2019-03-28-ff1ded63 && [ \! -d fstools-2019-03-28-ff1ded63 ] && git clone  https://git.openwrt.org/project/fstools.git fstools-2019-03-28-ff1ded63 && (cd fstools-2019-03-28-ff1ded63 && git checkout ff1ded63c51e84e239fb422ac8b9d15251d1221f && git submodule update --init --recursive) && echo "Packing checkout..." && export TAR_TIMESTAMP=`cd fstools-2019-03-28-ff1ded63 && git log -1 --format='\''@%ct'\''` && rm -rf fstools-2019-03-28-ff1ded63/.git && 	tar --numeric-owner --owner=0 --group=0 --mode=a-s --sort=name ${TAR_TIMESTAMP:+--mtime="$TAR_TIMESTAMP"} -c fstools-2019-03-28-ff1ded63 | 	xz -zc -7e > /workdir/openwrt/tmp/dl/fstools-2019-03-28-ff1ded63.tar.xz && mv /workdir/openwrt/tmp/dl/fstools-2019-03-28-ff1ded63.tar.xz /workdir/openwrt/dl/ && rm -rf fstools-2019-03-28-ff1ded63;  )    '
rm -f /workdir/openwrt/build_dir/target-x86_64_musl/fstools-2019-03-28-ff1ded63/.built
touch /workdir/openwrt/build_dir/target-x86_64_musl/fstools-2019-03-28-ff1ded63/.built_check
MAKEFLAGS="" /workdir/openwrt/staging_dir/host/bin/ninja  -j1 -C /workdir/openwrt/build_dir/target-x86_64_musl/fstools-2019-03-28-ff1ded63 
ninja: Entering directory `/workdir/openwrt/build_dir/target-x86_64_musl/fstools-2019-03-28-ff1ded63'
[1/12] Building C object CMakeFiles/fstools.dir/libfstools/overlay.c.o
make[3]: *** [Makefile:138: /workdir/openwrt/build_dir/target-x86_64_musl/fstools-2019-03-28-ff1ded63/.built] Error 1
FAILED: CMakeFiles/fstools.dir/libfstools/overlay.c.o 
/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/bin/x86_64-openwrt-linux-musl-gcc -Dfstools_EXPORTS  -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/workdir/openwrt/build_dir/target-x86_64_musl/fstools-2019-03-28-ff1ded63=fstools-2019-03-28-ff1ded63 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DNDEBUG -fPIC   -Os -ggdb -Wall -Werror --std=gnu99 -Wmissing-declarations -Wno-format-truncation -MD -MT CMakeFiles/fstools.dir/libfstools/overlay.c.o -MF CMakeFiles/fstools.dir/libfstools/overlay.c.o.d -o CMakeFiles/fstools.dir/libfstools/overlay.c.o -c /workdir/openwrt/build_dir/target-x86_64_musl/fstools-2019-03-28-ff1ded63/libfstools/overlay.c
/workdir/openwrt/build_dir/target-x86_64_musl/fstools-2019-03-28-ff1ded63/libfstools/overlay.c: In function 'foreachdir':
/workdir/openwrt/build_dir/target-x86_64_musl/fstools-2019-03-28-ff1ded63/libfstools/overlay.c:92:49: error: 'GLOB_ONLYDIR' undeclared (first use in this function); did you mean 'GLOB_TILDE'?
  if (!glob(globdir, GLOB_NOESCAPE | GLOB_MARK | GLOB_ONLYDIR, NULL, &gl))
                                                 ^~~~~~~~~~~~
                                                 GLOB_TILDE
/workdir/openwrt/build_dir/target-x86_64_musl/fstools-2019-03-28-ff1ded63/libfstools/overlay.c:92:49: note: each undeclared identifier is reported only once for each function it appears in
ninja: build stopped: subcommand failed.
make[3]: Leaving directory '/workdir/openwrt/package/system/fstools'
time: package/system/fstools/compile#0.24#0.07#0.28
    ERROR: package/system/fstools failed to build.
make[2]: *** [package/Makefile:115: package/system/fstools/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:109: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2
```
本人最后一次正常编译是9月26日凌晨，直到今天上午就出现报错，期间没编译过。
“偷窥”了一下题主，29日中午开始报错。
辛苦大佬了。

jh0072110, 2021-10-03:
> 同样的问题，以下是我的log：
>
> ```
> make[3]: Entering directory '/workdir/openwrt/package/system/fstools'
> mkdir -p /workdir/openwrt/dl
> SHELL= flock /workdir/openwrt/tmp/.fstools-2019-03-28-ff1ded63.tar.xz.flock -c [...]

不然你去开个新issue好了，我这个被close而且invalid了啊，我不太确定大佬们还能不能看到（当然也确实是我没有提供详版log=。=）……

kendraase, 2021-10-03:
> 不然你去开个新issue好了，我这个被close而且invalid了啊，我不太确定大佬们还能不能看到（当然也确实是我没有提供详版log=。=）……

不熟悉，以为回复后就会自动Open了。感谢提醒。