immortalwrt/immortalwrt #478: 18.06 fstools 编译报错

kendraase, 2021-10-03:
#474 相同的问题，fstools编译失败。

原issue因缺少log已被关闭，故重开一个新的，以下是我的log：
```
make[3]: Entering directory '/workdir/openwrt/package/system/fstools'
mkdir -p /workdir/openwrt/dl
SHELL= flock /workdir/openwrt/tmp/.fstools-2019-03-28-ff1ded63.tar.xz.flock -c '  	/workdir/openwrt/scripts/download.pl "/workdir/openwrt/dl" "fstools-2019-03-28-ff1ded63.tar.xz" "2731bbca42c0eafda557d545ebeca243fa4048c433c3b27d31256aca356886bc" "" || (  	echo "Checking out files from the git repository..."; mkdir -p /workdir/openwrt/tmp/dl && cd /workdir/openwrt/tmp/dl && rm -rf fstools-2019-03-28-ff1ded63 && [ \! -d fstools-2019-03-28-ff1ded63 ] && git clone  https://git.openwrt.org/project/fstools.git fstools-2019-03-28-ff1ded63 && (cd fstools-2019-03-28-ff1ded63 && git checkout ff1ded63c51e84e239fb422ac8b9d15251d1221f && git submodule update --init --recursive) && echo "Packing checkout..." && export TAR_TIMESTAMP=`cd fstools-2019-03-28-ff1ded63 && git log -1 --format='\''@%ct'\''` && rm -rf fstools-2019-03-28-ff1ded63/.git && 	tar --numeric-owner --owner=0 --group=0 --mode=a-s --sort=name ${TAR_TIMESTAMP:+--mtime="$TAR_TIMESTAMP"} -c fstools-2019-03-28-ff1ded63 | 	xz -zc -7e > /workdir/openwrt/tmp/dl/fstools-2019-03-28-ff1ded63.tar.xz && mv /workdir/openwrt/tmp/dl/fstools-2019-03-28-ff1ded63.tar.xz /workdir/openwrt/dl/ && rm -rf fstools-2019-03-28-ff1ded63;  )    '
rm -f /workdir/openwrt/build_dir/target-x86_64_musl/fstools-2019-03-28-ff1ded63/.built
touch /workdir/openwrt/build_dir/target-x86_64_musl/fstools-2019-03-28-ff1ded63/.built_check
MAKEFLAGS="" /workdir/openwrt/staging_dir/host/bin/ninja  -j1 -C /workdir/openwrt/build_dir/target-x86_64_musl/fstools-2019-03-28-ff1ded63 
ninja: Entering directory `/workdir/openwrt/build_dir/target-x86_64_musl/fstools-2019-03-28-ff1ded63'
[1/12] Building C object CMakeFiles/fstools.dir/libfstools/overlay.c.o
make[3]: *** [Makefile:138: /workdir/openwrt/build_dir/target-x86_64_musl/fstools-2019-03-28-ff1ded63/.built] Error 1
FAILED: CMakeFiles/fstools.dir/libfstools/overlay.c.o 
/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/bin/x86_64-openwrt-linux-musl-gcc -Dfstools_EXPORTS  -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/workdir/openwrt/build_dir/target-x86_64_musl/fstools-2019-03-28-ff1ded63=fstools-2019-03-28-ff1ded63 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DNDEBUG -fPIC   -Os -ggdb -Wall -Werror --std=gnu99 -Wmissing-declarations -Wno-format-truncation -MD -MT CMakeFiles/fstools.dir/libfstools/overlay.c.o -MF CMakeFiles/fstools.dir/libfstools/overlay.c.o.d -o CMakeFiles/fstools.dir/libfstools/overlay.c.o -c /workdir/openwrt/build_dir/target-x86_64_musl/fstools-2019-03-28-ff1ded63/libfstools/overlay.c
/workdir/openwrt/build_dir/target-x86_64_musl/fstools-2019-03-28-ff1ded63/libfstools/overlay.c: In function 'foreachdir':
/workdir/openwrt/build_dir/target-x86_64_musl/fstools-2019-03-28-ff1ded63/libfstools/overlay.c:92:49: error: 'GLOB_ONLYDIR' undeclared (first use in this function); did you mean 'GLOB_TILDE'?
  if (!glob(globdir, GLOB_NOESCAPE | GLOB_MARK | GLOB_ONLYDIR, NULL, &gl))
                                                 ^~~~~~~~~~~~
                                                 GLOB_TILDE
/workdir/openwrt/build_dir/target-x86_64_musl/fstools-2019-03-28-ff1ded63/libfstools/overlay.c:92:49: note: each undeclared identifier is reported only once for each function it appears in
ninja: build stopped: subcommand failed.
make[3]: Leaving directory '/workdir/openwrt/package/system/fstools'
time: package/system/fstools/compile#0.24#0.07#0.28
    ERROR: package/system/fstools failed to build.
```
根据和前issue出错时间的核对，应该是在9月26至29日之间出现的错误。

辛苦各位大佬了。

AmadeusGhost, 2021-10-03:
Ref: https://github.com/immortalwrt/immortalwrt/commit/c710efe18064e8e33f1f1348d19dd89e25c5bf0c
The changes on fstools will be migrated later.

kendraase, 2021-10-03:
额，25号凌晨的
还好我提供的信息没有产生误导
但我26号凌晨确实是编译成功的😂

无论如何，再次谢过大佬了

AmadeusGhost, 2021-10-03:
应该在 88c0c42306f7b724053882736838415965d99a6f 上修复了。

commit 88c0c423 (2021-10-03): fstools: backport changes to fix build error

Backport: "fstoools: add define for GLOB_ONLYDIR"
Fixes: #478
closed 2021-10-04 by 1715173329