immortalwrt/immortalwrt #479: [master] build failed(profile x86_64)

klever1988, 2021-10-04:
... 
make[3] -C package/network/utils/ethtool compile
 make[3] -C feeds/luci/libs/luci-lib-nixio compile
    ERROR: package/kernel/linux failed to build.
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
/home/runner/lede/include/toplevel.mk:228: recipe for target 'world' failed

$ make package/kernel/linux/compile V=sc | tee err.log
make[1]: Entering directory '/home/runner/lede'
make[2]: Entering directory '/home/runner/lede/package/libs/toolchain'
echo "libc" >> /home/runner/lede/staging_dir/target-x86_64_musl/pkginfo/toolchain.default.install
echo "libgcc" >> /home/runner/lede/staging_dir/target-x86_64_musl/pkginfo/toolchain.default.install
echo "libpthread" >> /home/runner/lede/staging_dir/target-x86_64_musl/pkginfo/toolchain.default.install
echo "librt" >> /home/runner/lede/staging_dir/target-x86_64_musl/pkginfo/toolchain.default.install
make[2]: Leaving directory '/home/runner/lede/package/libs/toolchain'
time: package/libs/toolchain/compile#0.19#0.08#0.24
make[2]: Entering directory '/home/runner/lede/package/firmware/linux-firmware'
mkdir -p /home/runner/lede/dl
SHELL= flock /home/runner/lede/tmp/.linux-firmware-20210511.tar.xz.flock -c '   /home/runner/lede/scripts/download.pl "/home/runner/lede/dl" "linux-firmware-20210511.tar.xz" "2aa6ae8b9808408f9811ac38f00c188e53e984a2b3990254f6c9c02c1ab13417" "" "@KERNEL/linux/kernel/firmware"    '
echo "ath9k-htc-firmware" >> /home/runner/lede/staging_dir/target-x86_64_musl/pkginfo/linux-firmware.default.install
echo "r8169-firmware" >> /home/runner/lede/staging_dir/target-x86_64_musl/pkginfo/linux-firmware.default.install
echo "bnx2-firmware" >> /home/runner/lede/staging_dir/target-x86_64_musl/pkginfo/linux-firmware.default.install
make[2]: Leaving directory '/home/runner/lede/package/firmware/linux-firmware'
time: package/firmware/linux-firmware/compile#1.00#0.53#1.17
make[2]: Entering directory '/home/runner/lede/package/firmware/prism54-firmware'
make[2]: Nothing to be done for 'compile'.
make[2]: Leaving directory '/home/runner/lede/package/firmware/prism54-firmware'
time: package/firmware/prism54-firmware/compile#0.32#0.06#0.37
make[2]: Entering directory '/home/runner/lede/package/kernel/linux'
modules/crypto.mk:748: recipe for target '/home/runner/lede/bin/targets/x86/64/packages/kmod-crypto-misc_5.10.70-1_x86_64.ipk' failed
make[2]: Leaving directory '/home/runner/lede/package/kernel/linux'
time: package/kernel/linux/compile#2.01#0.14#2.14
    ERROR: package/kernel/linux failed to build.
package/Makefile:114: recipe for target 'package/kernel/linux/compile' failed
make[1]: Leaving directory '/home/runner/lede'
/home/runner/lede/include/toplevel.mk:228: recipe for target 'package/kernel/linux/compile' failed

1715173329, 2021-10-10:
I cannot reproduce your problem.

klever1988, 2021-10-10:
我再看看