immortalwrt/immortalwrt #480: 将内核改回 5.4 网络设备驱动缺失

imoldchen, 2021-10-05:
5.10 对某些旧插件不兼容，改回 5.4 的话，make menuconfig -》 Kernel modules -》Network Devices 列表里只剩少数几个驱动了，比较重要的 Intel 万兆网卡驱动
kmod-igbvf
kmod-ixgbe
kmod-ixgbevf
都不见了。改回 5.10 是有的。

zxlhhyccc, 2021-10-05:
查看一下驱动是不是做了内核标记

1715173329, 2021-10-05:
[image]
建议检查下自己的配置。
btw, 5.4 was gone on master branch ;)

closed 2021-10-05 by 1715173329

zxlhhyccc, 2021-10-05:
> [image] 建议检查下自己的配置。 btw, 5.4 was gone on master branch ;)

提问不说哪个分支。。。。

imoldchen, 2021-10-05:
> 查看一下驱动是不是做了内核标记

查了一下，netdevices.mk 有指定 5.10
define KernelPackage/ixgbe
  SUBMENU:=$(NETWORK_DEVICES_MENU)
  TITLE:=Intel(R) 82598/82599 PCI-Express 10 Gigabit Ethernet support
  DEPENDS:=@PCI_SUPPORT +kmod-mdio +kmod-ptp +kmod-hwmon-core +kmod-libphy +LINUX_5_10:kmod-mdio-devres
  KCONFIG:=CONFIG_IXGBE \
    CONFIG_IXGBE_VXLAN=n \
    CONFIG_IXGBE_HWMON=y \
    CONFIG_IXGBE_DCA=n
  FILES:=$(LINUX_DIR)/drivers/net/ethernet/intel/ixgbe/ixgbe.ko
  AUTOLOAD:=$(call AutoLoad,35,ixgbe)
endef

define KernelPackage/ixgbe/description
 Kernel modules for Intel(R) 82598/82599 PCI-Express 10 Gigabit Ethernet adapters.
endef

$(eval $(call KernelPackage,ixgbe))

zxlhhyccc, 2021-10-05:
> > 查看一下驱动是不是做了内核标记
> > master 分支
> > 查了一下，netdevices.mk 有指定 5.10
> > define KernelPackage/ixgbe
> > SUBMENU:=$(NETWORK_DEVICES_MENU)
> > TITLE:=Intel(R) 82598/82599 PCI-Express 10 Gigabit Ethernet support
> > DEPENDS:=@PCI_SUPPORT +kmod-mdio +kmod-ptp +kmod-hwmon-core +kmod-libphy +LINUX_5_10:kmod-mdio-devres
> > KCONFIG:=CONFIG_IXGBE 
> >     CONFIG_IXGBE_VXLAN=n 
> >     CONFIG_IXGBE_HWMON=y 
> >     CONFIG_IXGBE_DCA=n
> > FILES:=$(LINUX_DIR)/drivers/net/ethernet/intel/ixgbe/ixgbe.ko
> > AUTOLOAD:=$(call AutoLoad,35,ixgbe)
> > endef
> 
> define KernelPackage/ixgbe/description Kernel modules for Intel(R) 82598/82599 PCI-Express 10 Gigabit Ethernet adapters. endef
> 
> $(eval $(call KernelPackage,ixgbe))

貌似缺少依赖吧？

imoldchen, 2021-10-05:
> LINUX_5_10:kmod-mdio

> 貌似缺少依赖吧？

是不是要有 LINUX_5_4:kmod-mdio 才行？

imoldchen, 2021-10-05:
mdio 好像也是必须 LINUX_5_10

```
define KernelPackage/mdio-devres
  SUBMENU:=$(NETWORK_DEVICES_MENU)
  TITLE:=Supports MDIO device registration
  DEPENDS:=@LINUX_5_10 +kmod-libphy PACKAGE_kmod-of-mdio:kmod-of-mdio
  KCONFIG:=CONFIG_MDIO_DEVRES
  HIDDEN:=1
  FILES:=$(LINUX_DIR)/drivers/net/phy/mdio_devres.ko
  AUTOLOAD:=$(call AutoProbe,mdio-devres)
endef
```

define KernelPackage/mdio-devres/description
 Supports MDIO device registration
endef

$(eval $(call KernelPackage,mdio-devres))

1715173329, 2021-10-05:
I would like to advice you to take a look at wiki, about how dependencies works.
https://openwrt.org/docs/guide-developer/dependencies

zxlhhyccc, 2021-10-05:
> mdio 好像也是必须 LINUX_5_10
>
> ```
> define KernelPackage/mdio-devres
>   SUBMENU:=$(NETWORK_DEVICES_MENU)
>   TITLE:=Supports MDIO device registration
>   DEPENDS:=@LINUX_5_10 +kmod-libphy [...]

你把指定的去掉就可以有了，这么简单的问题。。。。