immortalwrt/immortalwrt #501: [18.06]-Newifi-D2 MTK7621 编译后不显示无线wifi-openwrt，且无法进去路由界面

lgfyxx1, 2021-11-02:
如标题，多次尝试编译后的完整固件均是相同问题， 应该是固件问题。

刷回一个老固件一切正常。

感谢大佬！

AmadeusGhost, 2021-11-03:
你不接个 ttl 我们怎么知道是什么情况🌚

lgfyxx1, 2021-11-04:
不知道。。。。ttl 是啥

Thanf-code, 2021-11-05:
自9月中旬源码更新以来，新路由三的cpu使用率就比较高，一直在28%以上，不知道为啥？

lgfyxx1, 2021-11-05:
> 自9月中旬源码更新以来，新路由三的cpu使用率就比较高，一直在28%以上，不知道为啥？

你有编译最新的固件吗? 想问下能不能进去路由器管理界面呢？谢谢

Thanf-code, 2021-11-05:
> > 自9月中旬源码更新以来，新路由三的cpu使用率就比较高，一直在28%以上，不知道为啥？
> 
> 你有编译最新的固件吗? 想问下能不能进去路由器管理界面呢？谢谢

有，但是cpu占用较高问题未解决

1715173329, 2021-11-05:
Well, can you guys send me a sample firmware that you compiled?

Thanf-code, 2021-11-05:
> Well, can you guys send me a sample firmware that you compiled? Also with `.config`.

I compiled it using the 21.02 branch

Thanf-code, 2021-11-07:
> > 自9月中旬源码更新以来，新路由三的cpu使用率就比较高，一直在28%以上，不知道为啥？
> 
> 你有编译最新的固件吗? 想问下能不能进去路由器管理界面呢？谢谢

我用master分支编译出来还算比较正常看你需不需要

lgfyxx1, 2021-11-07:
[immortalwrt-ramips-mt7621-d-team_newifi-d2-squashfs-sysupgrade.zip](https://github.com/immortalwrt/immortalwrt/files/7492400/immortalwrt-ramips-mt7621-d-team_newifi-d2-squashfs-sysupgrade.zip)
IP能进去了，无线还是有问题，好几个终端连不上，显示
“MediaTek MT76x2E 802.11nac (radio1)
信道: 36 (5.180 GHz) | 传输速率: ？ Mbit/s  
WIFI 0% ”

zxlhhyccc, 2021-11-07:
> [immortalwrt-ramips-mt7621-d-team_newifi-d2-squashfs-sysupgrade.zip](https://github.com/immortalwrt/immortalwrt/files/7492400/immortalwrt-ramips-mt7621-d-team_newifi-d2-squashfs-sysupgrade.zip) [...]

你是哪个分支编译的？如果是master分支编译的，一定是hostpad更新后造成的。。。
解决办法：
1、回退hostpad；
2、对mac80211.sh添加一行代码：
set wireless.default_radio${devidx}.iw_qos_map_set=none

lgfyxx1, 2021-11-08:
> 你是哪个分支编译的？如果是master分支编译的，一定是hostpad更新后造成的。。。 解决办法： 1、回退hostpad； 2、对mac80211.sh添加一行代码： set wireless.default_radio${devidx}.iw_qos_map_set=none

18.06 master

lgfyxx1, 2021-11-08:
> 你是哪个分支编译的？如果是master分支编译的，一定是hostpad更新后造成的。。。 解决办法： 1、回退hostpad； 2、对mac80211.sh添加一行代码： set wireless.default_radio${devidx}.iw_qos_map_set=none

非常感谢，我试试

lgfyxx1, 2021-11-08:
搜到好多，想问下每一个都需要改吗？ 谢谢

[image]

Thanf-code, 2021-11-08:
> [immortalwrt-ramips-mt7621-d-team_newifi-d2-squashfs-sysupgrade.zip](https://github.com/immortalwrt/immortalwrt/files/7492400/immortalwrt-ramips-mt7621-d-team_newifi-d2-squashfs-sysupgrade.zip) [...]

这是5G信号 要设备支持5G才能连上

zxlhhyccc, 2021-11-08:
> 搜到好多，想问下每一个都需要改吗？ 谢谢
> 
> [image]

机器上改的话：修改/etc/config/wireless，在2.4G和5G下面添加：option iw_qos_map_set 'none'试试。
比如：
[image]

AmadeusGhost, 2021-11-29:
[image]
之前我也遇到了 18.06 新三无法启动的问题，`make dirclean` 之后实测能正常启动。

closed 2021-12-08 by AmadeusGhost