immortalwrt/immortalwrt #504: [18.06]-R2s 使用 RTL8821CU芯片，COMFAST 811AC 插入后内核报错,如果插着开机,则无法启动

FirYuen, 2021-11-05:
内核打印如下
```
[  127.942699] usb 1-1: new high-speed USB device number 2 using ehci-platform
[  128.100772] RTW: [HALMAC]11692M
[  128.100772] HALMAC_MAJOR_VER = 1
[  128.100772] HALMAC_PROTOTYPE_VER = 4
[  128.100772] HALMAC_MINOR_VER = 19
[  128.100772] HALMAC_PATCH_VER = 3
[  128.103992] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[  128.364164] RTW: HW EFUSE
[  128.364421] RTW: 0x000: 29 81 00 BC  09 00 21 00  6E 04 A4 10  10 00 30 0B  
[  128.365104] RTW: 0x010: 25 25 25 25  25 25 2B 2B  2B 2B 2B 04  FF FF FF FF  
[  128.365785] RTW: 0x020: FF FF 29 28  28 28 24 24  25 25 26 26  23 23 23 23  
[  128.366467] RTW: 0x030: 24 FF FF FF  FF FF D0 FF  FF FF 25 25  25 25 25 25  
[  128.367562] RTW: 0x040: 2B 2B 2B 2B  2B 04 FF FF  FF FF FF FF  FF FF FF FF  
[  128.368250] RTW: 0x050: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  128.368930] RTW: 0x060: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  128.369611] RTW: 0x070: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  128.370291] RTW: 0x080: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  128.371346] RTW: 0x090: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  128.372030] RTW: 0x0A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  128.372711] RTW: 0x0B0: FF FF FF FF  FF FF FF FF  7F 30 1B 00  FF FF FF FF  
[  128.373392] RTW: 0x0C0: FF 22 00 41  00 00 00 00  00 FF 22 FF  FF FF FF FF  
[  128.374071] RTW: 0x0D0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  128.375102] RTW: 0x0E0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  128.375786] RTW: 0x0F0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  128.376467] RTW: 0x100: DA 0B 11 C8  E3 46 02 E0  E1 A9 18 5B  66 09 03 52  
[  128.377146] RTW: 0x110: 65 61 6C 74  65 6B 0E 03  38 30 32 2E  31 31 61 63  
[  128.377826] RTW: 0x120: 20 4E 49 43  08 03 31 32  33 34 35 36  FF FF FF FF  
[  128.378516] RTW: 0x130: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  128.379331] RTW: 0x140: FF FF FF FF  00 00 31 0F  FF FF FF FF  FF FF FF FF  
[  128.380107] RTW: 0x150: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  128.380788] RTW: 0x160: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  128.381469] RTW: 0x170: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  128.382150] RTW: 0x180: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  128.382861] RTW: 0x190: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  128.383686] RTW: 0x1A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  128.384367] RTW: 0x1B0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  128.385048] RTW: 0x1C0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  128.385729] RTW: 0x1D0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  128.386410] RTW: 0x1E0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  128.387114] RTW: 0x1F0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  128.388242] RTW: hal_com_config_channel_plan chplan:0x7F
[  128.401487] RTW: [HALMAC][ALWAYS]shall R reg twice!!
[  128.405615] RTW: WARN [HALMAC][WARN]H2C/C2H ver is compatible!!
[  128.515628] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[  128.516364] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[  128.517647] RTW: rtw_regsty_chk_target_tx_power_valid return _FALSE for band:0, path:0, rs:0, t:-1
[  128.521182] RTW: rtw_ndev_init(wlan0) if1 mac_addr=e0:e1:a9:18:5b:66
[  128.523228] RTW: rtw_ndev_init(wlan1) if2 mac_addr=e2:e1:a9:18:5b:66
[  128.525494] reg-fixed-voltage vcc-rtl8153-regulator: deferred probe timeout, ignoring dependency
[ 1321.386486] RTW: [HALMAC][ALWAYS]shall R reg twice!!
[ 1321.390485] RTW: WARN [HALMAC][WARN]H2C/C2H ver is compatible!!
[ 1322.360240] RTW: wlan0- hw port(0) mac_addr =e0:e1:a9:18:5b:66
[ 1322.360977] RTW: wlan1- hw port(1) mac_addr =e2:e1:a9:18:5b:66
```
使用驱动仓库有类似的[issue ](https://github.com/brektrou/rtl8821CU/issues/15)

FirYuen, 2021-11-05:
```
root@ImmortalWrt:~# uname -a
Linux ImmortalWrt 5.10.76 #0 SMP PREEMPT Wed Nov 3 07:19:36 2021 aarch64 GNU/Linux
root@ImmortalWrt:~# cat /etc/os-release 
NAME="ImmortalWrt"
VERSION="SNAPSHOT"
ID="immortalwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="ImmortalWrt SNAPSHOT"
VERSION_ID="snapshot"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r20689-feef75ea4d"
OPENWRT_BOARD="rockchip/armv8"
OPENWRT_ARCH="aarch64_generic"
OPENWRT_TAINTS="no-all busybox"
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="ImmortalWrt SNAPSHOT r20689-feef75ea4d"
root@ImmortalWrt:~# 
```

AmadeusGhost, 2021-11-05:
> COMFAST 811AC 插入后内核报错

我没有从你提供的日志里面看到任何报错。
> 如果插着开机，则无法启动

日志呢？BTW 这可能是友善的问题。

xiaobaishu1, 2021-11-07:
> > COMFAST 811AC 插入后崩溃报错
> 
> 我没有从你提供的日志里面看到的。
> 
> > 如果插着启动，则无法启动
> 
> 呢？顺便说一句，这可能是友善的问题。

-source --target_arch=arm
bash: npm: command not found
Makefile:76: recipe for target '/root/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-request-2.88.2/.built' failed
make[3]: *** [/root/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-request-2.88.2/.built] Error 127
make[3]: Leaving directory '/root/openwrt/feeds/packages/lang/node-request'
time: package/feeds/packages/node-request/compile#0.10#0.05#0.16
    ERROR: package/feeds/packages/node-request failed to build.
package/Makefile:113: recipe for target 'package/feeds/packages/node-request/compile' failed
make[2]: *** [package/feeds/packages/node-request/compile] Error 1
make[2]: Leaving directory '/root/openwrt'
package/Makefile:109: recipe for target '/root/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile' failed
make[1]: *** [/root/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/root/openwrt'
/root/openwrt/include/toplevel.mk:228: recipe for target 'world' failed

FirYuen, 2021-11-09:
> > COMFAST 811AC 插入后内核报错
> 
> 我没有从你提供的日志里面看到任何报错。
> 
> > 如果插着开机，则无法启动
> 
> 日志呢？BTW 这可能是友善的问题。

启动的日志如下:
```
root@ImmortalWrt:/# 򀽄DR version 1.17 20210528
ID:0x805 N
In
DDR4
333MHz
Bus Width=32 Col=10 Bank=4 Bank Group=2 Row=15 CS=1 Die Bus-Width=16 Size=768MB
ddrconfig:14
OUT
Boot1 Release Time: May 13 2019 17:34:36, version: 2.50
ChipType = 0x11, 248
mmc2:cmd1,20
emmc reinit
mmc2:cmd1,20
emmc reinit
mmc2:cmd1,20
SdmmcInit=2 1
mmc0:cmd5,20
SdmmcInit=0 0
BootCapSize=0
UserCapSize=3772MB
FwPartOffset=2000 , 0
StorageInit ok = 20830
Raw SecureMode = 0
SecureInit read PBA: 0x4
SecureInit read PBA: 0x404
SecureInit read PBA: 0x804
SecureInit read PBA: 0xc04
SecureInit read PBA: 0x1004
SecureInit ret = 0, SecureMode = 0
atags_set_bootdev: ret:(0)
GPT 0x337a9f0 signature is wrong
recovery gpt...
GPT 0x337a9f0 signature is wrong
recovery gpt fail!
LoadTrust Addr:0x4000
No find bl30.bin
No find bl32.bin
Load uboot, ReadLba = 2000
hdr 000000000337a3b0 + 0x0:0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,

Load OK, addr=0x200000, size=0xa7480
RunBL31 0x40000
INFO:    Preloader serial: 2
NOTICE:  BL31: v1.3(release):dfd73e4
NOTICE:  BL31: Built : 16:45:02, May 28 2021
NOTICE:  BL31:Rockchip release version: v1.3
INFO:    ARM GICv2 driver initialized
INFO:    Using opteed sec cpu_context!
INFO:    boot cpu mask: 1
INFO:    plat_rockchip_pmu_init: pd status 0xe
INFO:    BL31: Initializing runtime services
WARNING: No OPTEE provided by BL2 boot loader, Booting device without OPTEE initialization. SMC`s destined for OPTEE will return SMC_UNK
ERROR:   Error initializing runtime service opteed_fast
INFO:    BL31: Preparing for EL3 exit to normal world
INFO:    Entry point address = 0x200000
INFO:    SPSR = 0x3c9

U-Boot 2021.07 (Nov 03 2021 - 07:19:36 +0000) OpenWrt

Model: FriendlyElec NanoPi R2S
[... 442 lines cut ...]
[   14.327298] rk_gmac-dwmac ff540000.ethernet eth0: PTP not supported by HW
[   14.328507] rk_gmac-dwmac ff540000.ethernet eth0: configuring for phy/rgmii link mode

BusyBox v1.34.1 (2021-11-03 07:19:36 UTC) built-in shell (ash)

┌─────────────────────────────────────────────┐
│                                             │
│ mmmmm                         m       ""#   │
│   #   mmmmm mmmmm  mmm  mmm mm#mm  mmm  #   │
│   #   # # # # # # #" "# #" "  #   "   # #   │
│   #   # # # # # # #   # #     #   m"""# #   │
│ mm#mm # # # # # # "#m#" #     "mm "mm"# "mm │
│                                             │
│─────────────────────────────────────────────│
│            ImmortalWrt SNAPSHOT             │
└─────────────────────────────────────────────┘
root@ImmortalWrt:/# 
root@ImmortalWrt:/# 
root@ImmortalWrt:/# 
root@ImmortalWrt:/# 
root@ImmortalWrt:/# ifconfig 

```

启动后红灯一直闪烁,输入 ifconfig后不显示命令内容直接卡死

FirYuen, 2021-11-09:
这是输入 logread后的局部打印
```
Thu Jan 21 16:50:18 2016 kern.notice kernel: [    8.706355] ntfs3: Activated 32 bits per cluster
Thu Jan 21 16:50:18 2016 kern.info kernel: [    8.712689] usbcore: registered new interface driver r8152
Thu Jan 21 16:50:18 2016 kern.info kernel: [    8.738098] usbcore: registered new interface driver ums-alauda
Thu Jan 21 16:50:18 2016 kern.info kernel: [    8.740176] usbcore: registered new interface driver ums-cypress
Thu Jan 21 16:50:18 2016 kern.info kernel: [    8.741927] usbcore: registered new interface driver ums-datafab
Thu Jan 21 16:50:18 2016 kern.info kernel: [    8.743618] usbcore: registered new interface driver ums-freecom
Thu Jan 21 16:50:18 2016 kern.info kernel: [    8.745236] usbcore: registered new interface driver ums-isd200
Thu Jan 21 16:50:18 2016 kern.info kernel: [    8.747163] usbcore: registered new interface driver ums-jumpshot
Thu Jan 21 16:50:18 2016 kern.info kernel: [    8.748982] usbcore: registered new interface driver ums-karma
Thu Jan 21 16:50:18 2016 kern.info kernel: [    8.751443] usbcore: registered new interface driver ums-sddr09
Thu Jan 21 16:50:18 2016 kern.info kernel: [    8.753536] usbcore: registered new interface driver ums-sddr55
Thu Jan 21 16:50:18 2016 kern.info kernel: [    8.755492] usbcore: registered new interface driver ums-usbat
Thu Jan 21 16:50:18 2016 kern.info kernel: [    8.757385] usbcore: registered new interface driver usblp
Thu Jan 21 16:50:18 2016 kern.info kernel: [    8.836103] xt_time: kernel timezone is -0000
Thu Jan 21 16:50:18 2016 kern.info kernel: [    8.895194] usb 4-1: reset SuperSpeed Gen 1 USB device number 2 using xhci-hcd
Thu Jan 21 16:50:18 2016 kern.err kernel: [    8.931616] r8152 4-1:1.0 (unnamed net_device) (uninitialized): Invalid ether addr 00:00:00:00:00:00
Thu Jan 21 16:50:18 2016 kern.info kernel: [    8.932449] r8152 4-1:1.0 (unnamed net_device) (uninitialized): Random ether addr 66:7b:7e:24:ba:3a
Thu Jan 21 16:50:18 2016 kern.info kernel: [    8.972117] r8152 4-1:1.0: load rtl8153b-2 v1 10/23/19 successfully
Thu Jan 21 16:50:18 2016 kern.info kernel: [    9.008153] r8152 4-1:1.0 eth1: v1.11.11
Thu Jan 21 16:50:18 2016 kern.warn kernel: [    9.008816] reg-fixed-voltage vcc-rtl8153-regulator: deferred probe timeout, ignoring dependency
Thu Jan 21 16:50:18 2016 kern.info kernel: [    9.027961] usbcore: registered new interface driver mt76x2u
Thu Jan 21 16:50:18 2016 kern.info kernel: [    9.032639] PPP generic driver version 2.4.2
Thu Jan 21 16:50:18 2016 kern.info kernel: [    9.034569] PPP MPPE Compression module registered
Thu Jan 21 16:50:18 2016 kern.info kernel: [    9.037296] NET: Registered protocol family 24
Thu Jan 21 16:50:18 2016 kern.warn kernel: [    9.243515] RTL871X: module init start
Thu Jan 21 16:50:18 2016 kern.warn kernel: [    9.243881] RTL871X: rtl8812au v4.3.14_13455.20150212_BTCOEX20150128-51
Thu Jan 21 16:50:18 2016 kern.warn kernel: [    9.244464] RTL871X: rtl8812au BT-Coex version = BTCOEX20150128-51
Thu Jan 21 16:50:18 2016 kern.info kernel: [    9.245236] usbcore: registered new interface driver rtl8812au
Thu Jan 21 16:50:18 2016 kern.warn kernel: [    9.245752] RTL871X: module init ret=0
Thu Jan 21 16:50:18 2016 kern.warn kernel: [    9.626542] RTW: module init start
Thu Jan 21 16:50:18 2016 kern.warn kernel: [    9.626952] RTW: rtl8821cu v5.4.1_28754.20180921_COEX20180712-3232
Thu Jan 21 16:50:18 2016 kern.warn kernel: [    9.627497] RTW: build time: Nov  3 2021 07:19:36
Thu Jan 21 16:50:18 2016 kern.warn kernel: [    9.627914] RTW: rtl8821cu BT-Coex version = COEX20180712-3232
Thu Jan 21 16:50:18 2016 kern.warn kernel: [    9.628762] RTW: [HALMAC]11692M
Thu Jan 21 16:50:18 2016 kern.warn kernel: [    9.628762] HALMAC_MAJOR_VER = 1
Thu Jan 21 16:50:18 2016 kern.warn kernel: [    9.628762] HALMAC_PROTOTYPE_VER = 4
Thu Jan 21 16:50:18 2016 kern.warn kernel: [    9.628762] HALMAC_MINOR_VER = 19
Thu Jan 21 16:50:18 2016 kern.warn kernel: [    9.628762] HALMAC_PATCH_VER = 3
Thu Jan 21 16:50:18 2016 kern.warn kernel: [    9.632635] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
Thu Jan 21 16:50:18 2016 kern.warn kernel: [    9.931675] RTW: HW EFUSE
Thu Jan 21 16:50:18 2016 kern.warn kernel: [    9.931941] RTW: 0x000: 29 81 00 BC  09 00 21 00  6E 04 A4 10  10 00 30 0B
Thu Jan 21 16:50:18 2016 kern.warn kernel: [    9.932615] RTW: 0x010: 25 25 25 25  25 25 2B 2B  2B 2B 2B 04  FF FF FF FF
Thu Jan 21 16:50:18 2016 kern.warn kernel: [    9.933287] RTW: 0x020: FF FF 29 28  28 28 24 24  25 25 26 26  23 23 23 23
Thu Jan 21 16:50:18 2016 kern.warn kernel: [    9.933957] RTW: 0x030: 24 FF FF FF  FF FF D0 FF  FF FF 25 25  25 25 25 25
Thu Jan 21 16:50:18 2016 kern.warn kernel: [    9.934627] RTW: 0x040: 2B 2B 2B 2B  2B 04 FF FF  FF FF FF FF  FF FF FF FF
Thu Jan 21 16:50:18 2016 kern.warn kernel: [    9.935324] RTW: 0x050: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
Thu Jan 21 16:50:18 2016 kern.warn kernel: [    9.935995] RTW: 0x060: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
Thu Jan 21 16:50:18 2016 kern.warn kernel: [    9.936666] RTW: 0x070: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
Thu Jan 21 16:50:18 2016 kern.warn kernel: [    9.937337] RTW: 0x080: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
Thu Jan 21 16:50:18 2016 kern.warn kernel: [    9.938008] RTW: 0x090: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
Thu Jan 21 16:50:18 2016 kern.warn kernel: [    9.938680] RTW: 0x0A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[... cut ...]
Thu Jan 21 16:50:21 2016 daemon.notice netifd: Interface 'wan6' is enabled
Thu Jan 21 16:50:21 2016 daemon.notice netifd: Network device 'lo' link is up
Thu Jan 21 16:50:21 2016 daemon.notice netifd: Interface 'loopback' has link connectivity
Thu Jan 21 16:50:22 2016 daemon.err odhcpd[1583]: Failed to send to ff02::1%lan@br-lan (Address not available)
Thu Jan 21 16:50:22 2016 authpriv.warn dropbear[2072]: Failed listening on '22': Error listening: Address not available
Thu Jan 21 16:50:22 2016 authpriv.info dropbear[2072]: Not backgrounding
root@ImmortalWrt:/# 
```

FirYuen, 2021-11-09:
正常启动的时候在上面的打印后面会有如下打印
```
Thu Jan 21 16:50:21 2016 kern.info kernel: [   14.751064] br-lan: port 1(eth1) entered disabled state
Thu Jan 21 16:50:21 2016 daemon.err odhcpd[1558]: Failed to send to ff02::1%lan@br-lan (Address not available)
Thu Jan 21 16:50:21 2016 authpriv.warn dropbear[2467]: Failed listening on '22': Error listening: Address not available
Thu Jan 21 16:50:21 2016 authpriv.info dropbear[2467]: Not backgrounding
Thu Jan 21 16:50:22 2016 daemon.err collectd[2690]: plugin_load: plugin "iwinfo" successfully loaded.
Thu Jan 21 16:50:22 2016 daemon.err collectd[2690]: plugin_load: plugin "memory" successfully loaded.
Thu Jan 21 16:50:22 2016 daemon.err collectd[2690]: plugin_load: plugin "cpu" successfully loaded.
Thu Jan 21 16:50:22 2016 daemon.err collectd[2690]: plugin_load: plugin "load" successfully loaded.
Thu Jan 21 16:50:22 2016 daemon.err collectd[2690]: plugin_load: plugin "rrdtool" successfully loaded.
Thu Jan 21 16:50:22 2016 daemon.err collectd[2690]: rrdtool plugin: RRASingle = true: creating only AVERAGE RRAs
Thu Jan 21 16:50:22 2016 daemon.err collectd[2690]: plugin_load: plugin "interface" successfully loaded.
Thu Jan 21 16:50:22 2016 cron.err crond[2800]: crond (busybox 1.34.1) started, log level 5
Thu Jan 21 16:50:22 2016 daemon.err collectd[2690]: Initialization complete, entering read-loop.

```

我经过测试,如果插着usb网卡开机, 输入 iwinfo 后终端也会同样卡死,命令阻塞,无输出
```
[   14.379457] rk_gmac-dwmac ff540000.ethernet eth0: PTP not supported by HW
[   14.380734] rk_gmac-dwmac ff540000.ethernet eth0: configuring for phy/rgmii link mode

BusyBox v1.34.1 (2021-11-03 07:19:36 UTC) built-in shell (ash)

┌─────────────────────────────────────────────┐
│                                             │
│ mmmmm                         m       ""#   │
│   #   mmmmm mmmmm  mmm  mmm mm#mm  mmm  #   │
│   #   # # # # # # #" "# #" "  #   "   # #   │
│   #   # # # # # # #   # #     #   m"""# #   │
│ mm#mm # # # # # # "#m#" #     "mm "mm"# "mm │
│                                             │
│─────────────────────────────────────────────│
│            ImmortalWrt SNAPSHOT             │
└─────────────────────────────────────────────┘
root@ImmortalWrt:/# 
root@ImmortalWrt:/# 
root@ImmortalWrt:/# 
root@ImmortalWrt:/# iwinfo 

```

AmadeusGhost, 2021-11-09:
Trying this patch, it seems we only need to change this on kernel 5.10.
By the way, the rtl wifi driver on 18.06 branch has not changed recently.
```
diff --git a/target/linux/rockchip/patches-5.10/102-rockchip-enable-LAN-port-on-NanoPi-R2S.patch b/target/linux/rockchip/patches-5.10/102-rockchip-enable-LAN-port-on-NanoPi-R2S.patch
index 82e2137a8b..e6b1a2fede 100644
--- a/target/linux/rockchip/patches-5.10/102-rockchip-enable-LAN-port-on-NanoPi-R2S.patch
+++ b/target/linux/rockchip/patches-5.10/102-rockchip-enable-LAN-port-on-NanoPi-R2S.patch
@@ -13,38 +13,38 @@ Signed-off-by: David Bauer <mail@david-bauer.net>
 
 --- a/arch/arm64/boot/dts/rockchip/rk3328-nanopi-r2s.dts
 +++ b/arch/arm64/boot/dts/rockchip/rk3328-nanopi-r2s.dts
-@@ -44,6 +44,18 @@
- 		};
+@@ -103,6 +103,18 @@
+ 		regulator-min-microvolt = <5000000>;
+ 		regulator-max-microvolt = <5000000>;
  	};
- 
-+	vcc_rtl8153: vcc-rtl8153-regulator {
++
++	vdd_5v_lan: vdd-5v-lan {
 +		compatible = "regulator-fixed";
++		enable-active-high;
 +		gpio = <&gpio2 RK_PC6 GPIO_ACTIVE_HIGH>;
++		pinctrl-0 = <&lan_vdd_pin>;
 +		pinctrl-names = "default";
-+		pinctrl-0 = <&rtl8153_en_drv>;
++		regulator-name = "vdd_5v_lan";
 +		regulator-always-on;
-+		regulator-name = "vcc_rtl8153";
-+		regulator-min-microvolt = <5000000>;
-+		regulator-max-microvolt = <5000000>;
-+		enable-active-high;
++		regulator-boot-on;
++		vin-supply = <&vdd_5v>;
 +	};
-+
- 	leds {
- 		compatible = "gpio-leds";
- 		pinctrl-0 = <&lan_led_pin>,  <&sys_led_pin>, <&wan_led_pin>;
-@@ -273,6 +285,12 @@
- 			};
+ };
+ 
+ &cpu0 {
+@@ -313,6 +325,12 @@
  		};
  	};
-+
-+	usb {
-+		rtl8153_en_drv: rtl8153-en-drv {
+ 
++	lan {
++		lan_vdd_pin: lan-vdd-pin {
 +			rockchip,pins = <2 RK_PC6 RK_FUNC_GPIO &pcfg_pull_none>;
 +		};
 +	};
- };
- 
- &io_domains {
++
+ 	pmic {
+ 		pmic_int_l: pmic-int-l {
+ 			rockchip,pins = <1 RK_PD0 RK_FUNC_GPIO &pcfg_pull_up>;
 @@ -379,3 +397,12 @@
  &usb_host0_ohci {
  	status = "okay";
```

1715173329, 2021-11-22:
Ping @FirYuen.

AmadeusGhost, 2021-11-27:
r2s 5.10 内核实测，插上 cf811ac 后重启几次均无问题。此 issue 视为无效，实在有问题建议找友善公司解决。

closed 2021-11-27 by AmadeusGhost