immortalwrt/immortalwrt #507: ...R2S 掉网卡，usb的那个，固件18.06 K5.4的

tao593, 2021-11-13:
[  352.668347] br-lan: port 1(eth1) entered disabled state
[  352.671153] r8152 4-1:1.0 eth1: carrier off
[  365.085025] r8152 4-1:1.0 eth1: Promiscuous mode enabled
[  365.085837] r8152 4-1:1.0 eth1: carrier on
[  365.086794] br-lan: port 1(eth1) entered blocking state
[  365.087270] br-lan: port 1(eth1) entered forwarding state
[  451.355260] xhci-hcd xhci-hcd.0.auto: WARN: HC couldn't access mem fast enough for slot 1 ep 6
[  451.356060] r8152 4-1:1.0 eth1: intr status -63
[  460.075763] br-lan: port 1(eth1) entered disabled state
[  460.081277] r8152 4-1:1.0 eth1: carrier off
[  474.507263] xhci-hcd xhci-hcd.0.auto: WARN: HC couldn't access mem fast enough for slot 1 ep 6
[  474.508064] r8152 4-1:1.0 eth1: intr status -63
[  474.795257] xhci-hcd xhci-hcd.0.auto: WARN: HC couldn't access mem fast enough for slot 1 ep 6
[  474.796054] r8152 4-1:1.0 eth1: intr status -63
[  476.204995] r8152 4-1:1.0 eth1: Promiscuous mode enabled
[  476.205730] r8152 4-1:1.0 eth1: carrier on
[  476.206667] br-lan: port 1(eth1) entered blocking state
[  476.207211] br-lan: port 1(eth1) entered forwarding state

tao593, 2021-11-13:
它会自己恢复，然后网口速率变成10Mbps了，必须重启，之后隔断时间往复循环

1715173329, 2021-11-13:
switching to use kmod-usb-net-rtl8152-vendor may help, idk, it mostly depends on the physique of your device, and some magic.

AmadeusGhost, 2021-11-13:
唯一的建议是找友善退钱。

tao593, 2021-11-13:
会不是是内存卡的原因，我在前几天保存配置时设备文件系统崩溃，重刷了固件恢复然后才出现这个掉LAN口的，之前一直没有遇到。

tao593, 2021-11-13:
我刚刚已经替换了<*> kmod-usb-net-rtl8152  为kmod-usb-net-rtl8152-vendor  我再观察一下看看，希望不要再有问题了，你们有推荐的设备吗，我也想换了，这东西毛病太多了。

closed 2021-11-14 by AmadeusGhost