immortalwrt/immortalwrt #511: [18.06] asus rt-acrh17路由 wifi无法连接，持续报错：hostpad wlan1:Could not add STA to kernel driver

dansir118, 2021-11-20:
硬件型号：asus rt-acrh17
源码版本：openwrt-18.06: 9faf432 / packages: 6803cad / luci: e56bc0a / routing: 785f1bb
问题：2.4G/5G wifi能搜索到，但均无法连接，系统日志报错如下：
`
Sat Nov 20 10:16:35 2021 kern.warn kernel: [   12.417030] ath10k_pci 0000:01:00.0: Direct firmware load for ath10k/QCA9984/hw1.0/firmware-6.bin
Sat Nov 20 10:16:35 2021 kern.warn kernel: [   12.417079] ath10k_pci 0000:01:00.0: Load Wireless ART files
Sat Nov 20 10:16:35 2021 kern.err kernel: [   12.480260] firmware ath10k!QCA9984!hw1.0!firmware-6.bin: firmware_loading_store: map pages failed
`

`
Sat Nov 20 10:16:35 2021 kern.warn kernel: [   21.244927] ath10k_ahb a000000.wifi: Direct firmware load for ath10k/QCA4019/hw1.0/firmware-6.bin
Sat Nov 20 10:16:35 2021 kern.warn kernel: [   21.244969] ath10k_ahb a000000.wifi: Load Wireless ART files
Sat Nov 20 10:16:35 2021 kern.err kernel: [   21.433154] firmware ath10k!QCA4019!hw1.0!firmware-6.bin: firmware_loading_store: map pages failed
`

持续报错：
`
Sat Nov 20 10:20:13 2021 daemon.notice hostapd: wlan0: STA 22:f8:5e:6d:0f:b4 IEEE 802.11: Could not add STA to kernel driver
Sat Nov 20 10:20:13 2021 daemon.notice hostapd: wlan0: STA 22:f8:5e:6d:0f:b4 IEEE 802.11: did not acknowledge authentication response
`

`
Sat Nov 20 10:20:45 2021 daemon.notice hostapd: wlan1: STA d8:2f:e6:e5:2b:cf IEEE 802.11: Could not add STA to kernel driver
Sat Nov 20 10:20:45 2021 daemon.notice hostapd: wlan1: STA d8:2f:e6:e5:2b:cf IEEE 802.11: did not acknowledge authentication response
`

AmadeusGhost, 2021-11-21:
Hi，我全新编译了一个竞斗云的不保留配置刷入（18.06 分支），初步试了下 wifi 连接没有问题，也许能做为参考（

dansir118, 2021-11-23:
> Hi，我全新编译了一个竞斗云的不保留配置刷入（18.06 分支），初步试了下 wifi 连接没有问题，也许能做为参考（

此型号与17的CPU同为IPQ4019，但5G WiFi一个为IPQ4019内置，一个为QCA9984，略有不同；
同时我也编译了一个默认配置的8M大小固件，不保留配置、从opboot干净刷入，反复测试。
排除第三方插件因素，WiFi无法连接问题依旧，报错都是这个。

Aspiriniii, 2021-11-26:
hostapd报Could not add STA to kernel driver，这个问题18.06分支的确存在，我两个机器mi-nano（mt7628）和xiaomi-mini（mt7620）都可以复现

AmadeusGhost, 2021-12-04:
> 排除第三方插件、编译缓存、旧固件保留配置因素，WiFi无法连接问题依旧，报错都是这个。

编译个 4.19 内核的固件试试？4.14 内核群友测试确实有这问题。

1715173329, 2021-12-17:
> hostapd报Could not add STA to kernel driver，这个问题18.06分支的确存在，我两个机器mi-nano（mt7628）和xiaomi-mini（mt7620）都可以复现，这时候手机连接提示接入点满载

https://downloads.immortalwrt.org/?releases/18.06-SNAPSHOT/targets/ramips
请尝试使用此处提供的预编译固件，暂无法复现此问题。

1715173329, 2022-03-18:
Should be fixed in d44eb0095e8784fb8b704bfd2f68d9e57d9bcec2.

closed 2022-03-18 by 1715173329