immortalwrt/immortalwrt #542: [18.06] P&W R619AC WIFI 接入点暂时满载

houfukude, 2021-12-27:
竞斗云 2.0 刷入 release 下的 18.06 出现与 #511 同款的问题

wlan0 和 wlan1 均不可接入 显示 `接入点暂时满载`

刷入的固件地址
>https://downloads.immortalwrt.org/releases/18.06-SNAPSHOT/targets/ipq40xx/generic/immortalwrt-ipq40xx-generic-p2w_r619ac-squashfs-nand-factory.ubi

固件版本
>ImmortalWrt 18.06-SNAPSHOT r7854-8938eed10 / LuCI openwrt-18.06 branch (git-21.350.04268-4931556)

内核版本
>4.14.256

日志脱敏后如下

```
Tue Dec 28 00:09:33 2021 daemon.notice hostapd: nl80211: Failed to remove interface wlan0 from bridge br-lan: Invalid argument
Tue Dec 28 00:09:33 2021 kern.warn kernel: [ 3117.000020] ath10k_ahb a000000.wifi: peer-unmap-event: unknown peer id 1
Tue Dec 28 00:09:33 2021 kern.warn kernel: [ 3117.000135] ath10k_ahb a000000.wifi: peer-unmap-event: unknown peer id 1
Tue Dec 28 00:09:33 2021 daemon.notice netifd: radio0 (9005): command failed: Not supported (-95)
Tue Dec 28 00:09:33 2021 user.notice mac80211: Failed command: iw phy phy0 set distance 0
Tue Dec 28 00:09:38 2021 daemon.notice netifd: radio0 (9075): command failed: Not supported (-95)
Tue Dec 28 00:09:38 2021 user.notice mac80211: Failed command: iw phy phy0 set distance 0
Tue Dec 28 00:09:38 2021 daemon.err hostapd: Configuration file: /var/run/hostapd-phy0.conf
Tue Dec 28 00:09:40 2021 kern.info kernel: [ 3124.224555] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
Tue Dec 28 00:09:40 2021 kern.info kernel: [ 3124.231284] br-lan: port 2(wlan0) entered blocking state
Tue Dec 28 00:09:40 2021 kern.info kernel: [ 3124.231339] br-lan: port 2(wlan0) entered disabled state
Tue Dec 28 00:09:40 2021 kern.info kernel: [ 3124.236034] device wlan0 entered promiscuous mode
Tue Dec 28 00:09:40 2021 daemon.notice hostapd: wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
Tue Dec 28 00:09:40 2021 daemon.notice hostapd: ACS: Automatic channel selection started, this may take a bit
Tue Dec 28 00:09:40 2021 daemon.notice hostapd: wlan0: interface state COUNTRY_UPDATE->ACS
Tue Dec 28 00:09:40 2021 daemon.notice hostapd: wlan0: ACS-STARTED
Tue Dec 28 00:09:41 2021 daemon.notice hostapd: handle_probe_req: send failed
Tue Dec 28 00:09:48 2021 daemon.notice hostapd: wlan1: STA 00:00:00:00:00:00 IEEE 802.11: Could not add STA to kernel driver
Tue Dec 28 00:09:48 2021 daemon.notice hostapd: wlan1: STA 00:00:00:00:00:00 IEEE 802.11: did not acknowledge authentication response
Tue Dec 28 00:09:49 2021 daemon.notice hostapd: wlan0: ACS-COMPLETED freq=2437 channel=6
Tue Dec 28 00:09:49 2021 daemon.err hostapd: Using interface wlan0 with hwaddr 00:00:00:00:00:01 and ssid "\xe4\xb8\xa..."
Tue Dec 28 00:09:49 2021 kern.info kernel: [ 3133.132030] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
Tue Dec 28 00:09:49 2021 kern.info kernel: [ 3133.132277] br-lan: port 2(wlan0) entered blocking state
Tue Dec 28 00:09:49 2021 kern.info kernel: [ 3133.137489] br-lan: port 2(wlan0) entered forwarding state
Tue Dec 28 00:09:49 2021 daemon.notice hostapd: wlan0: interface state ACS->ENABLED
Tue Dec 28 00:09:49 2021 daemon.notice hostapd: wlan0: AP-ENABLED
Tue Dec 28 00:09:49 2021 daemon.notice netifd: Network device 'wlan0' link is up
Tue Dec 28 00:09:50 2021 daemon.info hostapd: wlan0: STA 00:00:00:00:00:02 IEEE 802.11: authenticated
Tue Dec 28 00:09:50 2021 daemon.info hostapd: wlan0: STA 00:00:00:00:00:02 IEEE 802.11: associated (aid 1)
Tue Dec 28 00:09:50 2021 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 00:00:00:00:00:02
Tue Dec 28 00:09:50 2021 daemon.info hostapd: wlan0: STA 00:00:00:00:00:02 RADIUS: starting accounting session A7913F0A68A5E199
Tue Dec 28 00:09:50 2021 daemon.info hostapd: wlan0: STA 00:00:00:00:00:02 WPA: pairwise key handshake completed (RSN)
Tue Dec 28 00:09:53 2021 daemon.info hostapd: wlan0: STA 00:00:00:00:00:03 IEEE 802.11: authenticated
Tue Dec 28 00:09:53 2021 daemon.info hostapd: wlan0: STA 00:00:00:00:00:03 IEEE 802.11: associated (aid 2)
Tue Dec 28 00:09:53 2021 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 00:00:00:00:00:03
Tue Dec 28 00:09:53 2021 daemon.info hostapd: wlan0: STA 00:00:00:00:00:03 RADIUS: starting accounting session 72EAFA6A0768B45F
Tue Dec 28 00:09:53 2021 daemon.info hostapd: wlan0: STA 00:00:00:00:00:03 WPA: pairwise key handshake completed (RSN)
Tue Dec 28 00:09:57 2021 daemon.notice hostapd: wlan0: STA 00:00:00:00:00:04 IEEE 802.11: Could not add STA to kernel driver
Tue Dec 28 00:09:57 2021 daemon.notice hostapd: wlan0: STA 00:00:00:00:00:04 IEEE 802.11: did not acknowledge authentication response
Tue Dec 28 00:10:03 2021 daemon.info hostapd: wlan1: STA 00:00:00:00:00:05 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Tue Dec 28 00:10:05 2021 daemon.notice hostapd: wlan1: STA 00:00:00:00:00:04 IEEE 802.11: Could not add STA to kernel driver
Tue Dec 28 00:10:05 2021 daemon.notice hostapd: wlan1: STA 00:00:00:00:00:04 IEEE 802.11: did not acknowledge authentication response
...
Tue Dec 28 00:11:17 2021 daemon.notice hostapd: wlan0: STA 00:00:00:00:00:04 IEEE 802.11: Could not add STA to kernel driver
Tue Dec 28 00:11:17 2021 daemon.notice hostapd: wlan0: STA 00:00:00:00:00:04 IEEE 802.11: did not acknowledge authentication response
Tue Dec 28 00:11:21 2021 kern.info kernel: [ 3225.797110] device wlan1 left promiscuous mode
Tue Dec 28 00:11:21 2021 kern.info kernel: [ 3225.797262] br-lan: port 3(wlan1) entered disabled state
Tue Dec 28 00:11:22 2021 daemon.notice hostapd: wlan1: interface state ENABLED->DISABLED
Tue Dec 28 00:11:22 2021 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED 00:00:00:00:00:06
Tue Dec 28 00:11:22 2021 daemon.notice hostapd: wlan1: AP-DISABLED
Tue Dec 28 00:11:22 2021 daemon.notice hostapd: wlan1: CTRL-EVENT-TERMINATING
Tue Dec 28 00:11:22 2021 daemon.notice hostapd: nl80211: deinit ifname=wlan1 disabled_11b_rates=0
Tue Dec 28 00:11:22 2021 daemon.notice hostapd: nl80211: Failed to remove interface wlan1 from bridge br-lan: Invalid argument
Tue Dec 28 00:11:22 2021 kern.warn kernel: [ 3226.069922] ath10k_ahb a800000.wifi: peer-unmap-event: unknown peer id 1
[... 209 lines cut ...]
Tue Dec 28 00:21:27 2021 daemon.info hostapd: wlan0: STA 00:00:00:00:00:02 IEEE 802.11: associated (aid 1)
Tue Dec 28 00:21:27 2021 daemon.info hostapd: wlan0: STA 00:00:00:00:00:02 RADIUS: starting accounting session A7913F0A68A5E199
Tue Dec 28 00:21:27 2021 daemon.info hostapd: wlan0: STA 00:00:00:00:00:02 WPA: pairwise key handshake completed (RSN)
Tue Dec 28 00:22:08 2021 daemon.notice hostapd: wlan1: STA 00:00:00:00:00:00 IEEE 802.11: Could not add STA to kernel driver
Tue Dec 28 00:22:08 2021 daemon.notice hostapd: wlan1: STA 00:00:00:00:00:00 IEEE 802.11: did not acknowledge authentication response
...
Tue Dec 28 00:24:11 2021 daemon.notice hostapd: wlan0: STA 00:00:00:00:00:00 IEEE 802.11: Could not add STA to kernel driver
Tue Dec 28 00:24:11 2021 daemon.notice hostapd: wlan0: STA 00:00:00:00:00:00 IEEE 802.11: did not acknowledge authentication response
Tue Dec 28 00:24:23 2021 daemon.info hostapd: wlan0: STA 00:00:00:00:00:02 IEEE 802.11: authenticated
Tue Dec 28 00:24:25 2021 daemon.info hostapd: wlan0: STA 00:00:00:00:00:02 IEEE 802.11: authenticated
Tue Dec 28 00:24:25 2021 daemon.info hostapd: wlan0: STA 00:00:00:00:00:02 IEEE 802.11: associated (aid 1)
Tue Dec 28 00:24:26 2021 daemon.info hostapd: wlan0: STA 00:00:00:00:00:02 RADIUS: starting accounting session A7913F0A68A5E199
Tue Dec 28 00:24:26 2021 daemon.info hostapd: wlan0: STA 00:00:00:00:00:02 WPA: pairwise key handshake completed (RSN)
Tue Dec 28 00:24:30 2021 daemon.notice hostapd: wlan0: STA 00:00:00:00:00:04 IEEE 802.11: Could not add STA to kernel driver
Tue Dec 28 00:24:30 2021 daemon.notice hostapd: wlan0: STA 00:00:00:00:00:04 IEEE 802.11: did not acknowledge authentication response
Tue Dec 28 00:24:50 2021 daemon.notice hostapd: wlan0: STA 00:00:00:00:00:04 IEEE 802.11: Could not add STA to kernel driver
Tue Dec 28 00:24:50 2021 daemon.notice hostapd: wlan0: STA 00:00:00:00:00:04 IEEE 802.11: did not acknowledge authentication response
Tue Dec 28 00:25:06 2021 daemon.info hostapd: wlan0: STA 00:00:00:00:00:00 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Tue Dec 28 00:25:06 2021 daemon.info hostapd: wlan0: STA 00:00:00:00:00:04 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Tue Dec 28 00:25:27 2021 daemon.info hostapd: wlan1: STA 00:00:00:00:00:05 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Tue Dec 28 00:25:29 2021 daemon.notice hostapd: wlan1: STA 00:00:00:00:00:05 IEEE 802.11: Could not add STA to kernel driver
Tue Dec 28 00:25:29 2021 daemon.notice hostapd: wlan1: STA 00:00:00:00:00:05 IEEE 802.11: did not acknowledge authentication response
Tue Dec 28 00:25:36 2021 daemon.info hostapd: wlan0: STA 00:00:00:00:00:05 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Tue Dec 28 00:25:50 2021 daemon.notice hostapd: wlan0: STA 00:00:00:00:00:04 IEEE 802.11: Could not add STA to kernel driver
```

houfukude, 2021-12-27:
补充:  貌似 Android 设备中 `隐私` 设置成 `使用随机MAC` 可接入

```
Dec 28 00:55:12 R619AC hostapd: wlan1: STA ae:19:ba:ae:05:d3 IEEE 802.11: authenticated
Dec 28 00:55:12 R619AC hostapd: wlan1: STA ae:19:ba:ae:05:d3 IEEE 802.11: associated (aid 2)
Dec 28 00:55:12 R619AC hostapd: wlan1: AP-STA-CONNECTED ae:19:ba:ae:05:d3
Dec 28 00:55:12 R619AC hostapd: wlan1: STA ae:19:ba:ae:05:d3 RADIUS: starting accounting session 25BCE3899D068532
Dec 28 00:55:12 R619AC hostapd: wlan1: STA ae:19:ba:ae:05:d3 WPA: pairwise key handshake completed (RSN)
```

hh1100, 2022-02-11:
我编译newifi n1也是这个问题， ap满载无法接入 用的也是18.06分支  。21.02没这个问题

jxph1123, 2022-02-16:
我小米3g也是这样，甚至部分手机都接入不了，一样的18.06默认分支

1715173329, 2022-03-16:
Should be fixed in d44eb0095e8784fb8b704bfd2f68d9e57d9bcec2.

jxph1123, 2022-03-18:
感谢大佬，问题已解决

closed 2022-03-18 by 1715173329