immortalwrt/immortalwrt #575: ACRH17刷入从发布页面下载的固件后WIFI无法启动

yubanmeiqin9048, 2022-01-18:
Sat Dec 18 07:07:07 2021 daemon.notice netifd: radio0 (1765): cat: can't open '/var/run/wifi-phy0.pid': No such file or directory Sat Dec 18 07:07:07 2021 daemon.notice netifd: radio1 (1766): cat: can't open '/var/run/wifi-phy1.pid': No such file or directory Sat Dec 18 07:07:07 2021 daemon.notice netifd: radio0 (1765): WARNING (wireless_add_process): executable path /usr/sbin/wpad does not match process  path (/proc/exe) Sat Dec 18 07:07:07 2021 daemon.notice netifd: radio1 (1766): WARNING (wireless_add_process): executable path /usr/sbin/wpad does not match process  path (/proc/exe) Sat Dec 18 07:07:07 2021 daemon.notice netifd: radio0 (1765): Command failed: Invalid argument Sat Dec 18 07:07:07 2021 daemon.notice netifd: radio1 (1766): Command failed: Invalid argument Sat Dec 18 07:07:07 2021 daemon.notice netifd: radio0 (1765): Device setup failed: HOSTAPD_START_FAILED Sat Dec 18 07:07:07 2021 daemon.notice netifd: radio1 (1766): Device setup failed: HOSTAPD_START_FAILED

1715173329, 2022-01-19:
什么版本？

yubanmeiqin9048, 2022-01-19:
> 什么版本？

18.06

AmadeusGhost, 2022-01-19:
你这日志不能多给一点吗？

yubanmeiqin9048, 2022-01-19:
> 你这日志不能多给一点吗？

我只截了radio的部分，需要更多我可以再装回去

yubanmeiqin9048, 2022-01-22:
https://forum.openwrt.org/t/wireless-enabled-but-not-up-solved/13862/8 中说到：using wpad-mini solved it, now my wifi is up.

1715173329, 2022-03-16:
Should be fixed in d44eb0095e8784fb8b704bfd2f68d9e57d9bcec2.

closed 2022-03-18 by 1715173329