immortalwrt/immortalwrt #610: 18.06分支红米AC2100无线打开失败

cncherisher, 2022-03-02:
版本号
```
ImmortalWrt 18.06-SNAPSHOT r7926-d3e63a637e / LuCI openwrt-18.06 branch (git-22.059.46782-21f95c2)
```

内核日志
```
[   14.937366] register mt_drv
[   14.940234] bus=0x1, slot = 0x0, irq=0x0
[   14.944157] PCI: Enabling device 0000:00:00.0 (0004 -> 0006)
[   14.964547] 
[   14.964547] == pAd = c0b81000, size = 3982368, Status=0 ==
[   14.971545] pAd->PciHif.CSRBaseAddress =0xc0a80000, csr_addr=0xc0a80000!
[   14.978227] RTMPInitPCIeDevice():device_id=0x7615
[   14.982913] mt_pci_chip_cfg(): HWVer=0x8a10, FWVer=0x8a10, pAd->ChipID=0x7615
[   14.990025] mt_pci_chip_cfg(): HIF_SYS_REV=0x76150001
[   14.995048] AP Driver version-5.0.2.0
[   14.998705] RtmpChipOpsHook(223): Not support for HIF_MT yet! MACVersion=0x0
[   15.005729] mt7615_init()-->
[   15.008593] Use the default ePAeLNA bin image!
[   15.013008] Use the default /etc_ro/wlan/MT7615E_EEPROM1.bin bin image!
[   15.019618] <--mt7615_init()
[   15.026344] <-- RTMPAllocTxRxRingMemory, Status=0
[   15.077699] ctnetlink v0.93: registering with nfnetlink.
[   15.160575] xt_time: kernel timezone is -0000
[   15.229548] bus=0x2, slot = 0x1, irq=0xff
[   15.233582] PCI: Enabling device 0000:00:01.0 (0004 -> 0006)
[   15.239435] mt7603e 0000:02:00.0: ASIC revision: 76030010
[   15.246695] mt7603e 0000:02:00.0: Invalid MAC address, using random address 1a:15:28:c0:b3:54
[   16.276645] mt7603e 0000:02:00.0: Firmware Version: ap_pcie
[   16.282212] mt7603e 0000:02:00.0: Build Time: 20160107100755
[   16.325461] mt7603e 0000:02:00.0: firmware init done
```
系统日志
```
Tue Mar  1 20:53:30 2022 user.notice firewall: Reloading firewall due to ifup of lan (br-lan)
Tue Mar  1 20:53:30 2022 daemon.err odhcp6c[2202]: Failed to send RS (Address not available)
Tue Mar  1 20:53:30 2022 daemon.notice netifd: wan (2208): udhcpc: broadcasting discover
Tue Mar  1 20:53:30 2022 user.notice mac80211: Failed command: iw phy phy0 set antenna 0xffffffff 0xffffffff
Tue Mar  1 20:53:30 2022 kern.info kernel: [   39.956483] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
Tue Mar  1 20:53:30 2022 daemon.err odhcpd[1833]: Failed to send to ff02::1%lan@br-lan (Address not available)
Tue Mar  1 20:53:30 2022 authpriv.warn dropbear[2388]: Failed listening on '22': Error listening: Address not available
Tue Mar  1 20:53:30 2022 authpriv.info dropbear[2388]: Not backgrounding
Tue Mar  1 20:53:31 2022 daemon.err odhcp6c[2202]: Failed to send SOLICIT message to ff02::1:2 (Address not available)
Tue Mar  1 20:53:31 2022 daemon.err hostapd: Configuration file: /var/run/hostapd-phy0.conf
Tue Mar  1 20:53:31 2022 daemon.err hostapd: Could not set interface wlan0 flags (UP): Address not available
Tue Mar  1 20:53:31 2022 daemon.err hostapd: nl80211: Could not set interface 'wlan0' UP
Tue Mar  1 20:53:31 2022 daemon.notice hostapd: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Tue Mar  1 20:53:31 2022 daemon.err hostapd: nl80211 driver initialization failed.
Tue Mar  1 20:53:31 2022 daemon.notice hostapd: wlan0: interface state UNINITIALIZED->DISABLED
Tue Mar  1 20:53:31 2022 daemon.notice hostapd: wlan0: AP-DISABLED
Tue Mar  1 20:53:31 2022 daemon.notice hostapd: wlan0: CTRL-EVENT-TERMINATING
Tue Mar  1 20:53:31 2022 daemon.err hostapd: hostapd_free_hapd_data: Interface wlan0 wasn't started
Tue Mar  1 20:53:31 2022 daemon.notice netifd: radio0 (2144): cat: can't open '/var/run/wifi-phy0.pid': No such file or directory
Tue Mar  1 20:53:31 2022 daemon.notice netifd: radio0 (2144): WARNING (wireless_add_process): executable path /usr/sbin/wpad does not match process  path (/proc/exe)
Tue Mar  1 20:53:31 2022 daemon.notice netifd: radio0 (2144): Command failed: Invalid argument
Tue Mar  1 20:53:31 2022 daemon.notice netifd: radio0 (2144): Device setup failed: HOSTAPD_START_FAILED
```
网络->无线
```
[radio0: 未知 "OpenWrt"]
MediaTek MT7603E 802.11bgn (radio0)
SSID: OpenWrt | 模式: Unknown
无线未开启或未关联
```
无论是否编译luci-app-mtwifi均无法启动无线

cncherisher, 2022-03-02:
使用下载站的4.14固件，可以开启无线但是手机无法连接，提示接入点暂时满载
```
https://downloads.immortalwrt.org/?releases/18.06-SNAPSHOT/targets/ramips/mt7621
ImmortalWrt 18.06-SNAPSHOT r7810+10-43792ff18 / LuCI openwrt-18.06 branch (git-21.336.18025-ae7d404)
```
系统日志
```
Wed Mar  2 10:17:34 2022 daemon.notice hostapd: wlan1: STA 94:87:e0:1e:b4:06 IEEE 802.11: Could not add STA to kernel driver
Wed Mar  2 10:17:34 2022 daemon.notice hostapd: wlan1: STA 94:87:e0:1e:b4:06 IEEE 802.11: did not acknowledge authentication response
Wed Mar  2 10:17:37 2022 daemon.notice hostapd: wlan1: STA 94:87:e0:1e:b4:06 IEEE 802.11: Could not add STA to kernel driver
Wed Mar  2 10:17:37 2022 daemon.notice hostapd: wlan1: STA 94:87:e0:1e:b4:06 IEEE 802.11: did not acknowledge authentication response
Wed Mar  2 10:17:39 2022 daemon.notice hostapd: wlan1: STA 94:87:e0:1e:b4:06 IEEE 802.11: Could not add STA to kernel driver
Wed Mar  2 10:17:39 2022 daemon.notice hostapd: wlan1: STA 94:87:e0:1e:b4:06 IEEE 802.11: did not acknowledge authentication response
Wed Mar  2 10:17:40 2022 daemon.notice hostapd: wlan1: STA 94:87:e0:1e:b4:06 IEEE 802.11: Could not add STA to kernel driver
Wed Mar  2 10:17:40 2022 daemon.notice hostapd: wlan1: STA 94:87:e0:1e:b4:06 IEEE 802.11: did not acknowledge authentication response
Wed Mar  2 10:17:46 2022 kern.info kernel: [  213.839416] device wlan1 left promiscuous mode
Wed Mar  2 10:17:46 2022 kern.info kernel: [  213.844037] br-lan: port 3(wlan1) entered disabled state
Wed Mar  2 10:17:46 2022 daemon.notice hostapd: wlan1: interface state ENABLED->DISABLED
Wed Mar  2 10:17:46 2022 daemon.notice hostapd: wlan1: AP-DISABLED
Wed Mar  2 10:17:46 2022 daemon.notice hostapd: wlan1: CTRL-EVENT-TERMINATING
Wed Mar  2 10:17:46 2022 daemon.notice hostapd: nl80211: deinit ifname=wlan1 disabled_11b_rates=0
Wed Mar  2 10:17:46 2022 daemon.notice hostapd: nl80211: Failed to remove interface wlan1 from bridge br-lan: Invalid argument
Wed Mar  2 10:17:48 2022 user.notice mac80211: Failed command: iw phy phy1 set antenna 0xffffffff 0xffffffff
Wed Mar  2 10:17:48 2022 daemon.notice netifd: radio1 (2917): command failed: Not supported (-122)
Wed Mar  2 10:17:48 2022 user.notice mac80211: Failed command: iw phy phy1 set distance 0
Wed Mar  2 10:18:30 2022 user.notice mac80211: Failed command: iw phy phy0 set antenna 0xffffffff 0xffffffff
Wed Mar  2 10:18:31 2022 daemon.err hostapd: Configuration file: /var/run/hostapd-phy0.conf
Wed Mar  2 10:18:31 2022 kern.info kernel: [  258.304171] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
Wed Mar  2 10:18:31 2022 kern.info kernel: [  258.320824] br-lan: port 2(wlan0) entered blocking state
Wed Mar  2 10:18:31 2022 kern.info kernel: [  258.326219] br-lan: port 2(wlan0) entered disabled state
Wed Mar  2 10:18:31 2022 kern.info kernel: [  258.332206] device wlan0 entered promiscuous mode
Wed Mar  2 10:18:31 2022 daemon.err hostapd: Using interface wlan0 with hwaddr d6:b6:69:1b:43:b0 and ssid "OpenWrt"
Wed Mar  2 10:18:31 2022 kern.info kernel: [  258.349840] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
Wed Mar  2 10:18:31 2022 kern.info kernel: [  258.356520] br-lan: port 2(wlan0) entered blocking state
Wed Mar  2 10:18:31 2022 kern.info kernel: [  258.361840] br-lan: port 2(wlan0) entered forwarding state
Wed Mar  2 10:18:31 2022 daemon.notice hostapd: wlan0: interface state UNINITIALIZED->ENABLED
Wed Mar  2 10:18:31 2022 daemon.notice hostapd: wlan0: AP-ENABLED
Wed Mar  2 10:18:31 2022 daemon.notice netifd: Network device 'wlan0' link is up
Wed Mar  2 10:18:52 2022 daemon.notice hostapd: wlan0: STA 94:87:e0:1e:b4:06 IEEE 802.11: Could not add STA to kernel driver
Wed Mar  2 10:18:52 2022 daemon.notice hostapd: wlan0: STA 94:87:e0:1e:b4:06 IEEE 802.11: did not acknowledge authentication response

```

AmadeusGhost, 2022-03-03:
@skbeh mt7615 的闭源驱动好像是你上次改的来着（

1715173329, 2022-03-03:
18.06 4.14 的代码应该没有更改过？

skbeh, 2022-03-05:
@AmadeusGhost 
`AP Driver version-5.0.2.0`
不是我改过的版本

jxph1123, 2022-03-07:
18.06版本的7621一直有这个问题，无线满载，手机连不上，不知道什么原因

jxph1123, 2022-03-18:
感谢大佬，问题已解决

closed 2022-03-21 by AmadeusGhost