immortalwrt/immortalwrt #615: 进你们群组 换了5个TG号 回答问题都是错误的，刷个固件是真的难，实在费解，看不懂

Assassincc, 2022-03-05:
进你们群组 换了5个TG号 回答问题都是错误的，刷个固件是真的难，实在费解，看不懂进你们群组 换了5个TG号 回答问题都是错误的，刷个固件是真的难，实在费解，看不懂进你们群组 换了5个TG号 回答问题都是错误的，刷个固件是真的难，实在费解，看不懂进你们群组 换了5个TG号 回答问题都是错误的，刷个固件是真的难，实在费解，看不懂进你们群组 换了5个TG号 回答问题都是错误的，刷个固件是真的难，实在费解，看不懂进你们群组 换了5个TG号 回答问题都是错误的，刷个固件是真的难，实在费解，看不懂进你们群组 换了5个TG号 回答问题都是错误的，刷个固件是真的难，实在费解，看不懂进你们群组 换了5个TG号 回答问题都是错误的，刷个固件是真的难，实在费解，看不懂进你们群组 换了5个TG号 回答问题都是错误的，刷个固件是真的难，实在费解，看不懂进你们群组 换了5个TG号 回答问题都是错误的，刷个固件是真的难，实在费解，看不懂进你们群组 换了5个TG号 回答问题都是错误的，刷个固件是真的难，实在费解，看不懂

Assassincc, 2022-03-05:
进你们群组 换了5个TG号 回答问题都是错误的，刷个固件是真的难，实在费解，看不懂进你们群组 换了5个TG号 回答问题都是错误的，刷个固件是真的难，实在费解，看不懂进你们群组 换了5个TG号 回答问题都是错误的，刷个固件是真的难，实在费解，看不懂

1715173329, 2022-03-05:
您好，本项目是完全开源的，您可以自行编译；预编译固件可在 downloads.immortalwrt.org 或 firmware-selector.immortalwrt.org 获取。
群组仅供日常交流解惑使用，不包含额外内容。

closed 2022-03-05 by 1715173329