immortalwrt/immortalwrt #622: Orangepi One Plus  coremark跑分

ZZMHA227, 2022-03-11:
前段时间捡了个Orangepi one plus的板子，H6芯片。用K5.4编译后跑了一下分，1.2W。这和H5的跑分差不多呀。应该不至于那么低，应该比RK3328高才对，求大佬看看哪里出了问题。

[image]

AmadeusGhost, 2022-03-11:
你 cpu 主频拉到 1.8G 了嘛？

ZZMHA227, 2022-03-11:
> 默认频率跑的，CPU频率调节没适配这款机器。
[image]

ZZMHA227, 2022-03-13:
@AmadeusGhost 佬，H6用的是H5的频率设置，能fix吗？
[image]
[image]

AmadeusGhost, 2022-03-13:
```
diff --git a/target/linux/sunxi/patches-5.4/102-arm64-dts-allwinner-h6-Enable-CPU-opp-tables-for-orangepi.patch b/target/linux/sunxi/patches-5.4/102-arm64-dts-allwinner-h6-Enable-CPU-opp-tables-for-orangepi.patch
new file mode 100644
index 000000000..2e4f3ea4e
--- /dev/null
+++ b/target/linux/sunxi/patches-5.4/102-arm64-dts-allwinner-h6-Enable-CPU-opp-tables-for-orangepi.patch
@@ -0,0 +1,38 @@
+--- a/arch/arm64/boot/dts/allwinner/sun50i-h6-orangepi.dtsi
++++ b/arch/arm64/boot/dts/allwinner/sun50i-h6-orangepi.dtsi
+@@ -7,6 +7,7 @@
+ /dts-v1/;
+ 
+ #include "sun50i-h6.dtsi"
++#include "sun50i-h6-cpu-opp.dtsi"
+ 
+ #include <dt-bindings/gpio/gpio.h>
+ 
+@@ -47,6 +48,10 @@
+ 	};
+ };
+ 
++&cpu0 {
++	cpu-supply = <&reg_dcdca>;
++};
++
+ &ehci0 {
+ 	status = "okay";
+ };
+@@ -158,13 +163,15 @@
+ 			reg_dcdca: dcdca {
+ 				regulator-always-on;
+ 				regulator-min-microvolt = <810000>;
+-				regulator-max-microvolt = <1080000>;
++				regulator-max-microvolt = <1160000>;
++				regulator-ramp-delay = <2500>;
+ 				regulator-name = "vdd-cpu";
+ 			};
+ 
+ 			reg_dcdcc: dcdcc {
+ 				regulator-min-microvolt = <810000>;
+ 				regulator-max-microvolt = <1080000>;
++				regulator-ramp-delay = <2500>;
+ 				regulator-name = "vdd-gpu";
+ 			};
+
```
你把补丁打进去编译试试吧。

ZZMHA227, 2022-03-24:
[image]
patch 打进去之后编译失败

ZZMHA227, 2022-03-24:
用lean的代码可以编译成功，但是频率还是1.2GHz,不是1.8GHz

AmadeusGhost, 2022-03-25:
你要在 18.06-k5.4 分支下测试，还有你就这样发一张 action 的图片可能只有上帝才能帮助你了，实在不会弄就本地编译。

ZZMHA227, 2022-03-25:
[image]
[image]
[image]
用的5.4分支，patches-5.4
目录下也有新加的patch，但最高还是1.2GHz，没到1.8GHz。另外，新加的patch不需要在make menuconfig里勾选吧。

ZZMHA227, 2022-03-25:
随便搜索看到这个，佬，要不要加？https://github.com/armbian/build/blob/735a3679e2de12db555e17b0485f99f86a1d106a/patch/kernel/sunxi-dev-h6/add-1.8ghz.patch
曾经试过openwrt源码，也是1.2GHz

AmadeusGhost, 2022-03-25:
发一下内核日志看看？你这个是 cpufreq 驱动根本没工作。

ZZMHA227, 2022-03-25:
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 5.4.162 (runner@fv-az132-586) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r11797-d16776bc33)) #0 SMP Fri Mar 25 02:39:55 2022
[    0.000000] Machine model: OrangePi One Plus
[    0.000000] On node 0 totalpages: 262144
[    0.000000]   DMA32 zone: 4096 pages used for memmap
[    0.000000]   DMA32 zone: 0 pages reserved
[    0.000000]   DMA32 zone: 262144 pages, LIFO batch:63
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.0
[    0.000000] percpu: Embedded 21 pages/cpu s45784 r8192 d32040 u86016
[    0.000000] pcpu-alloc: s45784 r8192 d32040 u86016 alloc=21*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 258048
[    0.000000] Kernel command line: console=ttyS0,115200 earlyprintk root=PARTUUID=5452574f-02 rootwait
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 1016188K/1048576K available (7678K kernel code, 698K rwdata, 2196K rodata, 576K init, 287K bss, 32388K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=8 to nr_cpu_ids=4.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] random: get_random_bytes called from start_kernel+0x2dc/0x42c with crng_init=0
[    0.000000] arch_timer: cp15 timer(s) running at 24.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
[    0.000004] sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 4398046511097ns
[    0.000185] Console: colour dummy device 80x25
[    0.000213] Calibrating delay loop (skipped), value calculated using timer frequency.. 48.00 BogoMIPS (lpj=240000)
[    0.000222] pid_max: default: 32768 minimum: 301
[    0.000343] Mount-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.000355] Mountpoint-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.001521] ASID allocator initialised with 32768 entries
[    0.001587] rcu: Hierarchical SRCU implementation.
[    0.001995] smp: Bringing up secondary CPUs ...
[    0.002491] Detected VIPT I-cache on CPU1
[    0.002539] CPU1: Booted secondary processor 0x0000000001 [0x410fd034]
[    0.002975] Detected VIPT I-cache on CPU2
[    0.003002] CPU2: Booted secondary processor 0x0000000002 [0x410fd034]
[    0.003402] Detected VIPT I-cache on CPU3
[    0.003427] CPU3: Booted secondary processor 0x0000000003 [0x410fd034]
[    0.003482] smp: Brought up 1 node, 4 CPUs
[    0.003498] SMP: Total of 4 processors activated.
[    0.003503] CPU features: detected: 32-bit EL0 Support
[    0.003508] CPU features: detected: CRC32 instructions
[    0.003540] CPU features: emulated: Privileged Access Never (PAN) using TTBR0_EL1 switching
[    0.003547] CPU: All CPU(s) started at EL2
[    0.003563] alternatives: patching kernel code
[    0.008071] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.008091] futex hash table entries: 1024 (order: 4, 65536 bytes, linear)
[    0.008249] pinctrl core: initialized pinctrl subsystem
[    0.008921] NET: Registered protocol family 16
[    0.009819] DMA: preallocated 2048 KiB pool for atomic allocations
[    0.022300] iommu: Default domain type: Translated 
[    0.022562] SCSI subsystem initialized
[    0.022691] libata version 3.00 loaded.
[    0.022886] usbcore: registered new interface driver usbfs
[    0.022922] usbcore: registered new interface driver hub
[    0.022956] usbcore: registered new device driver usb
[    0.023140] pps_core: LinuxPPS API ver. 1 registered
[    0.023145] pps_core: Software ver. 5.3.6 - Copyright 2005-2007 Rodolfo Giometti <giometti@linux.it>
[    0.023162] PTP clock support registered
[    0.023630] Advanced Linux Sound Architecture Driver Initialized.
[    0.023678] workqueue: max_active 576 requested for napi_workq is out of range, clamping between 1 and 512
[... 206 lines cut ...]
[    8.170071] kmodloader: loading kernel modules from /etc/modules.d/*
[    8.214718] NET: Registered protocol family 38
[    8.237653] async_tx: api initialized (async)
[    8.252852] device-mapper: ioctl: 4.41.0-ioctl (2019-09-16) initialised: dm-devel@redhat.com
[    8.301528] cryptodev: driver 1.12 loaded.
[    8.356253] Bridge firewalling registered
[    8.378187] xt_time: kernel timezone is -0000
[    8.408467] PPP generic driver version 2.4.2
[    8.413565] PPP MPPE Compression module registered
[    8.419420] NET: Registered protocol family 24
[    8.431527] kmodloader: done loading kernel modules from /etc/modules.d/*
[... cut ...]
[   11.178384] dwmac-sun8i 5020000.ethernet eth0: configuring for phy/rgmii link mode
[   11.187188] br-lan: port 1(eth0) entered blocking state
[   11.192415] br-lan: port 1(eth0) entered disabled state
[   11.197892] device eth0 entered promiscuous mode
[   15.365007] dwmac-sun8i 5020000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[   15.373413] br-lan: port 1(eth0) entered blocking state
[   15.378685] br-lan: port 1(eth0) entered forwarding state
[   15.384515] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready

ZZMHA227, 2022-03-27:
佬，这块板子还有得救吗？靠，不会全网就我一个人用这板子吧(给大佬递🚬

AmadeusGhost, 2022-03-27:
> 靠，不会全网就我一个人用这板子吧

目前来看是的，我也不知道为什么 cpufreq 驱动没加载成功。后面我还跑去 armbian 看了下相关部分，补丁并无什么差别。只能自己解决或者选择寄给我们（

ZZMHA227, 2022-03-27:
我记得有的矿难产品，我家云，用的也是H6的芯片。但它的那个cpu频率是1.8GHz，还可以动态调节。

ZZMHA227, 2022-03-27:
你们地址是啥，寄过去给你们修修看

AmadeusGhost, 2022-03-27:
> 你们地址是啥，寄过去给你们修修看

mailto: amadeus@openjmu.xyz，留下你的联系方式（最好是用 tg，其次 qq）

closed 2022-03-27 by ZZMHA227