immortalwrt/immortalwrt #640: ATH79(GL-MIFI) curl-7.82.0编译不过

CWZ7605, 2022-03-21:
Device: GL-MiFi

curl-7.82.0编译出错。
已make distclean，错误依旧。

报错详情如下：

`/home/gaozi/openwrt/imwrt-master/staging_dir/toolchain-mips_24kc_gcc-11.2.0_musl/lib/gcc/mips-openwrt-linux-musl/11.2.0/../../../../mips-openwrt-linux-musl/bin/ld: ../lib/.libs/libcurl.so: undefined reference to `Curl_sasl_can_authenticate'
/home/gaozi/openwrt/imwrt-master/staging_dir/toolchain-mips_24kc_gcc-11.2.0_musl/lib/gcc/mips-openwrt-linux-musl/11.2.0/../../../../mips-openwrt-linux-musl/bin/ld: ../lib/.libs/libcurl.so: undefined reference to `Curl_sasl_parse_url_auth_option'
/home/gaozi/openwrt/imwrt-master/staging_dir/toolchain-mips_24kc_gcc-11.2.0_musl/lib/gcc/mips-openwrt-linux-musl/11.2.0/../../../../mips-openwrt-linux-musl/bin/ld: ../lib/.libs/libcurl.so: undefined reference to `Curl_sasl_decode_mech'
/home/gaozi/openwrt/imwrt-master/staging_dir/toolchain-mips_24kc_gcc-11.2.0_musl/lib/gcc/mips-openwrt-linux-musl/11.2.0/../../../../mips-openwrt-linux-musl/bin/ld: ../lib/.libs/libcurl.so: undefined reference to `Curl_sasl_continue'
/home/gaozi/openwrt/imwrt-master/staging_dir/toolchain-mips_24kc_gcc-11.2.0_musl/lib/gcc/mips-openwrt-linux-musl/11.2.0/../../../../mips-openwrt-linux-musl/bin/ld: ../lib/.libs/libcurl.so: undefined reference to `Curl_sasl_start'
/home/gaozi/openwrt/imwrt-master/staging_dir/toolchain-mips_24kc_gcc-11.2.0_musl/lib/gcc/mips-openwrt-linux-musl/11.2.0/../../../../mips-openwrt-linux-musl/bin/ld: ../lib/.libs/libcurl.so: undefined reference to `Curl_sasl_init'
/home/gaozi/openwrt/imwrt-master/staging_dir/toolchain-mips_24kc_gcc-11.2.0_musl/lib/gcc/mips-openwrt-linux-musl/11.2.0/../../../../mips-openwrt-linux-musl/bin/ld: ../lib/.libs/libcurl.so: undefined reference to `Curl_sasl_cleanup'
collect2: error: ld returned 1 exit status
make[6]: *** [Makefile:903: curl] Error 1
make[6]: Leaving directory '/home/gaozi/openwrt/imwrt-master/build_dir/target-mips_24kc_musl/curl-7.82.0/src'
make[5]: *** [Makefile:1375: all-recursive] Error 1
make[5]: Leaving directory '/home/gaozi/openwrt/imwrt-master/build_dir/target-mips_24kc_musl/curl-7.82.0/src'
make[4]: *** [Makefile:1217: all-recursive] Error 1
make[4]: Leaving directory '/home/gaozi/openwrt/imwrt-master/build_dir/target-mips_24kc_musl/curl-7.82.0'
make[3]: *** [Makefile:185: /home/gaozi/openwrt/imwrt-master/build_dir/target-mips_24kc_musl/curl-7.82.0/.built] Error 2
make[3]: Leaving directory '/home/gaozi/openwrt/imwrt-master/feeds/packages/net/curl'
time: package/feeds/packages/curl/compile#0.55#0.07#0.59
    ERROR: package/feeds/packages/curl failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/curl/compile] Error 1
make[2]: Leaving directory '/home/gaozi/openwrt/imwrt-master'
make[1]: *** [package/Makefile:110: /home/gaozi/openwrt/imwrt-master/staging_dir/target-mips_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/gaozi/openwrt/imwrt-master'
make: *** [/home/gaozi/openwrt/imwrt-master/include/toplevel.mk:230: world] Error 2
`

ksong008, 2022-03-21:
@CWZ7605 请看已关闭的问题中有解决办法

CWZ7605, 2022-03-21:
> 

WOW, It is this that causes the problem. Thank you very much!

closed 2022-03-21 by CWZ7605